RoboStack/ros-humble #1: error C2491 "definition of dllimport function not allowed" on downstream package on Windows

traversaro, 2022-06-13:
I tried to migrate of a simple ROS2 package from RoboStack's Galactic (where it was working) to Humble (where the Windows build is not working).

The PR is https://github.com/robotology-playground/yarp-ros2/pull/31, on Humble macOS and Linux are working but windows is failing with:
~~~
2022-06-13T07:23:12.5560863Z   Building Custom Rule D:/a/yarp-ros2/yarp-ros2/ros2_interfaces_ws/src/map2d_nws_ros2_msgs/CMakeLists.txt
2022-06-13T07:23:12.6743586Z   get_map_by_name__type_support.cpp
2022-06-13T07:23:13.0765930Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(34,1): error C2491: 'map2d_nws_ros2_msgs::srv::typesupport_fastrtps_cpp::cdr_serialize': definition of dllimport function not allowed [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0795019Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(45,1): error C2491: 'map2d_nws_ros2_msgs::srv::typesupport_fastrtps_cpp::cdr_deserialize': definition of dllimport function not allowed [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0799494Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(57,1): error C2491: 'map2d_nws_ros2_msgs::srv::typesupport_fastrtps_cpp::get_serialized_size': definition of dllimport function not allowed [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0804346Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(79,1): error C2491: 'map2d_nws_ros2_msgs::srv::typesupport_fastrtps_cpp::max_serialized_size_GetMapByName_Request': definition of dllimport function not allowed [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0891238Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(190,134): warning C4273: 'rosidl_typesupport_fastrtps_cpp__get_message_type_support_handle__map2d_nws_ros2_msgs__srv__GetMapByName_Request': inconsistent dll linkage [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0893518Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs/srv/detail/get_map_by_name__rosidl_typesupport_fastrtps_cpp.hpp(74,3): message : see previous definition of 'rosidl_typesupport_fastrtps_cpp__get_message_type_support_handle__map2d_nws_ros2_msgs__srv__GetMapByName_Request' [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0895571Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(258,1): error C2491: 'map2d_nws_ros2_msgs::srv::typesupport_fastrtps_cpp::cdr_serialize': definition of dllimport function not allowed [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0898227Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(271,1): error C2491: 'map2d_nws_ros2_msgs::srv::typesupport_fastrtps_cpp::cdr_deserialize': definition of dllimport function not allowed [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0902110Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(284,1): error C2491: 'map2d_nws_ros2_msgs::srv::typesupport_fastrtps_cpp::get_serialized_size': definition of dllimport function not allowed [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0904937Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(307,1): error C2491: 'map2d_nws_ros2_msgs::srv::typesupport_fastrtps_cpp::max_serialized_size_GetMapByName_Response': definition of dllimport function not allowed [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0915119Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(421,135): warning C4273: 'rosidl_typesupport_fastrtps_cpp__get_message_type_support_handle__map2d_nws_ros2_msgs__srv__GetMapByName_Response': inconsistent dll linkage [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0917017Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs/srv/detail/get_map_by_name__rosidl_typesupport_fastrtps_cpp.hpp(151,3): message : see previous definition of 'rosidl_typesupport_fastrtps_cpp__get_message_type_support_handle__map2d_nws_ros2_msgs__srv__GetMapByName_Response' [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0929394Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs\srv\detail\dds_fastrtps\get_map_by_name__type_support.cpp(482,126): warning C4273: 'rosidl_typesupport_fastrtps_cpp__get_service_type_support_handle__map2d_nws_ros2_msgs__srv__GetMapByName': inconsistent dll linkage [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
2022-06-13T07:23:13.0932847Z D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\rosidl_typesupport_fastrtps_cpp\map2d_nws_ros2_msgs/srv/detail/get_map_by_name__rosidl_typesupport_fastrtps_cpp.hpp(171,3): message : see previous definition of 'rosidl_typesupport_fastrtps_cpp__get_service_type_support_handle__map2d_nws_ros2_msgs__srv__GetMapByName' [D:\a\yarp-ros2\yarp-ros2\build\ros2_interfaces_ws\src\map2d_nws_ros2_msgs\map2d_nws_ros2_msgs__rosidl_typesupport_fastrtps_cpp.vcxproj]
~~~

wolfv, 2022-06-13:
Hmm, strange. I didn't see this kind of issue when building any of the native message packages... Could there be some boilerplate CMake missing?

traversaro, 2022-06-28:
Related code:
* Galactic: https://github.com/ros2/rosidl_typesupport_fastrtps/blob/galactic/rosidl_typesupport_fastrtps_cpp/cmake/rosidl_typesupport_fastrtps_cpp_generate_interfaces.cmake#L119
* Humble: https://github.com/ros2/rosidl_typesupport_fastrtps/blob/humble/rosidl_typesupport_fastrtps_cpp/cmake/rosidl_typesupport_fastrtps_cpp_generate_interfaces.cmake#L125

traversaro, 2022-06-28:
Manually changing the related code to the Galactic version makes the library work, so I think that indeed it is a regression due to https://github.com/ros2/rosidl_typesupport_fastrtps/pull/87 .

traversaro, 2022-06-28:
Ok, I understood the problem, and it will probably take a bit of time for describing it and opening appropriate issue in the related repositories. However, the good news is that it occurs only if you compile a repo with `BUILD_SHARED_LIBS` set to `OFF`, if you set `BUILD_SHARED_LIBS` set to `ON` everything works fine.

Tobias-Fischer, 2022-06-28:
Ah.. shouldn't we always build shared libs in our case?

traversaro, 2022-06-28:
> Ah.. shouldn't we always build shared libs in our case?

Yes, the problem is that the downstream repo (i.e. https://github.com/robotology-playground/yarp-ros2) was hardcoding `BUILD_SHARED_LIBS` to OFF, not a problem of our packages. Indeed, I think the issue is not related to robostack packaging at all, but I would leave it open until I open an appropraite issue upstream.

Tobias-Fischer, 2022-08-02:
I think we can close this now that an upstream issue was created: https://github.com/ros2/rosidl_typesupport_fastrtps/issues/90

closed 2022-08-02 by Tobias-Fischer

traversaro, 2022-08-02:
Sure, thanks for closing. I forgot close it myself after opening the issue upstream.