RoboStack/ros-humble #3: Remaining issues

wolfv, 2022-06-26:
- [ ] macOS: rclpy segfault because of Python linking
- [ ] macOS ARM64: need to use this patch in rosidl_python https://github.com/ros2/rosidl_python/pull/164 to prevent hard coded path to numpy (that prevents cross-compilation) (it was there, then reverted: https://github.com/ros2/rosidl_python/pull/160
- [ ] macOS ARM64: need to fix SIP build in qt_gui_cpp, something is going wrong
- [ ] linux-aarch64: needs rebuild against qt-main to get all the way to desktop-full
- [ ] Rebuild everything against latest pins (incl. qt-main on all platforms)

Tobias-Fischer, 2022-08-03:
The problem for osx-64 with qt-gui-cpp is - I think - the following:

The qmake-generated makefile contains paths to system tools rather than conda-forge tools, i.e.:
```
#############################################################################
# Makefile for building: libqt_gui_cpp_sip
# Generated by qmake (3.1) (Qt 5.15.4)
# Project:  libqt_gui_cpp_sip.pro
# Template: lib
# Command: /Users/fischert/mambaforge/conda-bld/ros-humble-qt-gui-cpp_1657237172523/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/bin/qmake -o Makefile libqt_gui_cpp_sip.pro
#############################################################################

MAKEFILE      = Makefile

EQ            = =

####### Compiler, tools and options

CC            = /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang
CXX           = /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++
DEFINES       = -DSIP_PROTECTED_IS_PUBLIC -Dprotected=public -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB
CFLAGS        = -pipe -O2 $(EXPORT_ARCH_ARGS) -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk -mmacosx-version-min=10.13 -Wall -Wextra -fPIC $(DEFINES)
CXXFLAGS      = -pipe -stdlib=libc++ -O2 -std=gnu++11 $(EXPORT_ARCH_ARGS) -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk -mmacosx-version-min=10.13 -Wall -Wextra -fPIC $(DEFINES)
INCPATH       = -I. -I. -I.. -I../../../../ros-humble-qt-gui-cpp/src/work/include -I../../../../../_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/pluginlib -I../../../../../_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/ament_index_cpp -I../../../../../_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/class_loader -I../../../../../_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include -I../../../../../_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/rcpputils -I../../../../../_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/rcutils -I../../../../../_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/python3.9 -I/Users/fischert/mambaforge/conda-bld/ros-humble-qt-gui-cpp_1657237172523/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/python3.9 -I/Users/fischert/mambaforge/conda-bld/ros-humble-qt-gui-cpp_1657237172523/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/qt -I/Users/fischert/mambaforge/conda-bld/ros-humble-qt-gui-cpp_1657237172523/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/qt/QtWidgets -I/Users/fischert/mambaforge/conda-bld/ros-humble-qt-gui-cpp_1657237172523/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/qt/QtGui -I/Users/fischert/mambaforge/conda-bld/ros-humble-qt-gui-cpp_1657237172523/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/include/qt/QtCore -I. -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/AGL.framework/Headers -I/Users/fischert/mambaforge/conda-bld/ros-humble-qt-gui-cpp_1657237172523/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/mkspecs/macx-clang
QMAKE         = /Users/fischert/mambaforge/conda-bld/ros-humble-qt-gui-cpp_1657237172523/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/bin/qmak [... cut ...]
RANLIB        = /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib -s
SED           = sed
STRIP         = strip
```

I am not sure how to avoid that. This file is generated from another Makefile (also generated by qmake I think) which contains:
```
qmake: FORCE
	@$(QMAKE) -o Makefile libqt_gui_cpp_sip.pro
```

This file in turn is generated by the pyqt-builder .. looks like fun. I wonder if it's easier to patch the Makefiles and substitute these lines?

/cc @wolfv @traversaro

wolfv, 2022-08-04:
This post mentions that we could put `QMAKE_CC` and `QMAKE_CXX` in our ".pro" file (https://forum.qt.io/topic/95073/qmake-uses-wrong-gcc-version-how-to-change-gcc-version-used).

Maybe that'd help.
If we can use env vars in the `.pro` file, it would be `$CC` and `$CXX`.

Tobias-Fischer, 2022-08-04:
I think pyqt builder dynamically creates the .pro file ..

Tobias-Fischer, 2022-08-20:
Fixed the OpenGL issue on osx-64.

Tobias-Fischer, 2022-08-24:
Closing here, opened separate issues instead to easier keep track

closed 2022-08-24 by Tobias-Fischer