RoboStack/ros-humble #6: Fresh Windows installation not finding FastDDS implementation.

danzimmerman, 2022-08-21:
## Issue 
I installed ROS2 Humble on Windows today and had an issue locating the FastDDS RMW implementation.

To reproduce the error, it's sufficient to `import rosbag2_py` in Python. 

```
[ERROR] [1661106823.276844500] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), 
with error message 'failed to load shared library 'rmw_fastrtps_cpp.dll' due to LoadLibrary error: 126, 
at C:\bld\ros-humble-rcutils_1655024829144\work\ros-humble-rcutils\src\work\src\shared_library.c:159, 
at C:\bld\ros-humble-rmw-implementation_1655039337688\work\ros-humble-rmw-implementation\src\work\src\functions.cpp:67', exiting with 1., 
at C:\bld\ros-humble-rcl_1655040514413\work\ros-humble-rcl\src\work\src\rcl\rmw_implementation_identifier_check.c:145
```

The error kills the Python interpreter as well ("Kernel has died" in Jupyter or a silent exit from a Python prompt)

I have two older installs where this does not seem to have been a problem.

On one of the older installs and this fresh install, there _is_ a `rmw_fastrtps_cpp.dll` in the `C:\Users\<username>\mambaforge\envs\humble\Library\bin`, but it seems like the fresh install can't find it.

On the broken machine, I tried installing Fast DDS with the Windows installer from the eProsima website, which does resolve the error (installs the required files and sets `FASTRTPSHOME` environment variable). This does resolve the error, but I didn't need to do this in the past on the other working machines. The problem returns when I uninstall it.

## Broken Config Details

`mamba info`

<details>

```
```

</details>

`mamba list`

<details>

```
```

</details>

## Working Config Details

`mamba info`

<details>

```

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (0.22.1) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████
[... 728 lines cut ...]
## Setup

I followed the "Getting Started" Instructions with the following mods to add the `robostack-humble` channel (maybe I should delete some others?)

```
# now create a new environment
mamba create -n humble python=3.9
conda activate humble

# this adds the conda-forge channel to the new created environment configuration 
conda config --env --add channels conda-forge
# and the robostack channels
conda config --env --add channels robostack
conda config --env --add channels robostack-experimental
conda config --env --add channels robostack-humble

# Install the version of ROS you are interested in:
mamba install ros-humble-desktop  # (or: mamba install ros-noetic-desktop)

# optionally, install some compiler packages if you want to e.g. build packages in a colcon_ws:
mamba install compilers cmake pkg-config make ninja colcon-common-extensions

# on Windows, install the Visual Studio command prompt:
mamba install vs2019_win-64
```

danzimmerman, 2022-08-21:
Possibly related issues

https://github.com/ros2/ros2/issues/844
https://github.com/ros2/ros2_documentation/issues/488
https://answers.ros.org/question/356763/ros2-windows-rmw_implementation-error-when-running-examples/

Can confirm according to the second link that CLI commands like `ros2 topic list` throw the same error.

### Working Config OpenSSL

```
(humble) c:\Code\>mamba list | findstr openssl
openssl                   1.1.1q               h8ffe710_0    conda-forge
pyopenssl                 22.0.0             pyhd8ed1ab_0    conda-forge
```

### Broken Config OpenSSL

Tobias-Fischer, 2022-08-21:
Just to clarify - in our case the issue cannot be related to OpenSSL as both working and broken configs are the same?

danzimmerman, 2022-08-21:
@Tobias-Fischer Yes, looks that way, doesn't seem to be OpenSSL version issue at least.

danzimmerman, 2022-08-21:
Trying to downgrade `ros-humble-fastrtps` to match the working config's `2.6.1`, which will only change that package:

<details>

```
(humble) c:\code\>mamba install ros-humble-fastrtps=2.6.1 --dry-run

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (0.25.0) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['ros-humble-fastrtps=2.6.1']

robostack-experimental/win-64                                 No change
robostack-humble/noarch                                       No change
robostack-humble/win-64                                       No change
robostack/win-64                                              No change
robostack-experimental/noarch                                 No change
robostack/noarch                                              No change
conda-forge/noarch                                   9.0MB @   3.9MB/s  2.9s
conda-forge/win-64                                  15.9MB @   3.5MB/s  5.8s

Pinned packages:
  - python 3.9.*

Transaction

  Prefix: C:\Users\dan\mambaforge\envs\humble

  Updating specs:

   - ros-humble-fastrtps=2.6.1
   - ca-certificates
   - certifi
   - openssl

  Package                Version  Build           Channel                     Size
------------------------------------------------------------------------------------
  Downgrade:
------------------------------------------------------------------------------------

  - ros-humble-fastrtps    2.6.2  py39hf2f0fb7_1  robostack-humble
  + ros-humble-fastrtps    2.6.1  py39hf2f0fb7_0  robostack-humble/win-64      3MB

  Summary:

  Downgrade: 1 packages

  Total download: 3MB

------------------------------------------------------------------------------------

Dry run. Exiting.

DryRunExit: Dry run. Exiting.
```
</details>

However, no change:

```
(humble) C:\code\ros\humble_ws>ros2 topic list
[ERROR] [1661120351.902239800] [rcl]: 
Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), 
with error message 'failed to load shared library 'rmw_fastrtps_cpp.dll' due to LoadLibrary error: 126, 
at C:\bld\ros-humble-rcutils_1655024829144\work\ros-humble-rcutils\src\work\src\shared_library.c:159, 
at C:\bld\ros-humble-rmw-implementation_1655039337688\work\ros-humble-rmw-implementation\src\work\src\functions.cpp:67', 
exiting with 1., 
at C:\bld\ros-humble-rcl_1655040514413\work\ros-humble-rcl\src\work\src\rcl\rmw_implementation_identifier_check.c:145
```

```
(humble) C:\code\ros\humble_ws>mamba list | findstr fastrtps
ros-humble-fastrtps       2.6.1            py39hf2f0fb7_0    robostack-humble
ros-humble-fastrtps-cmake-module 2.2.0            py39he8739fe_1    robostack-humble
ros-humble-rmw-fastrtps-cpp 6.2.1            py39he8739fe_0    robostack-humble
ros-humble-rmw-fastrtps-dynamic-cpp 6.2.1            py39he8739fe_0    robostack-humble
ros-humble-rmw-fastrtps-shared-cpp 6.2.1            py39he8739fe_0    robostack-humble
ros-humble-rosidl-typesupport-fastrtps-c 2.2.0            py39he8739fe_0    robostack-humble
ros-humble-rosidl-typesupport-fastrtps-cpp 2.2.0            py39he8739fe_0    robostack-humble
```

traversaro, 2022-08-22:
@danzimmerman can you share the `mamba list` of the broken environment. I followed your instructions, but I am unable to reproduce your problem. My mamba list is:

<details>

~~~
(humble) C:\Users\STraversaro>mamba list
# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\humble:
#
# Name                    Version                   Build  Channel
anyio                     3.6.1              pyhd8ed1ab_1    conda-forge
aom                       3.4.0                h0e60522_1    conda-forge
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
argon2-cffi               21.3.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0           py39hb82d6ee_2    conda-forge
assimp                    5.2.4                hc2aa0de_0    conda-forge
asttokens                 2.0.8              pyhd8ed1ab_0    conda-forge
atomicwrites              1.4.1              pyhd8ed1ab_0    conda-forge
attrs                     22.1.0             pyh71513ae_1    conda-forge
babel                     2.10.3             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
beautifulsoup4            4.11.1             pyha770c72_0    conda-forge
bleach                    5.0.1              pyhd8ed1ab_0    conda-forge
boost                     1.74.0           py39hefe7e4c_5    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
brotli                    1.0.9                h8ffe710_7    conda-forge
brotli-bin                1.0.9                h8ffe710_7    conda-forge
brotlipy                  0.7.0           py39hb82d6ee_1004    conda-forge
bullet                    3.21                 h832f523_3    conda-forge
bullet-cpp                3.21                 h2e25243_3    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.5.0                h8ffe710_0    conda-forge
ca-certificates           2022.6.15            h5b45459_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.16.0            h0ac17fb_1012    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.6.15        py39hcbf5309_0    conda-forge
cffi                      1.15.1           py39h0878f49_0    conda-forge
charset-normalizer        2.1.0              pyhd8ed1ab_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.24.1               h39d44d4_0    conda-forge
colcon-cmake              0.2.26                     py_0    conda-forge
colcon-common-extensions  0.3.0            py39hcbf5309_0    conda-forge
colcon-core               0.10.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.5                      py_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-notification       0.2.13           py39hcbf5309_3    conda-forge
colcon-output             0.2.12                     py_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colorama                  0.4.5              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compilers                 1.5.0                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
coverage                  6.4.4            py39hb82d6ee_0    conda-forge
cppcheck                  2.7.5            py39hd48a9a8_0    conda-forge
cryptography              37.0.4           py39h7bc7c5c_0    conda-forge
curl                      7.83.1               h789b8ee_0    conda-forge
cxx-compiler              1.5.0                h2d74725_0    conda-forge
[... 558 lines cut ...]
win_inet_pton             1.1.0            py39hcbf5309_4    conda-forge
winpty                    0.4.3                         4    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
[... cut ...]
yaml                      0.2.5                h8ffe710_2    conda-forge
yaml-cpp                  0.7.0                h0e60522_1    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.8.1              pyhd8ed1ab_0    conda-forge
zlib                      1.2.12               h8ffe710_2    conda-forge
zstd                      1.5.2                h7755175_4    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~
</details>

danzimmerman, 2022-08-22:
@traversaro Thanks, I'll diff that against mine and see if I can isolate my problem.

danzimmerman, 2022-08-22:
@traversaro There aren't many differences 

Yours
Mine

```
```

traversaro, 2022-08-22:
My suspect is that one of us has something corrupting/interfering with the system/environment in some way.

danzimmerman, 2022-08-22:
It seems that a `LoadLibrary Error 126` can be a dependency error.

```
(humble) c:\Windows\System32>dumpbin /DEPENDENTS %CONDA_PREFIX%\Library\bin\rmw_fastrtps_cpp.dll
Microsoft (R) COFF/PE Dumper Version 14.29.30038.1
Copyright (C) Microsoft Corporation.  All rights reserved.

Dump of file C:\Users\<username>\mambaforge\envs\humble\Library\bin\rmw_fastrtps_cpp.dll

File Type: DLL

  Image has the following dependencies:

    rmw_fastrtps_shared_cpp.dll - 
    fastrtps-2.6.dll
    rosidl_typesupport_fastrtps_c.dll
    rosidl_typesupport_fastrtps_cpp.dll
    fastcdr-1.0.dll
    rmw_dds_common.dll
    rmw_dds_common__rosidl_typesupport_cpp.dll
    rmw.dll
    rosidl_runtime_c.dll
    rcutils.dll
    MSVCP140.dll
    VCRUNTIME140.dll
    VCRUNTIME140_1.dll - NOT in env, but in C:\Windows\System32 which is on the %PATH%
    api-ms-win-crt-runtime-l1-1-0.dll
    api-ms-win-crt-heap-l1-1-0.dll
    api-ms-win-crt-stdio-l1-1-0.dll
    api-ms-win-crt-string-l1-1-0.dll
    api-ms-win-crt-math-l1-1-0.dll
    KERNEL32.dll

  Summary

        A000 .data
        4000 .pdata
       15000 .rdata
        1000 .reloc
        1000 .rsrc
       3C000 .text
```

However, this doesn't seem to be an issue. All of the dependencies except `KERNEL32.dll` ship in the `Library/bin` folder.

```python
import os
import pathlib
from subprocess import check_output

dll_path = pathlib.Path(os.environ['CONDA_PREFIX']) / 'Library/bin'
DLL_NAME = 'rmw_fastrtps_cpp.dll'

print(f'??? Checking DLL Path {dll_path} for dependencies of {DLL_NAME}')

dumpbin_output_string = check_output(['DUMPBIN', '/DEPENDENTS', (dll_path / DLL_NAME).as_posix()]).decode('UTF-8')
deps_list = dumpbin_output_string.split('dependencies:')[1].split('Summary')[0].strip().split()

# make them all lowercase so the check isn't case dependent
env_dlls = [fn.lower() for fn in os.listdir(dll_path) if fn.count('.dll')]
not_in_env_dlls = [fn for fn in deps_list if not fn.lower() in env_dlls]
in_env_dlls = [fn for fn in deps_list if fn.lower() in env_dlls]

print(f'\n>>> Found {len(in_env_dlls)} of {len(deps_list)} dependencies in {dll_path}:\n{in_env_dlls}')
print(f'\n>>> Did not find these DLLs in {dll_path}:\n{not_in_env_dlls}')
```

Gives

```
??? Checking DLL Path C:\Users\<username>\mambaforge\envs\humble\Library\bin for dependencies of rmw_fastrtps_cpp.dll

>>> Found 18 of 19 dependencies in C:\Users\dan\<username>\envs\humble\Library\bin:
['rmw_fastrtps_shared_cpp.dll', 'fastrtps-2.6.dll', 'rosidl_typesupport_fastrtps_c.dll', 'rosidl_typesupport_fastrtps_cpp.dll', 'fastcdr-1.0.dll', 'rmw_dds_common.dll', 'rmw_dds_common__rosidl_typesupport_cpp.dll', 'rmw.dll', 'rosidl_runtime_c.dll', 'rcutils.dll', 'MSVCP140.dll', 'VCRUNTIME140.dll', 'VCRUNTIME140_1.dll', 'api-ms-win-crt-runtime-l1-1-0.dll', 'api-ms-win-crt-heap-l1-1-0.dll', 'api-ms-win-crt-stdio-l1-1-0.dll', 'api-ms-win-crt-string-l1-1-0.dll', 'api-ms-win-crt-math-l1-1-0.dll']

>>> Did not find these DLLs in C:\Users\<username>\mambaforge\envs\humble\Library\bin:
['KERNEL32.dll']
```

danzimmerman, 2022-08-22:
This documentation points at the need for VC2013 Redistributables to supply `msvcr20.dll` for FastRTPS. 

https://docs.ros.org/en/humble/How-To-Guides/Installation-Troubleshooting.html#failed-to-load-fast-rtps-shared-library

via 

https://github.com/ros2/ros2/issues/708

I tried installing that but no luck. Will try to restart and check again.

danzimmerman, 2022-08-22:
Looks like I'm barking up the wrong tree with the VC 2013 stuff. 

I checked the dependencies of `fastrtps-2.6.dll` and found that it's looking for `foonathan_memory-0.7.1.dll`. 

Output of `dumpbin /dependents fastrtps-2.6.dll`:

<details>

```(humble) C:\Users\<username>\mambaforge\envs\humble\Library\bin>dumpbin /dependents fastrtps-2.6.dll
Microsoft (R) COFF/PE Dumper Version 14.29.30038.1
Copyright (C) Microsoft Corporation.  All rights reserved.

Dump of file fastrtps-2.6.dll

File Type: DLL

  Image has the following dependencies:

    fastcdr-1.0.dll
    foonathan_memory-0.7.1.dll
    tinyxml2.dll
    libssl-1_1-x64.dll
    libcrypto-1_1-x64.dll
    IPHLPAPI.DLL
    SHLWAPI.dll
    KERNEL32.dll
    ADVAPI32.dll
    MSVCP140.dll
    WS2_32.dll
    MSWSOCK.dll
    VCRUNTIME140.dll
    VCRUNTIME140_1.dll
    api-ms-win-crt-runtime-l1-1-0.dll
    api-ms-win-crt-heap-l1-1-0.dll
    api-ms-win-crt-convert-l1-1-0.dll
    api-ms-win-crt-stdio-l1-1-0.dll
    api-ms-win-crt-string-l1-1-0.dll
    api-ms-win-crt-math-l1-1-0.dll
    api-ms-win-crt-time-l1-1-0.dll
    api-ms-win-crt-filesystem-l1-1-0.dll
    api-ms-win-crt-environment-l1-1-0.dll
    api-ms-win-crt-locale-l1-1-0.dll

  Summary

       73000 .data
       3C000 .pdata
      11C000 .rdata
        A000 .reloc
        1000 .rsrc
      4E5000 .text
```

</details>

However, I have `foonathan-memory-0.7.2` installed instead:

```
(humble) C:\Users\<username>\mambaforge\envs\humble\Library\bin>mamba list | findstr foonathan
foonathan-memory          0.7.2                h57928b3_0    conda-forge
ros-humble-foonathan-memory-vendor 1.2.0            py39he8739fe_1    robostack-humble
```

Downgrading with `mamba install foonathan-memory=0.7.1` doesn't change any other packages and seems to fix the problem:

```
(humble) c:\code>mamba list | findstr foonathan
foonathan-memory          0.7.1                h57928b3_0    conda-forge
ros-humble-foonathan-memory-vendor 1.2.0            py39he8739fe_1    robostack-humble

(humble) c:\code>ros2 topic list
/parameter_events
/rosout
```

@traversaro Do you have the `RMW_IMPLEMENTATION` env variable set? I have:

```
(humble) c:\code>set | findstr RMW
RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```

traversaro, 2022-08-22:
Great debug! The issue then is that the run_exports for foonathan-memory should be stricter: https://github.com/conda-forge/foonathan-memory-feedstock/blob/47bf93075f75cc4645b5c77f9a26bb9b84697242/recipe/meta.yaml#L21 . This is because the library has a form of manual soversion by changing the library name: https://github.com/foonathan/memory/blob/f75c9fc761fde7fa0f6a829df08ea0041ebf1f10/src/CMakeLists.txt#L109 .

danzimmerman, 2022-08-22:
Can confirm that the `foonathan-memory` downgrade allows me to load my rosbag in my Jupyter notebook.

So ultimately, I think there was an uninformative error on a chain of DLL loads here, and lots of red herrings.

I've unset `RMW_IMPLEMENTATION` now (with `foonathan-memory 0.7.1` and the notebook works fine. 

If I upgrade `foonathan-memory` back to `0.7.2` commands like `ros2 topic list` work fine without `RMW_IMPLEMENTATION` env variable set.

However, actually reading my rosbag crashes. 😓

traversaro, 2022-08-22:
https://github.com/conda-forge/foonathan-memory-feedstock/pull/7 should fix the issue for future builds, but for past build we should pepare a repodata patch.

traversaro, 2022-08-23:
> but for past build we should pepare a repodata patch and submit a PR to https://github.com/conda-forge/conda-forge-repodata-patches-feedstock, something like [conda-forge/conda-forge-repodata-patches-feedstock#286](https://github.com/conda-forge/conda-forge-repodata-patches-feedstock/pull/286) .

Actually, this is unfortunatly not the case. In the `robostack-humble` channel, we are not using the [`fastrtps` conda-forge package](https://anaconda.org/conda-forge/fast-rtps/), but a custom [`ros-humble-fastrtps`](https://anaconda.org/robostack-humble/ros-humble-fastrtps/files) (an instance of the more general problem discussed in https://github.com/RoboStack/ros-noetic/issues/20 .

As as far as I know we do not have any repodata patch infrastructure on robostack, the proper way of solving the problem is to rebuild the `ros-humble-fastrtps` package.

Tobias-Fischer, 2022-08-25:
I triggered a rebuild of `ros-humble-fastrtps` on Windows.

danzimmerman, 2022-08-31:
I tried to `update --all` tonight, but `ros-humble-fastrtps` isn't one that got updated. 

Tonight's revision:

<details>

```
2022-08-31 19:47:09  (rev 12)
     cairo  {1.16.0 (conda-forge/win-64) -> 1.16.0 (conda-forge/win-64)}
     certifi  {2022.6.15 (conda-forge/win-64) -> 2022.6.15 (conda-forge/noarch)}
     charset-normalizer  {2.1.0 (conda-forge/noarch) -> 2.1.1 (conda-forge/noarch)}
     eigenpy  {2.7.11 (conda-forge/win-64) -> 2.7.12 (conda-forge/win-64)}
     executing  {0.10.0 (conda-forge/noarch) -> 1.0.0 (conda-forge/noarch)}
     fonttools  {4.36.0 (conda-forge/win-64) -> 4.37.1 (conda-forge/win-64)}
     foonathan-memory  {0.7.1 (conda-forge/win-64) -> 0.7.2 (conda-forge/win-64)}
     graphviz  {5.0.0 (conda-forge/win-64) -> 5.0.1 (conda-forge/win-64)}
     ipykernel  {6.15.1 (conda-forge/noarch) -> 6.15.2 (conda-forge/noarch)}
     ipython  {8.4.0 (conda-forge/win-64) -> 8.4.0 (conda-forge/noarch)}
     jsonschema  {4.9.1 (conda-forge/noarch) -> 4.15.0 (conda-forge/noarch)}
     jupyter_client  {7.3.4 (conda-forge/noarch) -> 7.3.5 (conda-forge/noarch)}
     jupyterlab_server  {2.15.0 (conda-forge/noarch) -> 2.15.1 (conda-forge/noarch)}
     libignition-cmake2  {2.14.0 (conda-forge/win-64) -> 2.15.0 (conda-forge/win-64)}
     llvm-meta  {5.0.0 (conda-forge/noarch) -> 13.0.1 (conda-forge/noarch)}
     matplotlib  {3.5.3 (conda-forge/win-64) -> 3.5.3 (conda-forge/win-64)}
     matplotlib-base  {3.5.3 (conda-forge/win-64) -> 3.5.3 (conda-forge/win-64)}
     mistune  {0.8.4 (conda-forge/win-64) -> 2.0.4 (conda-forge/noarch)}
     nbclient  {0.6.6 (conda-forge/noarch) -> 0.6.7 (conda-forge/noarch)}
     nbconvert  {6.5.3 (conda-forge/noarch) -> 7.0.0 (conda-forge/noarch)}
     nbconvert-core  {6.5.3 (conda-forge/noarch) -> 7.0.0 (conda-forge/noarch)}
     nbconvert-pandoc  {6.5.3 (conda-forge/noarch) -> 7.0.0 (conda-forge/noarch)}
     openmp  {5.0.0 (conda-forge/win-64) -> 5.0.0 (conda-forge/win-64)}
     pandoc  {2.19.1 (conda-forge/win-64) -> 2.19.2 (conda-forge/win-64)}
     pinocchio  {2.6.8 (conda-forge/win-64) -> 2.6.9 (conda-forge/win-64)}
     pysocks  {1.7.1 (conda-forge/win-64) -> 1.7.1 (conda-forge/noarch)}
     qtconsole  {5.3.1 (conda-forge/noarch) -> 5.3.2 (conda-forge/noarch)}
     qtconsole-base  {5.3.1 (conda-forge/noarch) -> 5.3.2 (conda-forge/noarch)}
     requests  {2.28.1 (conda-forge/noarch) -> 2.28.1 (conda-forge/noarch)}
     scipy  {1.9.0 (conda-forge/win-64) -> 1.9.1 (conda-forge/win-64)}
     setuptools  {65.1.0 (conda-forge/win-64) -> 65.3.0 (conda-forge/noarch)}
     stack_data  {0.4.0 (conda-forge/noarch) -> 0.5.0 (conda-forge/noarch)}
     vs2015_runtime  {14.29.30037 (conda-forge/win-64) -> 14.29.30139 (conda-forge/win-64)}
     vtk  {9.1.0 (conda-forge/win-64) -> 9.1.0 (conda-forge/win-64)}
     websocket-client  {1.3.3 (conda-forge/noarch) -> 1.4.0 (conda-forge/noarch)}
```
</details>

I'm on 

```
ros-humble-fastrtps       2.6.2            py39hf2f0fb7_1    robostack-humble
```

Will check again later. Is there anything else I need to do to try testing an update?

Tobias-Fischer, 2022-08-31:
I was sneaky - I overwrote `ros-humble-fastrtps`. So you will need to uninstall and re-install unfortunately ..

danzimmerman, 2022-08-31:
Ok will try that 👍🏻

danzimmerman, 2022-09-01:
Uninstalled and reinstalled `ros-humble-fastrtps` and those things that depended on it. (Also ran `mamba clean --all` to make sure I didn't reinstall cached packages) and now it's working fine with `foonathan-memory 0.7.2` 👍

danzimmerman, 2022-09-15:
@traversaro @Tobias-Fischer This actually seems to be an issue on `linux-64` as well if I `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp` explicitly in my env variables:

```
(humble) dan@<computer>:~$ ros2 topic list
[ERROR] [1663265938.869761742] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), with error message 'failed to load shared library 'librmw_fastrtps_cpp.so' due to dlopen error: libfoonathan_memory-0.7.1.so: cannot open shared object file: No such file or directory, at /opt/conda/build_artifacts/ros-humble-rcutils_1656763445540/work/ros-humble-rcutils/src/work/src/shared_library.c:99, at /opt/conda/build_artifacts/ros-humble-rmw-implementation_1659331744961/work/ros-humble-rmw-implementation/src/work/src/functions.cpp:65', exiting with 1., at /opt/conda/build_artifacts/ros-humble-rcl_1659332063789/work/ros-humble-rcl/src/work/src/rcl/rmw_implementation_identifier_check.c:139
```
The error is a lot more informative on Linux.

By default Linux was using `rmw_cyclonedds_cpp`, which I don't THINK is supposed to be the default:

https://www.eprosima.com/index.php/company-all/news/270-ros2-humble-comes-with-fast-dds

Tobias-Fischer, 2022-12-12:
This is now working fine with the fresh builds, at least on Linux. However, see #30

Tobias-Fischer, 2022-12-13:
Hi @danzimmerman, could you please let us know whether this is still an issue with the new builds in the robostack-staging channel?

danzimmerman, 2022-12-13:
@Tobias-Fischer Sure, happy to. I just tried it with these channels:

```
(humble-staging) C:\code\ros>conda config --show channels
channels:
  - robostack-staging
  - conda-forge
```

and 

```
mamba install ros-humble-desktop
```

But I can't find the desktop package. Do I need to wait a while for it to propagate?

```
Encountered problems while solving:
  - nothing provides requested ros-humble-desktop
```

Tobias-Fischer, 2022-12-14:
Sorry, still working on ros-humble-desktop on Windows. However, https://anaconda.org/robostack-staging/ros-humble-rosbag2-py is available which should be enough to test this particular issue?

danzimmerman, 2022-12-14:
Seems to work fine here to import `rosbag2_py`. Also was able to install `ros-humble-ros-core` and verify that topic listing was working.

Didn't get a chance to actually open a bag file yet, which seemed to also give me some trouble. I'll try some other things later when I get a chance, but the basics seem to be working.

Tobias-Fischer, 2022-12-14:
Ok - let’s close here then, if you have rosbag troubles please open a separate issue.

closed 2022-12-14 by Tobias-Fischer