RoboStack/ros-humble #10: Avoid ros-humble-octomap clobbering octomap

Tobias-Fischer, 2022-08-24:


Tobias-Fischer, 2022-08-25:
How about doing sth like this: https://github.com/RoboStack/ros-galactic/blob/main/additional_recipes/ros-galactic-cartographer/recipe.yaml @wolfv?

traversaro, 2022-08-25:
Those repos shadow the upstream ones? In that case, that would be great. Cross-link to https://github.com/RoboStack/ros-noetic/issues/20 .

Tobias-Fischer, 2022-08-25:
Yeah - the issue is that they need manual updating and rebuilding when new versions come out.

traversaro, 2022-11-07:
This is kind of related to https://github.com/ros/rosdistro/issues/26527, even if that issue was just for Noetic.

Tobias-Fischer, 2022-11-08:
Fixed with #21 - not sure which other packages are vendored in ROS2

traversaro, 2022-11-18:
> Fixed with #21 - not sure which other packages are vendored in ROS2

Many of them, some of them are tracked in https://github.com/RoboStack/ros-noetic/issues/20, but probably we could keep a ROS2-specific issue. Anyhow, I am not sure if we want to fix all of them now, it may be non-trivial to do so.

Tobias-Fischer, 2022-12-11:
Fixed with the new builds :)

closed 2022-12-11 by Tobias-Fischer