RoboStack/ros-humble #14: `ros2controlcli` is broken

danzimmerman, 2022-09-01:
I'm having trouble with the ROS2 Control CLI on a fresh Linux install:

```python
(humble) <user>@<computer>:~$ ros2 control list_controllers
Failed to load entry point 'list_controllers': cannot import name 'ConfigureStartController' from 'controller_manager_msgs.srv' (/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/controller_manager_msgs/srv/__init__.py)
Traceback (most recent call last):
  File "/home/<user>/.local/opt/mambaforge/envs/humble/bin/ros2", line 10, in <module>
    sys.exit(main())
  File "/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/ros2cli/cli.py", line 50, in main
    add_subparsers_on_demand(
  File "/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/ros2cli/command/__init__.py", line 250, in add_subparsers_on_demand
    extension.add_arguments(
  File "/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/ros2controlcli/command/control.py", line 26, in add_arguments
    add_subparsers_on_demand(parser, cli_name, '_verb', 'ros2controlcli.verb', required=False)
  File "/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/ros2cli/command/__init__.py", line 237, in add_subparsers_on_demand
    extension = command_extensions[name]
KeyError: 'list_controllers'
```

I think this is related to https://github.com/ros-controls/ros2_control/blob/master/controller_manager_msgs/CHANGELOG.rst#2120-2022-07-09 

It seems I've got an installed mix of packages from the `2.10` and `2.13` releases of `ros2_control`

Looking through the packages provided by https://github.com/ros-controls/ros2_control and the versions that get installed in my environment, I see the following:

```
ros-humble-controller-interface 2.13.0           py39h63710d1_1    robostack-humble
ros-humble-controller-manager 2.13.0           py39h63710d1_1    robostack-humble
ros-humble-controller-manager-msgs 2.13.0           py39h6be7e7a_1    robostack-humble
ros-humble-hardware-interface 2.13.0           py39h63710d1_1    robostack-humble
ros-humble-joint-limits                        <not available or installed>
ros-humble-joint-limits-interface              <not available or installed>
ros-humble-ros2-control   2.10.0           py39h6fdeb60_0    robostack-humble
ros-humble-ros2-control-test-assets 2.13.0           py39h63710d1_1    robostack-humble
ros-humble-ros2controlcli 2.10.0           py39h6fdeb60_0    robostack-humble
ros-humble-transmission-interface 2.10.0           py39h6fdeb60_0    robostack-humble
```

If I try to install v `2.13.0` of any of the packages that are on `2.10.0` I get, for example

```
$ mamba install ros-humble-ros2controlcli=2.13.0
Encountered problems while solving:
  - nothing provides requested ros-humble-ros2controlcli 2.13.0**
```

danzimmerman, 2022-09-01:
Is this tangentially related to https://github.com/ros-controls/ros2_control/issues/803 ?

Tobias-Fischer, 2022-09-01:
It shouldn't be related, no - https://github.com/ros-controls/ros2_control/issues/803 specifically talks about osx/win, whereas your issue seems to be about Linux.

I'm not sure why some packages have been built for version 2.13.0 while others haven't - weird!

Tobias-Fischer, 2022-09-01:
Actually the reason is that in recent builds we haven't included some of these packages - should be fixed soon: https://github.com/RoboStack/ros-humble/blob/main/vinca_linux_64.yaml

Tobias-Fischer, 2022-09-01:
Packages are now rebuilt - let me know if it still doesn't work.

danzimmerman, 2022-09-02:
Yep, updated and it works now. Thanks!

closed 2022-09-02 by danzimmerman