RoboStack/ros-humble #15: `rqt` runtime issues.

danzimmerman, 2022-09-01:
If I try to run `rqt_gui` or `rqt_graph` or probably other similar packages on a fresh Linux install, I get

```python
(humble) <user>@<computer>:~/ros/humble_ws$ ros2 run rqt_gui rqt_gui
Could not import "pyqt" bindings of qt_gui_cpp library - so C++ plugins will not be available:
Traceback (most recent call last):
  File "/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ImportError: /home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/qt_gui_cpp/libqt_gui_cpp_sip.cpython-39-x86_64-linux-gnu.so: undefined symbol: _ZTIN10qt_gui_cpp14PluginProviderE
```

`(humble)` environment details:

<details>

```
```

</details>

Tobias-Fischer, 2022-09-01:
Could you try `mamba install pyqt=5.15.4` and try again?

danzimmerman, 2022-09-02:
`mamba install ros-humble-qt-gui=2.2.1 ros-humble-qt-gui-cpp=2.2.1 ros-humble-qt-gui-py-common=2.2.1`

No, that doesn't seem to solve the problem either.

Tobias-Fischer, 2022-12-12:
This problem unfortunately still persists with the newly built packages. I think something is wrong with this hacky code: 

https://github.com/RoboStack/ros-humble/blob/2f2e0e25a8e92db73626009f7321c8784341780e/patch/ros-humble-qt-gui-cpp.patch#L23-L24

See also
https://github.com/RoboStack/ros-humble/blob/main/patch/ros-humble-python-qt-binding.patch

That said, rqt_gui etc seems to run and behave okay as far as I can tell.

Tobias-Fischer, 2022-12-13:
Alright - this was tricker than first expected, but is resolved now. The issue was that we did not link against qt_gui_cpp.so within the qt_gui_cpp_sip_*.so. However, the build succeeded for some reason. All good with the latest patches in python-qt-binding and qt-gui-cpp.

Tobias-Fischer, 2022-12-13:
See #30 ..

closed 2022-12-13 by Tobias-Fischer