RoboStack/ros-humble #17: Adding vision-msgs package for osx arm64

EdanToledo, 2022-10-09:
Hello guys, great work here- I have been struggling to get ros 2 working natively on an m1 MacBook and this has almost sorted me out. I'm not sure on the difficulty or process of creating these packages for arm64 but for Webots and ros2 to interface the vision-msgs package is important. There are probably other important packages but I can't get passed the building stage due to this missing package. Do you have any idea of when this package might possibly be added?

Thanks again.

Tobias-Fischer, 2022-10-09:
Hi @EdanToledo - it's very easy to add packages that you would like via a pull request. All that is needed is adding one line to the vinca_osx_arm64.yaml file, see e.g. https://github.com/RoboStack/ros-humble/pull/7

Feel free to open a pull request and we will build the packages that you would like.

EdanToledo, 2022-10-10:
Hello again - thanks for the help - what an easy way to add packages! My next issue is that I am trying to add all the packages required for ros2_webots. I saw that on the linux version it says there are issues with it. Do you think I can add all the source ros2 packages and then build from ros2_webots source files?

Tobias-Fischer, 2022-10-11:
Hiya - webots-ros2 is available for linux-64: https://anaconda.org/robostack-humble/ros-humble-webots-ros2/files

So give it a shot and add it to the vinca_osx*.yaml in a pull request and we will see if it works :).

Tobi

Tobias-Fischer, 2022-10-11:
Closing here - see #19 instead.

closed 2022-10-11 by Tobias-Fischer