RoboStack/ros-humble #23: Undefined symbol running rviz2

Nicogene, 2022-10-24:
I installed `ros2` via this command:
```
mamba create -n ros2_env -c conda-forge -c robostack -c robostack-experimental -c robostack-humble yarp  ros-humble-desktop
```

When I try to run `rviz2` I get:

```
rviz2: symbol lookup error: /home/ngenesio/mambaforge/envs/ros2_env/bin/../lib/././librcl_logging_spdlog.so: undefined symbol: _ZN6spdlog7details7log_msgC1ENS_10source_locEN3fmt2v817basic_string_viewIcEENS_5level10level_enumES6_
```

Same error when I try to run `rqt`.
I am using WSL2, Ubuntu 22.04.1, this is my environment:
```
# packages in environment at /home/ngenesio/mambaforge/envs/ros2_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       7.0.8                h27087fc_0    conda-forge
alsa-lib                  1.2.7.2              h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.2.4                h4c92fa2_0    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     22.1.0             pyh71513ae_1    conda-forge
boost                     1.74.0           py39h5472131_5    conda-forge
boost-cpp                 1.74.0               h75c5d50_8    conda-forge
brotli                    1.0.9                h166bdaf_7    conda-forge
brotli-bin                1.0.9                h166bdaf_7    conda-forge
bullet                    3.21                 hef51801_3    conda-forge
bullet-cpp                3.21                 h1832856_3    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.9.24            ha878542_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.9.24          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_1    conda-forge
cmake                     3.24.2               h5432695_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.0.5            py39hf939315_0    conda-forge
cppcheck                  2.7.5            py39h94487b0_0    conda-forge
cryptography              38.0.2           py39hd97740a_1    conda-forge
curl                      7.85.0               h7bff187_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39hf3d152e_0    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.9                h27087fc_0    conda-forge
ffmpeg                    4.4.2           gpl_hc51e5dc_110    conda-forge
fftw                      3.3.10          nompi_hf0379b8_105    conda-forge
flake8                    5.0.4              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             hfe772e8_1010    conda-forge
fmt                       9.1.0                h924138e_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.1               hc2a2eb6_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0           py39hb9d737c_0    conda-forge
foonathan-memory          0.7.2                h27087fc_1    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freeimage                 3.18.0              h530e30e_10    conda-forge
[... 508 lines cut ...]
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.13               h7f98852_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.7.0                h27087fc_2    conda-forge
yarp                      3.7.2                ha770c72_0    conda-forge
yarp-cxx                  3.7.2                hfee3fe3_0    conda-forge
yarp-python               3.7.2            py39h5a03fae_0    conda-forge
ycm-cmake-modules         0.14.2               h27087fc_0    conda-forge
zipp                      3.9.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h6239696_4    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
```

Nicogene, 2022-10-24:
Maybe also the `ldd` of the library can be useful:

```
ldd /home/ngenesio/mambaforge/envs/ros2_env/bin/../lib/././librcl_logging_spdlog.so
        linux-vdso.so.1 (0x00007fff2c935000)
        librcpputils.so => /home/ngenesio/mambaforge/envs/ros2_env/bin/../lib/./././librcpputils.so (0x00007fab6e8b3000)
        libspdlog.so.1 => /home/ngenesio/mambaforge/envs/ros2_env/bin/../lib/./././libspdlog.so.1 (0x00007fab6e839000)
        librcl_logging_interface.so => /home/ngenesio/mambaforge/envs/ros2_env/bin/../lib/./././librcl_logging_interface.so (0x00007fab6e834000)
        librcutils.so => /home/ngenesio/mambaforge/envs/ros2_env/bin/../lib/./././librcutils.so (0x00007fab6e81c000)
        libstdc++.so.6 => /home/ngenesio/mambaforge/envs/ros2_env/bin/../lib/./././libstdc++.so.6 (0x00007fab6e666000)
        libgcc_s.so.1 => /home/ngenesio/mambaforge/envs/ros2_env/bin/../lib/./././libgcc_s.so.1 (0x00007fab6e64d000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fab6e408000)
        libfmt.so.9 => /home/ngenesio/mambaforge/envs/ros2_env/bin/../lib/././././libfmt.so.9 (0x00007fab6e3e3000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fab6e3de000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fab6e8cf000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fab6e3d7000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fab6e3d2000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fab6e2eb000)
```

Nicogene, 2022-10-24:
Interesting that I have two `librcl_logging_spdlog.so`
```
locate librcl_logging_spdlog.so
/home/ngenesio/mambaforge/envs/ros2_env/lib/librcl_logging_spdlog.so
/home/ngenesio/mambaforge/pkgs/ros-humble-rcl-logging-spdlog-2.3.0-py39h495f368_1/lib/librcl_logging_spdlog.so
```

But they seem the same

Tobias-Fischer, 2022-10-24:
There is a bug unfortunately with the version of spdlog that is installed. Could you please try the previous version?

Nicogene, 2022-10-24:
> There is a bug unfortunately with the version of spdlog that is installed. Could you please try the previous version?

This workaround worked fine thanks! I installed `1.9.2` instead of the `1.10.0`
```
mamba create -n ros2_env -c conda-forge -c robostack -c robostack-experimental -c robostack-humble yarp spdlog=1.9.2  ros-humble-desktop
```

I still have an issue opening gazebo, but maybe I can open a separate issue for that.

wolfv, 2022-10-25:
Thanks for providing this workaround @Nicogene! 

It's expected that you have the library in `envs` and `pkgs`: the `pkgs` directory is a central location where packages are downloaded and extracted. From there they are linked into the environment. If you would have two or three environments that use the same package they would all refer to the same 'physical' file (via a hard-link).

@Tobias-Fischer I guess once we've started the rebuild to the staging channel, this will be fixed, right?

Tobias-Fischer, 2022-10-25:
Yes. Shall we put this in the install instructions before the talk tomorrow?

Tobias-Fischer, 2022-10-25:
Same with #6

wolfv, 2022-10-25:
yes, sounds like a good idea

traversaro, 2022-10-25:
I suspect this is due to the ABI incompatibility introduced by https://github.com/conda-forge/spdlog-feedstock/pull/50 .

traversaro, 2022-10-25:
Probably another workaround is to install the `1.10.0 _0` build instead of the `1.10.0 _1` build.

traversaro, 2022-10-25:
Hopefully this will fixed once https://github.com/conda-forge/admin-requests/pull/501 is merged.

Tobias-Fischer, 2022-12-11:
This is fixed in the newly built packages - however, please be aware of #30.

closed 2022-12-11 by Tobias-Fischer