RoboStack/ros-humble #24: Unable to run gazebo installed using ros-humble conda package

Nicogene, 2022-10-25:
I installed ros humble with gazebo with the following command:

```
mamba create -n ros2_env -c conda-forge -c robostack -c robostack-experimental -c robostack-humble yarp spdlog=1.9.2  ros-humble-desktop ros-humble-gazebo-ros-pkgs ros-humble-gazebo-dev
```

Unfortunately `gzclient` does not start, I get only this warning:

```
[Msg] Loading world file [/home/ngenesio/mambaforge/envs/ros2_env/share/gazebo-11/worlds/empty.world]
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
```

Outside the conda environment, it works fine.
Also if I run `gzserver` with the conda environment activated and `gzclient` in a terminal without the conda environment it starts (also if it complaints about gazebo versions mismatch)

I am using WSL2, Ubuntu 22.04.1, and here is my `mamba list` output:
```
# packages in environment at /home/ngenesio/mambaforge/envs/ros2_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       7.0.8                h27087fc_0    conda-forge
alsa-lib                  1.2.7.2              h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.2.4                h4c92fa2_0    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     22.1.0             pyh71513ae_1    conda-forge
blosc                     1.21.1               h83bc5f7_3    conda-forge
boost                     1.74.0           py39h5472131_5    conda-forge
boost-cpp                 1.74.0               h75c5d50_8    conda-forge
brotli                    1.0.9                h166bdaf_7    conda-forge
brotli-bin                1.0.9                h166bdaf_7    conda-forge
bullet                    3.21                 hef51801_3    conda-forge
bullet-cpp                3.21                 h1832856_3    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.9.24            ha878542_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.9.24          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_1    conda-forge
cfitsio                   4.1.0                hd9d235c_0    conda-forge
cmake                     3.24.2               h5432695_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.0.5            py39hf939315_0    conda-forge
cppcheck                  2.7.5            py39h94487b0_0    conda-forge
cppzmq                    4.8.1                hf7cf922_0    conda-forge
cryptography              38.0.2           py39hd97740a_1    conda-forge
curl                      7.85.0               h7bff187_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.12.2               hfeb25b1_1    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39hf3d152e_0    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.9                h27087fc_0    conda-forge
fcl                       0.7.0                h156637f_1    conda-forge
ffmpeg                    4.4.2           gpl_hc51e5dc_110    conda-forge
fftw                      3.3.10          nompi_hf0379b8_105    conda-forge
flake8                    5.0.4              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             hfe772e8_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
[... 561 lines cut ...]
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.13               h7f98852_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.7.0                h27087fc_2    conda-forge
yarp                      3.7.2                ha770c72_0    conda-forge
yarp-cxx                  3.7.2                hfee3fe3_0    conda-forge
yarp-python               3.7.2            py39h5a03fae_0    conda-forge
ycm-cmake-modules         0.14.2               h27087fc_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.9.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h6239696_4    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
```

Tobias-Fischer, 2022-10-25:
/cc @traversaro

Tobias-Fischer, 2022-10-25:
Does this also happen in a “gazebo-only” environment?

Nicogene, 2022-10-25:
> Does this also happen in a “gazebo-only” environment?

I created and environment `gazebo_only` such as:
```
mamba create -n gazebo_only -c conda-forge gazebo
```

And `gazebo` works smoothly

Tobias-Fischer, 2022-10-25:
Thanks - this problem will be fixed by #21 which we will hopefully merge in soon (too scared to do it during IROS though).

Tobias-Fischer, 2022-12-13:
Unfortunately even after the rebuild, this is still blocked by https://github.com/conda-forge/gazebo-feedstock/issues/88

Tobias-Fischer, 2023-02-09:
Fixed in the latest rebuild - this now pulls in gazebo built with ogre 1.10 which works nicely :)

/cc @b1n-ch1kn

closed 2023-02-09 by Tobias-Fischer

Nicogene, 2023-02-09:
> Fixed in the latest rebuild - this now pulls in gazebo built with ogre 1.10 which works nicely :)
> 
> /cc @b1n-ch1kn

Thanks heap @Tobias-Fischer!