RoboStack/ros-humble #30: Copy over packages from robostack-staging to robostack-humble

Tobias-Fischer, 2022-12-11:
Hiya @wolfv, the linux-64 and osx-64 rebuilts are complete :). We still need to copy over the packages to the robostack-humble channel; currently it will be confusing for users as the install instructions mention the robostack-humble channel + python3.10, but this results in an error ..

traversaro, 2022-12-16:
Probably we could modify the README to mention robostack-staging until this is done?

Tobias-Fischer, 2022-12-16:
I’m even thinking of not copying the packages. It’s good to have a fresh start and avoids annoying errors due to the different mutex versions and which ros packages are available ..

Tobias-Fischer, 2023-05-24:
I think we are quite happy with the current solution to use robostack-staging as our new channel.

closed 2023-05-24 by Tobias-Fischer