RoboStack/ros-humble #31: nav2_smac_planner not finding ode_double.lib

Tobias-Fischer, 2022-12-14:
Hiya @traversaro,

The last system to rebuild packages for is Windows. Unfortunately I am stuck with the navigation2 packages. In particular, for nav2_smac_planner, I get:
```
LINK : fatal error LNK1181: cannot open input file 'ode_double.lib' [%SRC_DIR%\build\nav2_smac_planner.vcxproj]
```

I already tried adding `find_package(ode)` to the CMakeLists.txt, but it did not help. Oddly enough, the logs did not mention that it looked for ode?!? See this patch: https://github.com/RoboStack/ros-humble/blob/main/patch/ros-humble-nav2-smac-planner.win.patch

Do you have any ideas what else needs to be done?

traversaro, 2022-12-14:
For reference, this is the failing job: https://github.com/RoboStack/ros-humble/actions/runs/3689749512/jobs/6246043717 .

traversaro, 2022-12-14:
There is something strange in the `omplConfig.cmake`: 

~~~
foreach(_lib D:/bld/ompl_1667488141743/_h_env/Library/lib/boost_serialization.lib;D:/bld/ompl_1667488141743/_h_env/Library/lib/boost_filesystem.lib;D:/bld/ompl_1667488141743/_h_env/Library/lib/boost_system.lib;ode_double;)
    if(_lib)
        list(APPEND OMPL_LIBRARIES "${_lib}")
    endif()
endforeach()
~~~

Apparently just doing `target_link_libraries([..] ode_double)` does not work, even if `ode_double.lib` is in a known location. Probably a quick fix that we could attemp is just to replace `ode_double` with `ODE::ODE`, that should work fine, i.e. to add after ` find_package(ompl REQUIRED)` and `find_package(ode REQUIRED)`:

~~~cmake
string(REPLACE "ode_double" "ODE::ODE" OMPL_LIBRARIES ${OMPL_LIBRARIES})
~~~

traversaro, 2022-12-14:
> Oddly enough, the logs did not mention that it looked for ode?!?

This is indeed strange, even if I am not really sure who is printing `Found: <pkg>` .

traversaro, 2022-12-14:
> > Oddly enough, the logs did not mention that it looked for ode?!?
> 
> This is indeed strange, even if I am not really sure who is printing `Found: <pkg>` .

Probably we can add a `message(STATUS "=====================> MESSAGE")` to the patch to make sure that it is actually applied and we are not missing anything on that side?

Tobias-Fischer, 2022-12-14:
Thanks Silvio - let’s see what happens :)

traversaro, 2022-12-14:
Ok, I think only packages that export a version (via a `<pkg>Version.cmake` file) print a message when found.

Tobias-Fischer, 2022-12-14:
By the way, when trying to build move it another weird error occurs.. https://github.com/RoboStack/ros-humble/actions/runs/3693026840

traversaro, 2022-12-14:
> By the way, when trying to build move it another weird error occurs.. https://github.com/RoboStack/ros-humble/actions/runs/3693026840

A possible patch (untested) could be https://github.com/ros-planning/moveit2/commit/dd3cd3290fae5b8a60f4052f053e7ff75ea304aa .

Tobias-Fischer, 2022-12-14:
Both the moveit as well as the navigation2 issues have been resolved. Thanks so much @traversaro!!!