RoboStack/ros-humble #32: Topic related operation cause crash

mrkbac, 2022-12-14:
Hey, when doing anything related to topics, like `ros2 topic list`, starting `rviz2` or playing a bag the application crashes. My platform is `osx-arm64`.

```
(ros2) m@mbp ~ % ros2 topic list
[foonathan::memory] Allocator foonathan::memory::memory_pool (at 0x6000018f8050) received invalid size/alignment 56, max supported is 32
libc++abi: terminating with uncaught exception of type foonathan::memory::bad_node_size: allocation node size exceeds supported maximum of allocator
zsh: abort      ros2 topic list
(ros2) m@mbp ~ % rviz2
[foonathan::memory] Allocator foonathan::memory::memory_pool (at 0x6000010302f0) received invalid size/alignment 56, max supported is 32
libc++abi: terminating with uncaught exception of type foonathan::memory::bad_node_size: allocation node size exceeds supported maximum of allocator
zsh: abort      rviz2
(ros2) m@mbp ~ % rqt
[foonathan::memory] Allocator foonathan::memory::memory_pool (at 0x6000013ed390) received invalid size/alignment 56, max supported is 32
libc++abi: terminating with uncaught exception of type foonathan::memory::bad_node_size: allocation node size exceeds supported maximum of allocator
zsh: abort      rqt
```

traversaro, 2022-12-14:
Hello @mrkbac ! Can you report the output of `conda list` and `conda info` to help debugging? Thanks!

mrkbac, 2022-12-14:
Here are the outputs, I'm using `micromamba` I hope that doesn't cause problems.

`micromamba info`

```

                                           __
          __  ______ ___  ____ _____ ___  / /_  ____ _
         / / / / __ `__ \/ __ `/ __ `__ \/ __ \/ __ `/
        / /_/ / / / / / / /_/ / / / / / / /_/ / /_/ /
       / .___/_/ /_/ /_/\__,_/_/ /_/ /_/_.___/\__,_/
      /_/

            environment : ros2 (active)
           env location : /Users/m/micromamba/envs/ros2
      user config files : /Users/m/.mambarc
 populated config files : /Users/m/.condarc
       libmamba version : 1.1.0
     micromamba version : 1.1.0
           curl version : libcurl/7.86.0 SecureTransport (OpenSSL/1.1.1s) zlib/1.2.13 libssh2/1.10.0 nghttp2/1.47.0
     libarchive version : libarchive 3.6.1 zlib/1.2.13 bz2lib/1.0.8 libzstd/1.5.2
       virtual packages : __unix=0=0
                          __osx=13.0.1=0
                          __archspec=1=arm64
               channels : https://conda.anaconda.org/robostack-experimental/osx-arm64
                          https://conda.anaconda.org/robostack-experimental/noarch
                          https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/robostack/osx-arm64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/robostack-humble/osx-arm64
                          https://conda.anaconda.org/robostack-humble/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
       base environment : /Users/m/micromamba
               platform : osx-arm64
```

`micromamba list`

```
List of packages in environment: "/Users/m/micromamba/envs/ros2"

  Name                                               Version       Build                 Channel          
────────────────────────────────────────────────────────────────────────────────────────────────────────────
  aiohttp                                            3.8.3         py310h8e9501a_1       conda-forge      
  aiosignal                                          1.3.1         pyhd8ed1ab_0          conda-forge      
  aom                                                3.5.0         h7ea286d_0            conda-forge      
  argcomplete                                        1.12.3        pyhd8ed1ab_0          conda-forge      
  assimp                                             5.2.5         h276577b_0            conda-forge      
  async-timeout                                      4.0.2         pyhd8ed1ab_0          conda-forge      
  atk-1.0                                            2.38.0        hcb7b3dd_1            conda-forge      
  attrs                                              22.1.0        pyh71513ae_1          conda-forge      
  boost                                              1.78.0        py310h629746b_4       conda-forge      
  boost-cpp                                          1.78.0        h1cb353e_1            conda-forge      
  brotli                                             1.0.9         h1a8c8d9_8            conda-forge      
  brotli-bin                                         1.0.9         h1a8c8d9_8            conda-forge      
  bullet                                             3.21          h69fb684_4            conda-forge      
  bullet-cpp                                         3.21          py310h2b830bf_4       conda-forge      
  bzip2                                              1.0.8         h3422bc3_4            conda-forge      
  c-ares                                             1.18.1        h3422bc3_0            conda-forge      
  ca-certificates                                    2022.12.7     h4653dfc_0            conda-forge      
  cairo                                              1.16.0        h73a0509_1014         conda-forge      
  catkin_pkg                                         0.5.2         pyhd8ed1ab_0          conda-forge      
  certifi                                            2022.12.7     pyhd8ed1ab_0          conda-forge      
  cffi                                               1.15.1        py310h2399d43_2       conda-forge      
  charset-normalizer                                 2.1.1         pyhd8ed1ab_0          conda-forge      
  cmake                                              3.25.1        hf234bd0_0            conda-forge      
  colorama                                           0.4.6         pyhd8ed1ab_0          conda-forge      
  console_bridge                                     1.0.2         h3e96240_1            conda-forge      
[... 497 lines cut ...]
  xorg-libice                                        1.0.10        h27ca646_0            conda-forge      
  xorg-libsm                                         1.2.3         h27ca646_1000         conda-forge      
  xorg-libx11                                        1.7.2         h3422bc3_0            conda-forge      
  xorg-libxau                                        1.0.9         h27ca646_0            conda-forge      
  xorg-libxaw                                        1.0.14        h3422bc3_1            conda-forge      
  xorg-libxdmcp                                      1.1.3         h27ca646_0            conda-forge      
[... cut ...]
  yarl                                               1.8.1         py310h8e9501a_0       conda-forge      
  zipp                                               3.11.0        pyhd8ed1ab_0          conda-forge      
  zlib                                               1.2.13        h03a7124_4            conda-forge      
  zstd                                               1.5.2         h8128057_4            conda-forge      
  zziplib                                            0.13.69       he1e0b03_1            conda-forge   
```

traversaro, 2022-12-14:
Is this happening also if no other ROS2-program beside `ros2 topic list`? I briefly tried to reproduce this on linux-64, but I was not able to do reproduce the problem:

~~~
traversaro@IITICUBLAP257:~$ mamba activate roshumble
(roshumble) traversaro@IITICUBLAP257:~$ ros2 topic list
/parameter_events
/rosout
(roshumble) traversaro@IITICUBLAP257:~$ mamba list
# packages in environment at /home/traversaro/mambaforge/envs/roshumble:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohttp                   3.8.3           py310h5764c6d_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     22.1.0             pyh71513ae_1    conda-forge
boost                     1.78.0          py310hc4a4660_4    conda-forge
boost-cpp                 1.78.0               h75c5d50_1    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
bullet                    3.21                 hfdc917e_4    conda-forge
bullet-cpp                3.21                 h769672d_4    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.12.7            ha878542_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h255011f_2    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
cmake                     3.25.1               h816a3e0_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.0.6           py310hbf28c38_0    conda-forge
cppcheck                  2.7.5           py310h94ea96f_1    conda-forge
cryptography              38.0.4          py310h597c629_0    conda-forge
curl                      7.86.0               h7bff187_1    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19            py310hff52083_1    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.0.4              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h27087fc_0    conda-forge
ffmpeg                    5.1.2           gpl_h13415ed_104    conda-forge
fftw                      3.3.10          nompi_hf0379b8_106    conda-forge
flake8                    6.0.0              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             he05ef13_1011    conda-forge
fmt                       9.1.0                h924138e_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.1               hc2a2eb6_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0          py310h5764c6d_1    conda-forge
foonathan-memory          0.7.2                h27087fc_1    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freeimage                 3.18.0              h530e30e_10    conda-forge
freetype                  2.12.1               hca18f0e_1    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
[... 509 lines cut ...]
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxaw               1.0.14               h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.13               h7f98852_0    conda-forge
[... cut ...]
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.7.0                h27087fc_2    conda-forge
yarl                      1.8.1           py310h5764c6d_0    conda-forge
zipp                      3.11.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h6239696_4    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
~~~

mrkbac, 2022-12-14:
Yes, it also happens when starting rviz2 or when trying to use the topic monitor in rqt. Probably a macos bug?

traversaro, 2022-12-14:
I have a Mac mini in my lab, I will test with that tomorrow (CET time).

Tobias-Fischer, 2022-12-14:
I can locally reproduce, and the same happens for the osx-64 builds as well. No idea what the underlying problem is, though.

Tobias-Fischer, 2022-12-14:
So I was evil and built a foonathan-memory with the following modification:
```diff
diff --git a/recipe/build.sh b/recipe/build.sh
index 3a2a806..e3440cc 100644
--- a/recipe/build.sh
+++ b/recipe/build.sh
@@ -11,7 +11,8 @@ cmake ${CMAKE_ARGS} .. \
       -DFOONATHAN_MEMORY_BUILD_EXAMPLES=OFF \
       -DFOONATHAN_MEMORY_BUILD_TESTS=OFF \
       -DFOONATHAN_MEMORY_BUILD_TOOLS=ON \
-      -DBUILD_SHARED_LIBS=ON
+      -DBUILD_SHARED_LIBS=ON \
+      -DFOONATHAN_MEMORY_CHECK_ALLOCATION_SIZE=OFF
 
 cmake --build . --config Release -- -j$CPU_COUNT
 cmake --build . --config Release --target install
diff --git a/recipe/meta.yaml b/recipe/meta.yaml
index 570a089..90158bd 100644
--- a/recipe/meta.yaml
+++ b/recipe/meta.yaml
@@ -16,7 +16,7 @@ source:
       - fix-crosscompile.patch
 
 build:
-  number: 1
+  number: 2
   run_exports:
     - {{ pin_subpackage(name, max_pin='x.x.x') }}
 
```

Then running rviz2 works fine, but upon closing it, the following segfault occurs:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               rviz2 [53261]
Path:                  /Users/USER/*/rviz2
Identifier:            rviz2
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [40066]
Responsible:           Terminal [3926]
User ID:               504

Date/Time:             2022-12-15 07:28:25.0836 +1000
OS Version:            macOS 13.0.1 (22A400)
Report Version:        12
Anonymous UUID:        D2F57B18-67DF-1170-36B2-7552F5F0469A

Sleep/Wake UUID:       D37B2A08-4FCD-4A9E-8543-DE34461CC4AA

Time Awake Since Boot: 1000000 seconds
Time Since Wake:       780420 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000001a826
Exception Codes:       0x0000000000000001, 0x000000000001a826

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [53261]

VM Region Info: 0x1a826 is not in any region.  Bytes before following region: 4367669210
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
[... 554 lines cut ...]
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro18,3, BootROM 8419.41.10, proc 10:8:2 processors, 32 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: DELL U2719D, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Sep  3 2022 02:35:52 version 20.10.965.9.8.7.129 FWID 01-b0e84a9b
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: Laptop, Apple Inc.
Thunderbolt Bus: Laptop, Apple Inc.
Thunderbolt Bus: Laptop, Apple Inc.
```

This might help in debugging.

`ros2 topic list` still segfaults immediately.

Tobias-Fischer, 2022-12-14:
In general, things seem to go wrong when calling `destroy_node()`. Example code:
```python
import rclpy
from rclpy.node import Node
node_dummy = Node("_ros2cli_dummy_to_show_topic_list")
topic_list = node_dummy.get_topic_names_and_types()
node_dummy.destroy_node()  # error occurs here
```

Tobias-Fischer, 2022-12-16:
Note that the issue is likely fast-rtps (or fast-dds, I'm still confused by all those middlewares). The following runs perfectly:
```
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp rviz2
```

traversaro, 2022-12-16:
> Note that the issue is likely fast-rtps (or fast-dds, I'm still confused by all those middlewares). The following runs perfectly:
> 
> ```
> RMW_IMPLEMENTATION=rmw_cyclonedds_cpp rviz2
> ```

Perhaps as an quick fix we could set RMW_IMPLEMENTATION somewhere?

Tobias-Fischer, 2022-12-16:
I have recreated the ros-humble-ros-workspace package that now sets this env variable. I have overwritten the old package, so one needs to clear the cache.
Are we happy enough with this workaround and shall we close here?

mrkbac, 2022-12-17:
Yes, this works for me. Thanks, a lot!

wolfv, 2022-12-18:
Should we file a bug report with `fastrtps`? I briefly checked the `fastrtps_rmw` middleware implementation and there foonathan isn't mentioned, but I think in the FastRTPS package itself it's used.

jordanlv, 2024-01-17:
Hello @Tobias-Fischer ,
It would be great for me to use Fast DDS om my Mac.
The issue has the tag wontfix. Is it possible to fix it ?
If yes, I can try to.

Can you tell me what should I look at ?
I'm already investigating.

Thanks

Tobias-Fischer, 2024-02-06:
Hi @jordanlv - many thanks for getting in touch! I saw https://github.com/RoboStack/robostack.github.io/issues/37 and I really doubt that the DDS is at fault for you not seeing the topic list. Have you confirmed that you can ping the other machine (in either direction)? Also, have you made sure that your router/VPN etc. does not block the ports required by ROS2? And did you set the correct `ROS_DOMAIN_ID` if required?

For fixing Fast DDS: I think the best way is to dig deeper starting from https://github.com/RoboStack/ros-humble/issues/32#issuecomment-1352229655 - it will probably be not much fun ;). The StackTrace might help. It might also be the case that a new version of Fast DDS has resolved the issues; probably worth trying the new packages that we've recently built here and see if they still have the same issue.

jordanlv, 2024-02-06:
Yes, I already check things like this. 

The problem is that I'm using a Turtlebot4 with [discovery setup](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html) server. In this configuration, "every device in the system must use rmw_fastrtps_cpp as their DDS". WIth an Ubuntu PC I can also see this problem by switching between Cyclone DDS and Fast DDS. With Cyclone DDS, it cant work but with Fast DDS, I get all the nodes and the topics.

I will try the new packages asap

Tobias-Fischer, 2024-02-06:
Interesting, I wasn’t aware of this. I guess switching to simple discovery is not an option for you?

jordanlv, 2024-02-06:
Yes, I would preferer to stay in discovery server.

Tobias-Fischer, 2024-07-15:
@xingruiyang @traversaro - FYI, I just updated the version of foonathan-memory on conda-forge (https://github.com/conda-forge/foonathan-memory-feedstock/pull/8) which might contain some bugfixes.

traversaro, 2024-07-15:
> @xingruiyang @traversaro - FYI, I just updated the version of foonathan-memory on conda-forge ([conda-forge/foonathan-memory-feedstock#8](https://github.com/conda-forge/foonathan-memory-feedstock/pull/8)) which might contain some bugfixes.

Great! However the abi compatibility of foonathan-memory is at the patch level (see https://github.com/conda-forge/foonathan-memory-feedstock/blob/13588d02ef478e94c60b7ce82a347385789c34d7/recipe/meta.yaml#L21), so maybe this requires a package rebuild?

Tobias-Fischer, 2024-07-15:
Yes, it does ..

starlitxiling, 2025-01-06:
I also encountered this problem when using ros2 on osx. For some reason, I had to use `fastrtps`. The above pr has been merged into `fooathan-memory-feedstock0.7.3`, but ros humble installed through mamba is still `0.7.2`. When I force update to `0.7.3`, my ros cli will be deleted. There is a conflict between their dependencies. What should I do?

traversaro, 2025-01-09:
> I also encountered this problem when using ros2 on osx. For some reason, I had to use `fastrtps`. The above pr has been merged into `fooathan-memory-feedstock0.7.3`, but ros humble installed through [...]

@starlitxiling can you try to create a new environment? Three days ago we merged our rebuild (see https://github.com/RoboStack/ros-humble/pull/229), and the new uploaded packages should not be affected anymore (`ros2-distro-mutex==0.6.*`).

starlitxiling, 2025-01-13:
hello, I am sorry reply you until now.I create a new environment try new uploaded. It work correctly!!! Thanks for yours help.

traversaro, 2025-01-15:
> hello, I am sorry reply you until now.I create a new environment try new uploaded. It work correctly!!! Thanks for yours help.

No problem, 3 days with the weekend in the middle is a fast response. Happy that everything works, if you still have problems feel free to open a new issue, thanks!

traversaro, 2025-01-15:
@Tobias-Fischer I guess we can close the issue?

closed 2025-01-15 by Tobias-Fischer