RoboStack/ros-humble #33: moveit-setup-framework linking issues on Win

Tobias-Fischer, 2022-12-14:
For moveit-setup-framework, we have the following linker error:
```
2022-12-14T22:16:27.4366704Z   Generating Code...
2022-12-14T22:16:41.2728968Z   Auto build dll exports
2022-12-14T22:16:42.4919769Z      Creating library C:/bld/ros-humble-moveit-setup-framework-0_1671055682237/work/build/Release/moveit_setup_framework.lib and object C:/bld/ros-humble-moveit-setup-framework-0_1671055682237/work/build/Release/moveit_setup_framework.exp
2022-12-14T22:16:42.6939934Z rviz_panel.obj : error LNK2019: unresolved external symbol "public: __cdecl rviz_common::VisualizationManager::VisualizationManager(class rviz_common::RenderPanel *,class std::weak_ptr<class rviz_common::ros_integration::RosNodeAbstractionIface>,class rviz_common::WindowManagerInterface *,class std::shared_ptr<class rclcpp::Clock>)" (??0VisualizationManager@rviz_common@@QEAA@PEAVRenderPanel@1@V?$weak_ptr@VRosNodeAbstractionIface@ros_integration@rviz_common@@@std@@PEAVWindowManagerInterface@1@V?$shared_ptr@VClock@rclcpp@@@4@@Z) referenced in function "public: void __cdecl moveit_setup::RVizPanel::initialize(void)" (?initialize@RVizPanel@moveit_setup@@QEAAXXZ) [%SRC_DIR%\build\moveit_setup_framework.vcxproj]
2022-12-14T22:16:42.6942951Z rviz_panel.obj : error LNK2019: unresolved external symbol "public: void __cdecl rviz_common::VisualizationManager::initialize(void)" (?initialize@VisualizationManager@rviz_common@@QEAAXXZ) referenced in function "public: void __cdecl moveit_setup::RVizPanel::initialize(void)" (?initialize@RVizPanel@moveit_setup@@QEAAXXZ) [%SRC_DIR%\build\moveit_setup_framework.vcxproj]
2022-12-14T22:16:42.6944973Z rviz_panel.obj : error LNK2019: unresolved external symbol "public: void __cdecl rviz_common::VisualizationManager::startUpdate(void)" (?startUpdate@VisualizationManager@rviz_common@@QEAAXXZ) referenced in function "public: void __cdecl moveit_setup::RVizPanel::initialize(void)" (?initialize@RVizPanel@moveit_setup@@QEAAXXZ) [%SRC_DIR%\build\moveit_setup_framework.vcxproj]
2022-12-14T22:16:42.6946542Z rviz_panel.obj : error LNK2019: unresolved external symbol "public: void __cdecl rviz_common::VisualizationManager::addDisplay(class rviz_common::Display *,bool)" (?addDisplay@VisualizationManager@rviz_common@@QEAAXPEAVDisplay@2@_N@Z) referenced in function "public: void __cdecl moveit_setup::RVizPanel::initialize(void)" (?initialize@RVizPanel@moveit_setup@@QEAAXXZ) [%SRC_DIR%\build\moveit_setup_framework.vcxproj]
2022-12-14T22:16:42.6948891Z rviz_panel.obj : error LNK2019: unresolved external symbol "public: void __cdecl rviz_common::VisualizationManager::removeAllDisplays(void)" (?removeAllDisplays@VisualizationManager@rviz_common@@QEAAXXZ) referenced in function "public: virtual __cdecl moveit_setup::RVizPanel::~RVizPanel(void)" (??1RVizPanel@moveit_setup@@UEAA@XZ) [%SRC_DIR%\build\moveit_setup_framework.vcxproj]
2022-12-14T22:16:42.6951198Z rviz_panel.obj : error LNK2019: unresolved external symbol "public: void __cdecl rviz_common::VisualizationManager::setFixedFrame(class QString const &)" (?setFixedFrame@VisualizationManager@rviz_common@@QEAAXAEBVQString@@@Z) referenced in function "public: void __cdecl moveit_setup::RVizPanel::initialize(void)" (?initialize@RVizPanel@moveit_setup@@QEAAXXZ) [%SRC_DIR%\build\moveit_setup_framework.vcxproj]
2022-12-14T22:16:42.7287992Z %SRC_DIR%\build\Release\moveit_setup_framework.dll : fatal error LNK1120: 6 unresolved externals [%SRC_DIR%\build\moveit_setup_framework.vcxproj]
```

I'm not sure why this occurs - rviz_common is added as a dependency correctly in https://github.com/ros-planning/moveit2/blob/e65f457734b3c9dc992d11a382651a058a3bcaa4/moveit_setup_assistant/moveit_setup_framework/CMakeLists.txt#L58

Sorry to bother you again @traversaro, but do you have any idea?

Tobias-Fischer, 2022-12-15:
I hope this is fixed with https://github.com/RoboStack/ros-humble/commit/f6fc7c6d6b80fd18dfe7e9c085d4c4f8fd30a0ba

Tobias-Fischer, 2022-12-15:
Now we are running into a similar problem for moveit-setup-core-plugins. However, the solution does not seem to be as easy:
```
Auto build dll exports
[1550](https://github.com/RoboStack/ros-humble/actions/runs/3701257631/jobs/6270413604#step:6:1551)
     Creating library C:/bld/ros-humble-moveit-setup-core-plugins-0_1671081099777/work/build/Release/moveit_setup_core_plugins.lib and object C:/bld/ros-humble-moveit-setup-core-plugins-0_1671081099777/work/build/Release/moveit_setup_core_plugins.exp
[1551](https://github.com/RoboStack/ros-humble/actions/runs/3701257631/jobs/6270413604#step:6:1552)
configuration_files.obj : error LNK2019: unresolved external symbol "public: static class std::vector<struct moveit_setup::TemplateVariable,class std::allocator<struct moveit_setup::TemplateVariable> > moveit_setup::TemplatedGeneratedFile::variables_" (?variables_@TemplatedGeneratedFile@moveit_setup@@2V?$vector@UTemplateVariable@moveit_setup@@V?$allocator@UTemplateVariable@moveit_setup@@@std@@@std@@A) referenced in function "public: void __cdecl moveit_setup::core::ConfigurationFiles::loadFiles(void)" (?loadFiles@ConfigurationFiles@core@moveit_setup@@QEAAXXZ) [%SRC_DIR%\build\moveit_setup_core_plugins.vcxproj]
[1552](https://github.com/RoboStack/ros-humble/actions/runs/3701257631/jobs/6270413604#step:6:1553)
moc_start_screen_widget.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const moveit_setup::SetupStepWidget::staticMetaObject" (?staticMetaObject@SetupStepWidget@moveit_setup@@2UQMetaObject@@B) [%SRC_DIR%\build\moveit_setup_core_plugins.vcxproj]
[1553](https://github.com/RoboStack/ros-humble/actions/runs/3701257631/jobs/6270413604#step:6:1554)
moc_configuration_files_widget.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const moveit_setup::SetupStepWidget::staticMetaObject" (?staticMetaObject@SetupStepWidget@moveit_setup@@2UQMetaObject@@B) [%SRC_DIR%\build\moveit_setup_core_plugins.vcxproj]
[1554](https://github.com/RoboStack/ros-humble/actions/runs/3701257631/jobs/6270413604#step:6:1555)
moc_author_information_widget.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const moveit_setup::SetupStepWidget::staticMetaObject" (?staticMetaObject@SetupStepWidget@moveit_setup@@2UQMetaObject@@B) [%SRC_DIR%\build\moveit_setup_core_plugins.vcxproj]
[1555](https://github.com/RoboStack/ros-humble/actions/runs/3701257631/jobs/6270413604#step:6:1556)
%SRC_DIR%\build\Release\moveit_setup_core_plugins.dll : fatal error LNK1120: 2 unresolved externals [%SRC_DIR%\build\moveit_setup_core_plugins.vcxproj]
[1556](https://github.com/RoboStack/ros-humble/actions/runs/3701257631/jobs/6270413604#step:6:1557)
```

I guess we need to extend moveit-setup-framework with export visibility headers? This was done for many packages in https://github.com/ros-planning/moveit2/pull/530 but not for moveit-setup-framework. What do you think @traversaro?

Tobias-Fischer, 2022-12-15:
Note that `set(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS ON)` already happens in the `moveit_package` macro .. :(

traversaro, 2022-12-15:
> Note that `set(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS ON)` already happens in the `moveit_package` macro .. :(

That does not work for static variables. For those we need to manually export the visi

Tobias-Fischer, 2022-12-15:
I’m going to use this issue as a general windows build issue thread. On another build something I haven’t seen before:
RuntimeError: copy_symlink: The file or directory is not a reparse point.: 
[5298](https://github.com/RoboStack/ros-humble/actions/runs/3701570027/jobs/6270998023#step:6:5299)
"C:\Users\runneradmin\conda_pkgs_dir\libignition-physics5-5.2.0-hef61a30_2\Libr
[5299](https://github.com/RoboStack/ros-humble/actions/runs/3701570027/jobs/6270998023#step:6:5300)
ary\lib\ign-physics-5\engine-plugins\ignition-physics-bullet-plugin.dll", 
[5300](https://github.com/RoboStack/ros-humble/actions/runs/3701570027/jobs/6270998023#step:6:5301)
"C:\bld\ros-humble-ros-gz-sim-0_1671085464153\_h_env\Library\lib\ign-physics-5\
[5301](https://github.com/RoboStack/ros-humble/actions/runs/3701570027/jobs/6270998023#step:6:5302)
engine-plugins\ignition-physics-bullet-plugin.dll"

https://github.com/RoboStack/ros-humble/actions/runs/3701570027/jobs/6270998023

this might be one for you @wolfv?

traversaro, 2022-12-15:
> > Note that `set(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS ON)` already happens in the `moveit_package` macro .. :(
> 
> That does not work for static variables. For those we need to manually export the visibility.

Sorry, I was brief as I was on the phone. Yes, for static variables `CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS` does not, work, so you manually need to set the visibility flag, quoting from https://www.kitware.com/create-dlls-on-windows-without-declspec-using-new-cmake-export-all-feature/ :

> This should turn all add_library calls that do not explicitly specify build type into shared builds. If there are no global data variables in the project, all libraries will be built as DLLs with no errors.  If you run into undefined symbols, check for global data like static data members of classes. 
> The easiest way to handle the global data is to use the CMake GenerateExportHeader module like this:
> ~~~cmake
> add_library(mylibrary ${mylibrary_SOURCES})
> # add these lines
> include(GenerateExportHeader)
> generate_export_header(mylibrary)
> ~~~
> 
> // edit the source like this:
> 
>
> ~~~cmake
> #include <mylibary_export.h>
> class myclass
> {
>  static mylibrary_EXPORT int GlobalCounter;
> …
>~~~
> 
> Note, if you use GenerateExportHeader and still want static builds to work, you will need to add a -Dmylibrary_STATIC during static builds. See the generated mylibary_export.h for more details.

In practice, this is a bit more tricky as we also need to make sure that the export header is properly installed.

Tobias-Fischer, 2022-12-15:
Thanks @traversaro! I created a new branch in my moveit2 fork and provided you with access to it: https://github.com/Tobias-Fischer/moveit2/tree/export_headers

Let's see whether this fixes the issue ..

Tobias-Fischer, 2022-12-15:
Ok, seems like I found the culprit for the symlink error on Windows as well: https://github.com/conda-forge/libignition-physics-feedstock/pull/31

Tobias-Fischer, 2022-12-16:
Both issues have been resolved .. let's see what's next ;)

Tobias-Fischer, 2022-12-16:
So we should have ros-humble-desktop-full, ros-humble-moveit2 and ros-humble-navigation2 any second now. Any chance anyone could test whether rviz2 and moveit2 works?

closed 2022-12-16 by Tobias-Fischer

traversaro, 2022-12-16:
> So we should have ros-humble-desktop-full, ros-humble-moveit2 and ros-humble-navigation2 any second now. Any chance anyone could test whether rviz2 and moveit2 works?

I tried rviz2 and it works. Do you have some simple check for moveit2 ? I tried the tutorial, but they do not work on Windows due to common Windows compilation problems (M_PI, ..).

Tobias-Fischer, 2022-12-16:
Awesome re rviz, thanks for checking. I’m not sure re moveit - I’ll have a look later