RoboStack/ros-humble #34: ROS2 custom msgs failing - AARCH64

b1n-ch1kn, 2022-12-24:
Reference query: https://gitter.im/RoboStack/Lobby?at=6379895a2cb657024035f2ab

Host: Jetson Nano, robostack-humble
Mamba Packages: https://pastebin.com/raw/ww0ax7Zr

When building custom msgs on Humble (and Python 3.9), we run into the following error: https://pastebin.com/raw/yA9JraPd.
Specifically:
```
    from __future__ import annotations
    ^
SyntaxError: future feature annotations is not defined
```

@Tobias-Fischer believes there's a CI setup issue on aarch64.

Tobias-Fischer, 2022-12-24:
Yes - we need to rebuild the packages but the CI times out after 20 minutes @wolfv

Tobias-Fischer, 2023-01-24:
Hi @aktech, sorry to ping you. Our cirun builds time out after 25 minutes. Is this normal?

aktech, 2023-01-24:
Hey @Tobias-Fischer that's not normal, I'll take a look. Can you share the build please? (also, don't hesitate to ping me)

aktech, 2023-01-24:
I guess you're referring these:

[image]

Looking at the summary, it seems they were potentially starving of memory/cpu, try a larger instance maybe?
[image]

I checked our database and saw they were terminated after instructed github's webhook, so un natural termination by cirun:

| name                                | terminated_at              | terminated_via |
| ----------------------------------- | -------------------------- | -------------- |
| cirun-robostack--ros-humble-c60ab63 | 2023-01-24 06:31:43.668931 | webhook        |
| cirun-robostack--ros-humble-e62946f | 2023-01-24 06:36:44.147264 | webhook        |

Tobias-Fischer, 2023-01-24:
@wolfv  any chance you could setup a larger server to test?

wolfv, 2023-01-24:
@Tobias-Fischer we can use xxlarge: https://github.com/RoboStack/ros-humble/blob/540c44e42d3fe63a7b5a8cc45c46aa3719227cb9/.cirun.yml#L4

wolfv, 2023-01-24:
Sorry, it's 2xlarge (or larger :)

[image]

aktech, 2023-01-24:
Let me know if it doesn't works with that too, we can debug together.

Tobias-Fischer, 2023-01-24:
Seems to have worked. Thanks everyone!

closed 2023-01-24 by Tobias-Fischer