RoboStack/ros-humble #37: running rqt_graph on macos 13.1 m1 chip comes with the following warning .

zhengyunqq, 2023-01-07:
Could not import "pyqt" bindings of qt_gui_cpp library - so C++ plugins will not be available:
Traceback (most recent call last):
  File "/Users/zy123/mambaforge/envs/humble/lib/python3.9/site-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ImportError: dlopen(/Users/zy123/mambaforge/envs/humble/lib/python3.9/site-packages/qt_gui_cpp/libqt_gui_cpp_sip.cpython-39-darwin.so, 0x0002): symbol not found in flat namespace '__ZN10qt_gui_cpp12PluginBridge16staticMetaObjectE'

[image]

Exactly follow the instruction at [](https://robostack.github.io/GettingStarted.html)

traversaro, 2023-01-07:
Hello @zhengyunqq, to simplify debugging can you report the output of the following commands:
* `conda info`
* `conda list`
?

Thanks!

zhengyunqq, 2023-01-07:
[image]

conda list                                                                                                                                                                                     
packages in environment at /Users/zy123/mambaforge/envs/humble:
Name                    Version                   Build  Channel
aom                       3.5.0                h7ea286d_0    conda-forge
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.2.4                h3287542_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
boost                     1.74.0           py39h776e71f_5    conda-forge
boost-cpp                 1.74.0               h1cb353e_8    conda-forge
brotli                    1.0.9                h1a8c8d9_8    conda-forge
brotli-bin                1.0.9                h1a8c8d9_8    conda-forge
bullet                    3.21                 h85045c0_4    conda-forge
bullet-cpp                3.21             py39hde7b980_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.18.1               h3422bc3_0    conda-forge
c-compiler                1.5.2                h5008568_0    conda-forge
ca-certificates           2022.12.7            h4653dfc_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.2              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hcbb26d4_11    conda-forge
cctools_osx-arm64         973.0.1             hef52d2f_11    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h7e6b969_3    conda-forge
clang                     14.0.6               hce30654_0    conda-forge
clang-14                  14.0.6          default_h81a5282_0    conda-forge
clang_osx-arm64           14.0.6               h15773ab_4    conda-forge
clangxx                   14.0.6          default_hb7ecf47_0    conda-forge
clangxx_osx-arm64         14.0.6               he29aa18_4    conda-forge
cmake                     3.25.1               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39h2804cbe_1    conda-forge
colcon-core               0.10.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.5                      py_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.12                     py_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.4.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compiler-rt               14.0.6               h30b49de_0    conda-forge
compiler-rt_osx-arm64     14.0.6               h48302dc_0    conda-forge
compilers                 1.5.2                hce30654_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.0.6            py39haaf3ac1_0    conda-forge
coreutils                 9.1                  h1c322ee_0    conda-forge
coverage                  7.0.3            py39h02fc5c5_0    conda-forge
cppcheck                  2.7.5            py39h7791993_1    conda-forge
cryptography              39.0.0           py39haa0b8cc_0    conda-forge
curl                      7.86.0               hd538317_1    conda-forge
cxx-compiler              1.5.2                hffc8910_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
distlib                   0.3.6              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
[... 509 lines cut ...]
x265                      3.5                  hbc6ce65_3    conda-forge
xorg-kbproto              1.0.7             h27ca646_1002    conda-forge
xorg-libice               1.0.10               h27ca646_0    conda-forge
xorg-libsm                1.2.3             h27ca646_1000    conda-forge
xorg-libx11               1.7.2                h3422bc3_0    conda-forge
[... cut ...]
xz                        5.2.6                h57fd34a_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
yaml-cpp                  0.7.0                hb7217d7_2    conda-forge
zipp                      3.11.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h03a7124_4    conda-forge
zstd                      1.5.2                h8128057_4    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge

Tobias-Fischer, 2023-01-07:
This is fixed with the new builds in the robostack-staging channel. We should update the readme ..

Tobias-Fischer, 2023-01-07:
Just updated the readme. Could you please recreate the environment with the new instructions?

closed 2023-01-07 by Tobias-Fischer

zhengyunqq, 2023-01-08:
Thanks very much for the quick fix.