RoboStack/ros-humble #39: cannot build custom messages on macos

tudoroancea, 2023-01-21:
Hi there,

I am trying to follow the following [tutorial](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html) to create custom ros interfaces but for some reason when I try to build the package with `colcon build` I get the following error:
```--- stderr: tutorial_interfaces
CMake Error at /Users/tudoroancea/miniforge3/envs/ros_humble/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_NumPy_INCLUDE_DIRS NumPy) (found
  version "3.11.1")
Call Stack (most recent call first):
  /Users/tudoroancea/miniforge3/envs/ros_humble/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /Users/tudoroancea/miniforge3/envs/ros_humble/share/cmake-3.25/Modules/FindPython/Support.cmake:3245 (find_package_handle_standard_args)
  /Users/tudoroancea/miniforge3/envs/ros_humble/share/cmake-3.25/Modules/FindPython3.cmake:490 (include)
  /Users/tudoroancea/miniforge3/envs/ros_humble/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:24 (find_package)
  /Users/tudoroancea/miniforge3/envs/ros_humble/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /Users/tudoroancea/miniforge3/envs/ros_humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:318 (ament_execute_extensions)
  CMakeLists.txt:15 (rosidl_generate_interfaces)
```
I think this has something to do with another python version I installed with pyenv (3.11.1 like mentioned in the error message) but I cannot figure out why because I init pyenv before conda in my `.zshrc` file and when I do `which python3` , the first interpreter that is found is the one installed in the conda env in which ros is installed.
Do you have any idea how to fix this and if this may come from the ros installation or is a colcon issue?

Just in case, here is the list of ros packages in my conda env:
```
# packages in environment at /Users/tudoroancea/miniforge3/envs/ros_humble:
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
ros-humble-action-msgs    1.2.1           py310h927cc32_2    robostack-staging
ros-humble-action-tutorials-cpp 0.20.2          py310h927cc32_2    robostack-staging
ros-humble-action-tutorials-interfaces 0.20.2          py310h927cc32_2    robostack-staging
ros-humble-action-tutorials-py 0.20.2          py310h927cc32_2    robostack-staging
ros-humble-actionlib-msgs 4.2.2           py310h927cc32_2    robostack-staging
ros-humble-ament-cmake    1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-auto 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-copyright 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-core 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-cppcheck 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-cpplint 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-export-definitions 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-export-dependencies 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-export-include-directories 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-export-interfaces 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-export-libraries 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-export-link-flags 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-export-targets 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-flake8 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-gen-version-h 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-gmock 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-gtest 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-include-directories 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-libraries 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-lint-cmake 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-pep257 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-pytest 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-python 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-ros 0.10.0          py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-target-dependencies 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-test 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-uncrustify 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-version 1.3.3           py310hd751b12_2    robostack-staging
ros-humble-ament-cmake-xmllint 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-copyright 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-cppcheck 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-cpplint  0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-flake8   0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-index-cpp 1.4.0           py310hd751b12_2    robostack-staging
ros-humble-ament-index-python 1.4.0           py310hd751b12_2    robostack-staging
ros-humble-ament-lint     0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-lint-auto 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-lint-cmake 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-lint-common 0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-package  0.14.0          py310hd751b12_2    robostack-staging
ros-humble-ament-pep257   0.12.4          py310hd751b12_2    robostack-staging
ros-humble-ament-uncrustify 0.12.4          py310hd751b12_2    robostack-staging
[... 223 lines cut ...]
ros-humble-tinyxml2-vendor 0.7.5           py310hb0861aa_2    robostack-staging
ros-humble-topic-monitor  0.20.2          py310h927cc32_2    robostack-staging
ros-humble-tracetools     4.1.1           py310hd751b12_2    robostack-staging
ros-humble-trajectory-msgs 4.2.2           py310h927cc32_2    robostack-staging
ros-humble-turtlesim      1.4.2           py310h927cc32_2    robostack-staging
ros-humble-uncrustify-vendor 2.0.2           py310hd751b12_2    robostack-staging
ros-humble-unique-identifier-msgs 2.2.1           py310h927cc32_2    robostack-staging
ros-humble-urdf           2.6.0           py310hd751b12_2    robostack-staging
ros-humble-urdf-parser-plugin 2.6.0           py310hd751b12_2    robostack-staging
ros-humble-urdfdom        3.0.2           py310h648553b_2    robostack-staging
ros-humble-urdfdom-headers 1.0.6           py310hd751b12_2    robostack-staging
ros-humble-visualization-msgs 4.2.2           py310h927cc32_2    robostack-staging
ros-humble-yaml-cpp-vendor 8.0.1           py310hd751b12_2    robostack-staging
ros-humble-zstd-vendor    0.15.3          py310h0ee586c_2    robostack-staging
ros2-distro-mutex         0.2.0                    humble    robostack-staging
rosdep                    0.22.1             pyhd8ed1ab_0    conda-forge
rosdistro                 0.9.0           py310hbe9552e_0    conda-forge
rospkg                    1.4.0              pyhd8ed1ab_0    conda-forge
```

Thanks a lot for your help!

Best wishes,

Ted

tudoroancea, 2023-01-21:
Interestingly enough, if I uninstall this python version from pyenv, it still finds a 3.11.1 python executable and I have no idea why

traversaro, 2023-01-21:
> Just in case, here is the list of ros packages in my conda env:

Can you report all the packages installed in your environment? As this seems to be related to the interaction with CMake with the Python packages, for example it is interesting to understand which version of numpy do you have installed, as CMake seems to complain it is not able to find numpy.

> Interestingly enough, if I uninstall this python version from pyenv, it still finds a 3.11.1 python executable and I have no idea why

Are you trying with a clean build directory? Otherwise this could influence the result. Anyhow, probably the best way to debug this is to upload the `CMakeCache.txt`  of the package whose build its failing (it should be in `ws/build/<nameofpkgthatisfailing>/CMakeCache.txt`) . Furthermore, it would be also interesting to get the output of `env` of the terminal in which you are running `colcon` (make sure that it does not include anything that it can't be shared such as API tokens or similar.

traversaro, 2023-01-21:
By the way, did you tried to not initialize pyenv at all?

tudoroancea, 2023-01-21:
I tried to comment out the lines initializing pyenv in my `.zshrc` and still obtained the same error, on a fresh new build (I removed the `build` and `install` dirs). Here is the list of installed packages in the conda env:
```
# packages in environment at /Users/tudoroancea/miniforge3/envs/ros_humble:
#
# Name                    Version                   Build  Channel
aiohttp                   3.8.3           py310h8e9501a_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argcomplete               1.12.3             pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
black                     22.12.0         py310hbe9552e_0    conda-forge
boost                     1.78.0          py310h629746b_4    conda-forge
boost-cpp                 1.78.0               h1cb353e_1    conda-forge
brotli                    1.0.9                h1a8c8d9_8    conda-forge
brotli-bin                1.0.9                h1a8c8d9_8    conda-forge
brotlipy                  0.7.0           py310h8e9501a_1005    conda-forge
bullet                    3.21                 h69fb684_4    conda-forge
bullet-cpp                3.21            py310h2b830bf_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.18.1               h3422bc3_0    conda-forge
c-compiler                1.5.2                h5008568_0    conda-forge
ca-certificates           2022.12.7            h4653dfc_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hcbb26d4_11    conda-forge
cctools_osx-arm64         973.0.1             hef52d2f_11    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h2399d43_3    conda-forge
cfgv                      3.3.1              pyhd8ed1ab_0    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
clang                     14.0.6               hce30654_0    conda-forge
clang-14                  14.0.6          default_h81a5282_0    conda-forge
clang_osx-arm64           14.0.6               h15773ab_4    conda-forge
clangxx                   14.0.6          default_hb7ecf47_0    conda-forge
clangxx_osx-arm64         14.0.6               he29aa18_4    conda-forge
click                     8.1.3           unix_pyhd8ed1ab_2    conda-forge
cmake                     3.25.1               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py310hbe9552e_1    conda-forge
colcon-core               0.10.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.5                      py_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.12                     py_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.4.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compiler-rt               14.0.6               h30b49de_0    conda-forge
compiler-rt_osx-arm64     14.0.6               h48302dc_0    conda-forge
compilers                 1.5.2                hce30654_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.0.7           py310h2887b22_0    conda-forge
[... 1297 lines cut ...]
//The paths to the Python include directories.
PythonExtra_INCLUDE_DIRS:INTERNAL=/Users/tudoroancea/miniforge3/envs/ros_humble/include/python3.10;/Users/tudoroancea/miniforge3/envs/ros_humble/include/python3.10
//The libraries that need to be linked against for Python extensions.
PythonExtra_LIBRARIES:INTERNAL=
//Compiler reason failure
[... cut ...]
_Python3_INTERPRETER_SIGNATURE:INTERNAL=9af7aa9dccd61f082b164a191cd68f9c
//Interpreter reason failure
_Python3_Interpreter_REASON_FAILURE:INTERNAL=
//Path to a library.
_Python3_LIBRARY_RELEASE:INTERNAL=/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/libpython3.11.dylib
//Path to a file.
_Python3_NumPy_INCLUDE_DIR:INTERNAL=_Python3_NumPy_INCLUDE_DIR-NOTFOUND
//NumPy reason failure
_Python3_NumPy_REASON_FAILURE:INTERNAL=
//Index for unique symlink install targets
__AMENT_CMAKE_SYMLINK_INSTALL_TARGETS_INDEX:INTERNAL=0
```

tudoroancea, 2023-01-21:
From this `CMakeCache.txt` I found out that I still have a python 3.11 that was installed in Homebrew as dependency of other packages. However, I checked and it had numpy installed.
When uninstalling it, everything worked fine :tada:
It is nonetheless a strange behavior, right?

traversaro, 2023-01-22:
> It is nonetheless a strange behavior, right?

In general in conda-forge and robostack packaging we try to make sure that non-conda packages are not found by CMake, but it is not simple to do so. For example, I suspect that in this case `FindPython.cmake` module (the code behind the `find_package(Python3 REQUIRED)` call) prefer always the newest Python, so this may be the reason why the homebrew Python. In general, it is convenient to minimize the amount of other python that are installed in the system. Another trick is to explicitly specify which Python to use to CMake, via: 

~~~
--cmake-args "-DPython_EXECUTABLE=$CONDA_PYTHON_EXE  -DPython3_EXECUTABLE=$CONDA_PYTHON_EXE`
~~~
if you are using colcon or:
~~~
-DPython_EXECUTABLE=$CONDA_PYTHON_EXE  -DPython3_EXECUTABLE=$CONDA_PYTHON_EXE
~~~
if you are invoking CMake directly.

tudoroancea, 2023-01-22:
That works just fine! thank you very much for all your help !

closed 2023-01-22 by tudoroancea

duyu, 2023-06-09:
Hi, I think the problem is this: `CMAKE_PREFIX_PATH=:/Users/tudoroancea/miniforge3/envs/ros_humble`, I'm not sure why this is set with a `:` symbol though, I checked in ros_noetic env, it's without `:`.

Anyone could help check this?

duyu, 2023-06-09:
You are right, thanks;)