RoboStack/ros-humble #40: Move urdfdom-py to conda-forge?

traversaro, 2023-01-25:
In our lab we currently have a pure Python library (https://github.com/ami-iit/ADAM) that was released on PyPI (https://pypi.org/project/adam-robotics/), and we would like to release on conda-forge (upstream issue: https://github.com/ami-iit/ADAM/issues/26).

The problem is this library depends on the Python module `urdf_parser_py`, that is available in PyPI in the `urdf-parser-py` package (see https://pypi.org/project/urdf-parser-py/). The latest version of the `urdf_parser_py` module is anyhow also release in ROS2, in the `urdfdom-py` package. To permit to package adam on conda-forge and without creating possible incompatibility/clobbering problems with robostack's ROS2 package, my idea is to:
* Release the `urdf_parser_py` python module from https://github.com/ros/urdf_parser_py as `urdfdom-py` package in conda-forge (sorry for the confusion between `urdf_parser_py`/`urdfdom-py`, I just following the choice made when the library/repo were written)
* Add in https://github.com/RoboStack/ros-humble/tree/main/additional_recipes a `ros-humble-urdfdom-py` recipe that does not contain anything, at only has a run dependency on the urdfdom-py package from conda-forge with the same version, a bit like we did for https://github.com/RoboStack/ros-humble/issues/10

Do you think it make sense or there could be problems I am not considering? Thanks in advance! @Tobias-Fischer @wolfv

Tobias-Fischer, 2023-01-25:
Sounds all good to me, thanks!

Tobias-Fischer, 2023-05-24:
That's done, thanks @traversaro

closed 2023-05-24 by Tobias-Fischer