RoboStack/ros-humble #41: Nodes from pure python packages can't be started via ros2 run on Linux

traversaro, 2023-01-26:
Running the command `ros2 run demo_nodes_py demo_talker` results in an error:
~~~
(rosurdfdom) traversaro@IITICUBLAP257:~$ ros2 run demo_nodes_py demo_talker
No executable found
~~~

While the same command on Windows works fine:
~~~
(ros-humble) C:\Users\STraversaro>ros2 run demo_nodes_py demo_talker
[INFO] [1674722953.116532500] [talker]: Publishing: "Hello World: 0"
[INFO] [1674722953.946947500] [talker]: Publishing: "Hello World: 1"
[INFO] [1674722954.944548600] [talker]: Publishing: "Hello World: 2"
~~~

The problem is that the executables/entry points on Linux are installed in `<install_prefix>/bin`, instead of the correct location `<install_prefix>/lib/<pkg_name>`.

This is kind of a regression of https://github.com/RoboStack/ros-galactic/issues/74, as there the problem affected Windows packages, now Linux (and I guess macOS) packages.

Tobias-Fischer, 2023-01-26:
Just to be sure, those are the packages in staging?

traversaro, 2023-01-26:
Yes, after asking to everyone to post the `mamba list` I forgot to do it myself. :D 

~~~
(roshumble) traversaro@IITICUBLAP257:~/mambaforge/envs/roshumble/share/ament_index/resource_index$ ros2 run demo_nodes_py demo_talker
No executable found
(roshumble) traversaro@IITICUBLAP257:~/mambaforge/envs/roshumble/share/ament_index/resource_index$ mamba list
# packages in environment at /home/traversaro/mambaforge/envs/roshumble:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohttp                   3.8.3           py310h5764c6d_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
argcomplete               1.12.3             pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
boost                     1.78.0          py310hc4a4660_4    conda-forge
boost-cpp                 1.78.0               h75c5d50_1    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
bullet                    3.21                 hfdc917e_4    conda-forge
bullet-cpp                3.21                 h769672d_4    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.12.7            ha878542_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h255011f_3    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
cmake                     3.25.2               h077f3f9_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.0.7           py310hdf3cbec_0    conda-forge
cppcheck                  2.7.5           py310h94ea96f_1    conda-forge
cryptography              39.0.0          py310h65dfdc0_0    conda-forge
curl                      7.87.0               h6312ad2_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19            py310hff52083_1    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.1.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h27087fc_0    conda-forge
ffmpeg                    5.1.2           gpl_h8dda1f0_106    conda-forge
fftw                      3.3.10          nompi_hf0379b8_106    conda-forge
flake8                    6.0.0              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             he05ef13_1011    conda-forge
fmt                       9.1.0                h924138e_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.1               hc2a2eb6_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0          py310h5764c6d_1    conda-forge
foonathan-memory          0.7.2                h27087fc_1    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freeimage                 3.18.0              h530e30e_10    conda-forge
freetype                  2.12.1               hca18f0e_1    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
frozenlist                1.3.3           py310h5764c6d_0    conda-forge
[... 509 lines cut ...]
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxaw               1.0.14               h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
[... cut ...]
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.7.0                h27087fc_2    conda-forge
yarl                      1.8.2           py310h5764c6d_0    conda-forge
zipp                      3.11.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h3eb15da_6    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
~~~

traversaro, 2023-01-26:
Ok, I think I understood what is going on. I tried to install a simple repo that installs a file in `<install_prefix>/lib/<pkg_name>`, i.e. https://github.com/ros/urdf_parser_py, and the entry point/executable is installed in the correct location for Python 3.7/3.8/3.9, while in the wrong location `<install_prefix>/bin` for Python 3.10/3.11 . Indeed, for Python 3.7/3.8/3.9 when installing a clear deprecation warning is printed:
~~~
Installing collected packages: urdfdom-py
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
  changing mode of /home/traversaro/mambaforge/envs/urdfpydev/lib/urdfdom_py/display_urdf to 755
~~~

So probably now the `setup.cfg` are ignored. I wonder how this is handled by humble's official apt packages, that anyhow are installed via Python 3.10 on Ubuntu 22.04 .

traversaro, 2023-01-27:
> So probably now the `setup.cfg` are ignored. I wonder how this is handled by humble's official apt packages, that anyhow are installed via Python 3.10 on Ubuntu 22.04 .

Interestingly, that issue only refers to `distutils.cfg`, so I am not 100% sure how it is related to our issue. I also tested `ros2 run demo_nodes_py demo_talker` installed on Ubuntu 22.04 with ROS2 Humble official apt packages and it is working fine.

traversaro, 2023-01-27:
However, using pure python apt on a Ubuntu 22.04 with ROS2 Humble results in the scripts being installed in `<install_prefix>/local/bin` instead of `<install_prefix>/lib/urdfdom_py/`:
~~~
traversaro@IITICUBLAP257:~/urdf_parser_py$ python -m pip install . -v  --prefix=./install
Using pip 22.0.2 from /usr/lib/python3/dist-packages/pip (python 3.10)
Processing /home/traversaro/urdf_parser_py
  Running command python setup.py egg_info
  running egg_info
  creating /tmp/pip-pip-egg-info-1k3_opic/urdfdom_py.egg-info
  writing /tmp/pip-pip-egg-info-1k3_opic/urdfdom_py.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-pip-egg-info-1k3_opic/urdfdom_py.egg-info/dependency_links.txt
  writing entry points to /tmp/pip-pip-egg-info-1k3_opic/urdfdom_py.egg-info/entry_points.txt
  writing requirements to /tmp/pip-pip-egg-info-1k3_opic/urdfdom_py.egg-info/requires.txt
  writing top-level names to /tmp/pip-pip-egg-info-1k3_opic/urdfdom_py.egg-info/top_level.txt
  writing manifest file '/tmp/pip-pip-egg-info-1k3_opic/urdfdom_py.egg-info/SOURCES.txt'
  reading manifest file '/tmp/pip-pip-egg-info-1k3_opic/urdfdom_py.egg-info/SOURCES.txt'
  writing manifest file '/tmp/pip-pip-egg-info-1k3_opic/urdfdom_py.egg-info/SOURCES.txt'
  Preparing metadata (setup.py) ... done
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from urdfdom-py==1.2.0) (59.6.0)
Building wheels for collected packages: urdfdom-py
  Running command python setup.py bdist_wheel
  running bdist_wheel
  running build
  running build_py
  /usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
    warnings.warn(
  installing to build/bdist.linux-x86_64/wheel
  running install
  running install_lib
  creating build/bdist.linux-x86_64/wheel
  creating build/bdist.linux-x86_64/wheel/urdf_parser_py
  copying build/lib/urdf_parser_py/sdf.py -> build/bdist.linux-x86_64/wheel/urdf_parser_py
  copying build/lib/urdf_parser_py/urdf.py -> build/bdist.linux-x86_64/wheel/urdf_parser_py
  copying build/lib/urdf_parser_py/display_urdf.py -> build/bdist.linux-x86_64/wheel/urdf_parser_py
  creating build/bdist.linux-x86_64/wheel/urdf_parser_py/xml_reflection
  copying build/lib/urdf_parser_py/xml_reflection/core.py -> build/bdist.linux-x86_64/wheel/urdf_parser_py/xml_reflection
  copying build/lib/urdf_parser_py/xml_reflection/basics.py -> build/bdist.linux-x86_64/wheel/urdf_parser_py/xml_reflection
  copying build/lib/urdf_parser_py/xml_reflection/__init__.py -> build/bdist.linux-x86_64/wheel/urdf_parser_py/xml_reflection
  copying build/lib/urdf_parser_py/__init__.py -> build/bdist.linux-x86_64/wheel/urdf_parser_py
  running install_data
  creating build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.data
  creating build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.data/data
  creating build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.data/data/share
  creating build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.data/data/share/ament_index
  creating build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.data/data/share/ament_index/resource_index
  creating build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.data/data/share/ament_index/resource_index/packages
  copying resource/urdfdom_py -> build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.data/data/share/ament_index/resource_index/packages
  creating build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.data/data/share/urdfdom_py
  copying package.xml -> build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.data/data/share/urdfdom_py
  running install_egg_info
  running egg_info
  writing src/urdfdom_py.egg-info/PKG-INFO
  writing dependency_links to src/urdfdom_py.egg-info/dependency_links.txt
  writing entry points to src/urdfdom_py.egg-info/entry_points.txt
  writing requirements to src/urdfdom_py.egg-info/requires.txt
  writing top-level names to src/urdfdom_py.egg-info/top_level.txt
  reading manifest file 'src/urdfdom_py.egg-info/SOURCES.txt'
  writing manifest file 'src/urdfdom_py.egg-info/SOURCES.txt'
  Copying src/urdfdom_py.egg-info to build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.egg-info
  running install_scripts
  creating build/bdist.linux-x86_64/wheel/urdfdom_py-1.2.0.dist-info/WHEEL
  creating '/tmp/pip-wheel-aet5_po1/urdfdom_py-1.2.0-py3-none-any.whl' and adding 'build/bdist.linux-x86_64/wheel' to it
  adding 'urdf_parser_py/__init__.py'
  adding 'urdf_parser_py/display_urdf.py'
  adding 'urdf_parser_py/sdf.py'
  adding 'urdf_parser_py/urdf.py'
  adding 'urdf_parser_py/xml_reflection/__init__.py'
  adding 'urdf_parser_py/xml_reflection/basics.py'
  adding 'urdf_parser_py/xml_reflection/core.py'
  adding 'urdfdom_py-1.2.0.data/data/share/ament_index/resource_index/packages/urdfdom_py'
  adding 'urdfdom_py-1.2.0.data/data/share/urdfdom_py/package.xml'
  adding 'urdfdom_py-1.2.0.dist-info/METADATA'
  adding 'urdfdom_py-1.2.0.dist-info/WHEEL'
  adding 'urdfdom_py-1.2.0.dist-info/entry_points.txt'
  adding 'urdfdom_py-1.2.0.dist-info/top_level.txt'
  adding 'urdfdom_py-1.2.0.dist-info/zip-safe'
  adding 'urdfdom_py-1.2.0.dist-info/RECORD'
  removing build/bdist.linux-x86_64/wheel
  Building wheel for urdfdom-py (setup.py) ... done
  Created wheel for urdfdom-py: filename=urdfdom_py-1.2.0-py3-none-any.whl size=15991 sha256=cb115c64a233776005373a28dc6de48c6b9dc806494954d782df375b67e67504
  Stored in directory: /tmp/pip-ephem-wheel-cache-g3fi39eu/wheels/e2/91/e9/6c87018216e4a8e53b53b8ffda1f6b738e989c30435075cbda
Successfully built urdfdom-py
Installing collected packages: urdfdom-py
  changing mode of install/local/bin/display_urdf to 755
Successfully installed urdfdom-py-1.2.0
~~~

traversaro, 2023-01-27:
Installing everything via colcon instead works fine:
~~~
traversaro@IITICUBLAP257:~/testurdf$ colcon build
Starting >>> urdfdom_py
--- stderr: urdfdom_py
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< urdfdom_py [0.99s]

Summary: 1 package finished [1.16s]
  1 package had stderr output: urdfdom_py
traversaro@IITICUBLAP257:~/testurdf$ cd install/
traversaro@IITICUBLAP257:~/testurdf/install$ find . | grep displa
./urdfdom_py/lib/python3.10/site-packages/urdf_parser_py/__pycache__/display_urdf.cpython-310.pyc
./urdfdom_py/lib/python3.10/site-packages/urdf_parser_py/display_urdf.py
./urdfdom_py/lib/urdfdom_py/display_urdf
~~~

Given the warning, it seems that `python setup.py` is being used in colcon, and indeed this is the case by printing the `command.log` file:

~~~
traversaro@IITICUBLAP257:~/testurdf$ cat ./log/build_2023-01-27_10-11-22/urdfdom_py/command.log
Invoking command in '/home/traversaro/testurdf/src/urdf_parser_py': PYTHONPATH=/home/traversaro/testurdf/build/urdfdom_py/prefix_override:/home/traversaro/testurdf/install/urdfdom_py/lib/python3.10/site-packages:${PYTHONPATH} /usr/bin/python3 setup.py egg_info --egg-base ../../build/urdfdom_py build --build-base /home/traversaro/testurdf/build/urdfdom_py/build install --record /home/traversaro/testurdf/build/urdfdom_py/install.log --single-version-externally-managed
Invoked command in '/home/traversaro/testurdf/src/urdf_parser_py' returned '0': PYTHONPATH=/home/traversaro/testurdf/build/urdfdom_py/prefix_override:/home/traversaro/testurdf/install/urdfdom_py/lib/python3.10/site-packages:${PYTHONPATH} /usr/bin/python3 setup.py egg_info --egg-base ../../build/urdfdom_py build --build-base /home/traversaro/testurdf/build/urdfdom_py/build install --record /home/traversaro/testurdf/build/urdfdom_py/install.log --single-version-externally-managed
~~~

traversaro, 2023-01-27:
Indeed, the code related to building python repos in colcon is https://github.com/colcon/colcon-core/blob/0.11.0/colcon_core/task/python/build.py#L51 , and it is possible to see that `python setup.py` is still used, and its removal is being tracked in https://github.com/colcon/colcon-core/issues/454 .

Tobias-Fischer, 2023-01-27:
Thanks for the debugging, @traversaro! The simplest solution seems to mirror https://github.com/RoboStack/vinca/blob/master/vinca/templates/bld_ament_python.bat.in to https://github.com/RoboStack/vinca/blob/master/vinca/templates/build_ament_python.sh.in?

traversaro, 2023-01-27:
> Thanks for the debugging, @traversaro! The simplest solution seems to mirror https://github.com/RoboStack/vinca/blob/master/vinca/templates/bld_ament_python.bat.in to https://github.com/RoboStack/vinca/blob/master/vinca/templates/build_ament_python.sh.in?

That could be an idea! For some reason, when testing that command locally I had some problem as the library was installed as an egg archive, but I do not know if there is something at the conda level to deal with it. Anyhow, given that it works for Windows we can try to use it and see if it works (I did not try to modify vinca and re-build the packages at the moment).

traversaro, 2023-01-27:
I also had problems in while using that command, as for some reason a `setuptools` was installed with the package. I avoid that by passing also the commands `--single-version-externally-managed --record ./installed_files`.

Tobias-Fischer, 2023-01-27:
Ok great. Do you happen to know if we also need to check this logic on Unix? https://github.com/RoboStack/vinca/blob/3b26d3ff52dd2c1108b3bef90e9bb0a08afd4cf4/vinca/templates/bld_ament_python.bat.in#L9-L14

Tobias-Fischer, 2023-02-04:
Reopening until packages are rebuilt.

Tobias-Fischer, 2023-02-09:
Fixed in the last rebuild :)

closed 2023-02-09 by Tobias-Fischer

Tobias-Fischer, 2024-02-23:
Can you please create a new issue and provide the requested information @TSoli?