RoboStack/ros-humble #44: Upgrading ament-cmake packages to 1.3.3

olivier-stasse, 2023-02-04:
Dear all, I ran into a problem while creating msgs in a conda environment using robostack/ros-humble.

Everything is fine for the creation phase but when reusing the msg I run into a problem similar to this one: https://github.com/ros-planning/moveit2/issues/1570

This was fixed with https://github.com/ament/ament_cmake/pull/395

It was backported in Aug, 11 2022.
But looking at https://anaconda.org/robostack-humble/ros-humble-ament-cmake-export-targets the current version in conda-force is 1.3.2 which was the release in May 2022.

The latest relesase 1.3.3 was done in Nov 2022 and therefore include the patch.
Would it be possible to have a the whole set of ament-cmake packages updated to 1.3.3 ?

Tobias-Fischer, 2023-02-04:
Hi @olivier-stasse thanks for getting in touch! We have new packages in the robostack-staging channel, hopefully this solves your issues :)

Tobias-Fischer, 2023-02-04:
Feel free to re-open if you still have issues.

closed 2023-02-04 by Tobias-Fischer

olivier-stasse, 2023-02-04:
Works like a charm ! Thanks.