RoboStack/ros-humble #46: library not found for -lrclcpp

baixianger, 2023-02-09:
Where can I find the right place for `rclcpp` library?

I want to add the right directory to LDLIBS variable of a makefile. Where is it?

Tobias-Fischer, 2023-02-09:
It's in `$CONDA_PREFIX/lib`

closed 2023-02-09 by Tobias-Fischer

baixianger, 2023-02-10:
Thank you for being so helpful, and I found it, but a new problem occurred.
My environment, `Python 3.10.9`, `Ventura 13.1`, `M2 chip`
I write a test cpp code for ROS2, 
```
#include "rclcpp/rclcpp.hpp"

int main(int argc, char **argv)
{
    // 调用rclcpp的初始化函数
    rclcpp::init(argc, argv);
    // 调用rclcpp的循环运行我们创建的first_node节点
    rclcpp::spin(std::make_shared<rclcpp::Node>("first_node"));
    return 0;
}
```
But when I compiled it, the compiler gave me such errors.
```
Undefined symbols for architecture arm64:
  "__ZN6rclcpp4NodeC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_11NodeOptionsE", referenced from:
      __ZSt10_ConstructIN6rclcpp4NodeEJRA11_KcEEvPT_DpOT0_ in first_node.o
  "__ZN6rclcpp4spinESt10shared_ptrINS_4NodeEE", referenced from:
      _main in first_node.o
  "__ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7_M_dataEv", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructIPKcEEvT_S8_St20forward_iterator_tag in first_node.o
  "__ZNSaIcEC1Ev", referenced from:
      __ZSt10_ConstructIN6rclcpp4NodeEJRA11_KcEEvPT_DpOT0_ in first_node.o
  "__ZNSaIcED1Ev", referenced from:
      __ZSt10_ConstructIN6rclcpp4NodeEJRA11_KcEEvPT_DpOT0_ in first_node.o
  "__ZNSaIcED2Ev", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_Alloc_hiderD1Ev in first_node.o
  "__ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_disposeEv", referenced from:
      __ZZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructIPKcEEvT_S8_St20forward_iterator_tagEN6_GuardD1Ev in first_node.o
  "__ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE11_M_capacityEm", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructIPKcEEvT_S8_St20forward_iterator_tag in first_node.o
  "__ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_Alloc_hiderC1EPcRKS3_", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC1IS3_EEPKcRKS3_ in first_node.o
  "__ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13_M_local_dataEv", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC1IS3_EEPKcRKS3_ in first_node.o
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructIPKcEEvT_S8_St20forward_iterator_tag in first_node.o
  "__ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13_M_set_lengthEm", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructIPKcEEvT_S8_St20forward_iterator_tag in first_node.o
  "__ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13_S_copy_charsEPcPKcS7_", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructIPKcEEvT_S8_St20forward_iterator_tag in first_node.o
  "__ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7_M_dataEPc", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructIPKcEEvT_S8_St20forward_iterator_tag in first_node.o
  "__ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructIPKcEEvT_S8_St20forward_iterator_tag in first_node.o
  "__ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev", referenced from:
      __ZN14rcl_interfaces3msg15ParameterValue_ISaIvEED1Ev in first_node.o
      __ZN6rclcpp9ParameterD1Ev in first_node.o
      __ZN6rclcpp20QosOverridingOptionsD1Ev in first_node.o
      __ZSt8_DestroyINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEvPT_ in first_node.o
      __ZSt10_ConstructIN6rclcpp4NodeEJRA11_KcEEvPT_DpOT0_ in first_node.o
  "__ZNSt8ios_base4InitC1Ev", referenced from:
      __Z41__static_initialization_and_destruction_0ii in first_node.o
  "__ZNSt8ios_base4InitD1Ev", referenced from:
      __Z41__static_initialization_and_destruction_0ii in first_node.o
  "__ZSt17__throw_bad_allocv", referenced from:
      __ZNSt15__new_allocatorISt23_Sp_counted_ptr_inplaceIN6rclcpp4NodeESaIvELN9__gnu_cxx12_Lock_policyE2EEE8allocateEmPKv in first_node.o
  "__ZSt19__throw_logic_errorPKc", referenced from:
      __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC1IS3_EEPKcRKS3_ in first_node.o
  "__ZSt28__throw_bad_array_new_lengthv", referenced from:
      __ZNSt15__new_allocatorISt23_Sp_counted_ptr_inplaceIN6rclcpp4NodeESaIvELN9__gnu_cxx12_Lock_policyE2EEE8allocateEmPKv in first_node.o
  "__ZTVN10__cxxabiv117__class_type_infoE", referenced from:
      __ZTIN6rclcpp11NodeOptionsE in first_node.o
      __ZTISt19_Sp_make_shared_tag in first_node.o
      __ZTISt11_Mutex_baseILN9__gnu_cxx12_Lock_policyE2EE in first_node.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
  "__ZTVN10__cxxabiv120__si_class_type_infoE", referenced from:
      __ZTISt23_Sp_counted_ptr_inplaceIN6rclcpp4NodeESaIvELN9__gnu_cxx12_Lock_policyE2EE in first_node.o
      __ZTISt16_Sp_counted_baseILN9__gnu_cxx12_Lock_policyE2EE in first_node.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
  "__ZdlPvm", referenced from:
      __ZNSt15__new_allocatorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEE10deallocateEPS5_m in first_node.o
      __ZNSt15__new_allocatorImE10deallocateEPmm in first_node.o
      __ZNSt15__new_allocatorIhE10deallocateEPhm in first_node.o
      __ZNSt15__new_allocatorIxE10deallocateEPxm in first_node.o
[... cut ...]
      __ZNSt15__new_allocatorISt23_Sp_counted_ptr_inplaceIN6rclcpp4NodeESaIvELN9__gnu_cxx12_Lock_policyE2EEE8allocateEmPKv in first_node.o
     (maybe you meant: __ZnwmPv)
  "___cxa_pure_virtual", referenced from:
      __ZTVSt16_Sp_counted_baseILN9__gnu_cxx12_Lock_policyE2EE in first_node.o
  "___gxx_personality_v0", referenced from:
      Dwarf Exception Unwind Info (__eh_frame) in first_node.o
ld: symbol(s) not found for architecture arm64
collect2: error: ld returned 1 exit status
make: *** [first_node] Error 1
```

baixianger, 2023-02-10:
This is my `makefile`:
```
CXX = gcc-12 # instead of the clang apple provided
CXXFLAGS = -g -Wall
INCLUDES = -I/Users/baixianger/miniconda3/envs/ros_env/include/rclcpp \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/rcl \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/rcutils \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/rmw \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/rcl_yaml_param_parser \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/rosidl_runtime_c \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/rosidl_typesupport_interface \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/rcpputils \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/builtin_interfaces \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/rosidl_runtime_cpp \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/tracetools \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/rcl_interfaces \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/libstatistics_collector \
           -I/Users/baixianger/miniconda3/envs/ros_env/include/statistics_msgs

LDFLAGS = -L/Users/baixianger/miniconda3/envs/ros_humble/lib
LDLIBS = -lrclcpp -lrcutils

Target_arch = -arch aarch64
LINK.cc = $(CXX) $(CXXFLAGS) $(TARGET_ARCH)

# 指定源文件和目标文件
SOURCES := $(wildcard *.cpp)
TARGET := $(SOURCES:.cpp=)
OBJECTS = $(SOURCES:.cpp=.o)

$(TARGET) : $(OBJECTS)
	$(LINK.cc) -o $(TARGET) $(OBJECTS) $(LDLIBS) $(LDFLAGS)

$(OBJECTS) : $(SOURCES)
	$(LINK.cc) -c $(SOURCES) $(INCLUDES)

clean :
	@rm $(OBJECTS)

deepclean: clean
	@rm $(TARGET)
```

Tobias-Fischer, 2023-02-10:
This is most certainly an issue with your Makefile - sorry but I can't provide support on that. Why don't you use the standard approach of ROS2 and use CMakeLists.txt etc. as per the tutorials? As per readme, please `mamba install compilers` and then use the compiler (+flags) that is automatically detected by cmake.

baixianger, 2023-02-10:
OK, i will try it later.

baixianger, 2023-02-10:
I followed your instruction and wrote this `cmakelists.txt`; it works well. 
I should not write `makefile` directly, which is not a good choice.
```
cmake_minimum_required(VERSION 3.22)
project(first_node)
find_package(rclcpp REQUIRED)
add_executable(first_node first_node.cpp)
target_link_libraries(first_node rclcpp::rclcpp)
```