RoboStack/ros-humble #47: No module named 'rclpy._rclpy_pybind11' on Apple M2 chip

baixianger, 2023-02-09:
I try to run the code below,
```
import rclpy
from rclpy.node import Node
rclpy.init() 
rclpy.spin(Node("second_node"))
```
But those errors occurred,
```
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension '/Users/baixianger/miniconda3/envs/ros_humble/lib/python3.10/site-packages/_rclpy_pybind11.cpython-311-darwin.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/humble/Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
```

Tobias-Fischer, 2023-02-09:
Hi, can you please provide the output of `mamba list`? This works without issues on my M1.

baixianger, 2023-02-10:
```
aiohttp
aiosignal
anyio
aom
appnope
argcomplete
argon2-cffi
argon2-cffi-bindings
arrow
assimp
asttokens
async-timeout
atk-1.0
attrs
backcall
beautifulsoup4
bleach
boost
boost-cpp
bqplot
brotli
brotli-bin
bullet
bullet-cpp
bzip2
c-ares
ca-certificates
cairo
catkin_pkg
certifi
cffi
charset-normalizer
cmake
colorama
comm
console_bridge
contourpy
cppcheck
cryptography
curl
cycler
debugpy
decorator
defusedxml
distro
docutils
double-conversion
eigen
empy
exceptiongroup
executing
expat
fastjsonschema
ffmpeg
flake8
flann
fmt
font-ttf-dejavu-sans-mono
font-ttf-inconsolata
font-ttf-source-code-pro
font-ttf-ubuntu
fontconfig
fonts-conda-ecosystem
fonts-conda-forge
fonttools
foonathan-memory
fqdn
freeimage
freetype
[... 529 lines cut ...]
xorg-libice
xorg-libsm
xorg-libx11
xorg-libxau
xorg-libxaw
xorg-libxdmcp
xorg-libxext
xorg-libxmu
xorg-libxpm
xorg-libxrandr
xorg-libxrender
xorg-libxt
xorg-randrproto
xorg-renderproto
xorg-xextproto
xorg-xproto
xz
yaml
yaml-cpp
yarl
zipp
zlib
zstd
zziplib
```

Tobias-Fischer, 2023-02-10:
Please include the full output including version number and channel information etc.

baixianger, 2023-02-10:
> Please include the full output including version number and channel information etc.

```
appnope                   0.1.3                    pypi_0    pypi
argcomplete               1.12.3             pyhd8ed1ab_0    conda-forge
argon2-cffi               21.3.0                   pypi_0    pypi
argon2-cffi-bindings      21.2.0                   pypi_0    pypi
arrow                     1.2.3                    pypi_0    pypi
assimp                    5.2.5                h276577b_0    conda-forge
asttokens                 2.2.1                    pypi_0    pypi
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
backcall                  0.2.0                    pypi_0    pypi
beautifulsoup4            4.11.2                   pypi_0    pypi
bleach                    6.0.0                    pypi_0    pypi
boost                     1.78.0          py310h629746b_4    conda-forge
boost-cpp                 1.78.0               h1cb353e_1    conda-forge
bqplot                    0.12.36                  pypi_0    pypi
brotli                    1.0.9                h1a8c8d9_8    conda-forge
brotli-bin                1.0.9                h1a8c8d9_8    conda-forge
bullet                    3.21                 h69fb684_4    conda-forge
bullet-cpp                3.21            py310h2b830bf_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.18.1               h3422bc3_0    conda-forge
ca-certificates           2022.12.7            h4653dfc_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h2399d43_3    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
cmake                     3.25.2               hf234bd0_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
comm                      0.1.2                    pypi_0    pypi
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.0.7           py310h2887b22_0    conda-forge
cppcheck                  2.7.5           py310hf3ebaa5_1    conda-forge
cryptography              39.0.1          py310hfc83b78_0    conda-forge
curl                      7.87.0               h9049daf_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
debugpy                   1.6.6                    pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
defusedxml                0.7.1                    pypi_0    pypi
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19            py310hbe9552e_1    conda-forge
double-conversion         3.2.0                hb7217d7_1    conda-forge
eigen                     3.4.0                hc021e02_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.1.0              pyhd8ed1ab_0    conda-forge
executing                 1.2.0                    pypi_0    pypi
expat                     2.5.0                hb7217d7_0    conda-forge
fastjsonschema            2.16.2                   pypi_0    pypi
ffmpeg                    5.1.2           gpl_hf318d42_106    conda-forge
flake8                    6.0.0              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             hd3e9afc_1011    conda-forge
fmt                       9.1.0                hffc8910_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h82840c6_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0          py310h8e9501a_1    conda-forge
foonathan-memory          0.7.2                hb7217d7_1    conda-forge
fqdn                      1.5.1                    pypi_0    pypi
freeimage                 3.18.0              hb513136_10    conda-forge
freetype                  2.12.1               hd633e50_1    conda-forge
fribidi                   1.0.10               h27ca646_0    conda-forge
frozenlist                1.3.3           py310h8e9501a_0    conda-forge
[... 530 lines cut ...]
xorg-libxau               1.0.9                h27ca646_0    conda-forge
xorg-libxaw               1.0.14               h3422bc3_1    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
xorg-libxext              1.3.4                h27ca646_1    conda-forge
xorg-libxmu               1.1.3                h3422bc3_0    conda-forge
xorg-libxpm               3.5.13               h3422bc3_0    conda-forge
xorg-libxrandr            1.5.2                h3422bc3_1    conda-forge
xorg-libxrender           0.9.10            h27ca646_1003    conda-forge
[... cut ...]
yaml-cpp                  0.7.0                hb7217d7_2    conda-forge
yarl                      1.8.2           py310h8e9501a_0    conda-forge
zipp                      3.13.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h03a7124_4    conda-forge
zstd                      1.5.2                hf913c23_6    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

Thank you, sir.
I re-installed the mamba env and it works this time.

Tobias-Fischer, 2023-02-10:
Ok thanks

closed 2023-02-10 by Tobias-Fischer