RoboStack/ros-humble #50: rosdep init failed in MacOS

andersdu, 2023-02-27:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

rosdep init failed after following the instruction in MacOS.

```
ERROR: cannot create /etc/ros/rosdep/sources.list.d:
	[Errno 13] Permission denied: '/etc/ros'
```

### Installed packages

```shell
# packages in environment at /Users/andersdu/miniconda3/envs/ros_humble:
#
# Name                    Version                   Build  Channel
aiohttp                   3.8.4           py310h90acd4f_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                hf0c8a7f_0    conda-forge
argcomplete               1.12.3             pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hd9e13b6_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
boost                     1.78.0          py310h3e792ce_4    conda-forge
boost-cpp                 1.78.0               h8b082ac_1    conda-forge
brotli                    1.0.9                hb7f2c08_8    conda-forge
brotli-bin                1.0.9                hb7f2c08_8    conda-forge
bullet                    3.21                 ha188af9_4    conda-forge
bullet-cpp                3.21                 hcd8b382_4    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2022.12.7            h033912b_0    conda-forge
cairo                     1.16.0            h904041c_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310ha78151a_3    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
cmake                     3.25.2               h4032537_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
contourpy                 1.0.7           py310ha23aa8a_0    conda-forge
cppcheck                  2.10.2          py310h944fa21_0    conda-forge
cryptography              39.0.1          py310hdd0c95c_0    conda-forge
curl                      7.88.1               h6df9250_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19            py310h2ec42d9_1    conda-forge
double-conversion         3.2.0                hf0c8a7f_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.1.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hf0c8a7f_0    conda-forge
ffmpeg                    5.1.2           gpl_h8b4fe81_106    conda-forge
flake8                    6.0.0              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             h56de9e4_1011    conda-forge
fmt                       9.1.0                hb8565cd_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h5bb23bf_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0          py310h90acd4f_1    conda-forge
foonathan-memory          0.7.2                hf0c8a7f_1    conda-forge
freeimage                 3.18.0              haafd79f_10    conda-forge
[... 507 lines cut ...]
       user config file : /Users/andersdu/.condarc
 populated config files :
          conda version : 23.1.0
    conda-build version : not installed
         python version : 3.10.9.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=13.2.1=0
                          __unix=0=0
       base environment : /Users/andersdu/miniconda3  (writable)
      conda av data dir : /Users/andersdu/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/andersdu/miniconda3/pkgs
                          /Users/andersdu/.conda/pkgs
       envs directories : /Users/andersdu/miniconda3/envs
                          /Users/andersdu/.conda/envs
               platform : osx-64
             user-agent : conda/23.1.0 requests/2.28.1 CPython/3.10.9 Darwin/22.3.0 OSX/13.2.1
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-02-27:
mamba install rosdep should help

Tobias-Fischer, 2023-02-27:
Feel free to reopen.

closed 2023-02-27 by Tobias-Fischer

andersdu, 2023-02-28:
thanks @Tobias-Fischer .

I figured I missed this installation as it's actually failed:
```
Looking for: ['rosdep']

pkgs/main/noarch                                              No change
pkgs/r/noarch                                                 No change
pkgs/main/osx-64                                              No change
pkgs/r/osx-64                                                 No change

Pinned packages:
  - python 3.10.*

Could not solve for environment specs
Encountered problems while solving:
  - nothing provides requested rosdep

The environment can't be solved, aborting the operation
```

Could you help with this as well? thanks.

Tobias-Fischer, 2023-02-28:
You are looking in the wrong channel - please use conda-forge (for everything, and please avoid mixing channels, see our README and paper)

andersdu, 2023-02-28:
thanks, I'll check.