RoboStack/ros-humble #53: Add ros-humble-turtlebot3 and ros-humble-turtlebot3-gazebo to more target platforms

nikolausWest, 2023-03-08:
First off, thanks for actually making it feasible to use ROS on a Mac! Really great job!

Gazebo and turtlebot3 are commonly used for introduction tutorials with ROS. We're putting together a tutorial on using ROS 2 (Humble) with [Rerun](https://github.com/rerun-io/rerun) and it would be super useful if it was possible to run it with other targets than linux-64.

traversaro, 2023-03-08:
Hello @nikolausWest, see https://github.com/RoboStack/ros-humble/blob/main/CONTRIBUTING.md for how to add packages to architectures in which they are not available. Once a PR is ready, the CI will check if the package build fine or if there are problems, thanks!

traversaro, 2023-03-08:
(And congrats on rerun, it seems quite cool!)

closed 2023-03-09 by Tobias-Fischer