RoboStack/ros-humble #59: OpenGL Error

BoyangZhangFromHKUST, 2023-04-09:
When running the command `colcon build` to build moveit workspace. There is a problem:

`CMake Error at /home/zby/miniforge3/envs/ros2_env/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenGL (missing: OPENGL_opengl_LIBRARY OPENGL_glx_LIBRARY)
Call Stack (most recent call first):
  /home/zby/miniforge3/envs/ros2_env/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /home/zby/miniforge3/envs/ros2_env/share/cmake-3.26/Modules/FindOpenGL.cmake:443 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:15 (find_package)
`

I have installed mesa already. 
Then I searched that I should install some packages like the below:
``mamba install -c conda-forge glfw`

`mamba install -c conda-forge mesa-libgl-devel-cos7-ppc64le``

But the problem still exists.

Tobias-Fischer, 2023-04-09:
Please use the issue template.

Tobias-Fischer, 2023-04-11:
Closing - feel free to reopen & fill the issue template.