RoboStack/ros-humble #62: Custom messages and rosidl-generator-c

b1n-ch1kn, 2023-05-06:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I'm using a Jetson Nano and robostack humble (py3.10) and have hit an issue with using custom messages. It seems to be having a problem with rosidl-generator-c.
The custom messages can built but can't be used in the python node - they crash when I try to run nodes that reference these messages in subscribers/publishers.
I also attempted to downgrade to an earlier version of rosidl-generator-c however there was only 3.1.4 available.

```
(humble_env) alastair@nano:~/QUTMS_Driverless$ ros2 run driverless_common display
Traceback (most recent call last):
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'driverless_msgs.driverless_msgs_s__rosidl_typesupport_c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/alastair/QUTMS_Driverless/install/driverless_common/lib/driverless_common/display", line 33, in <module>
    sys.exit(load_entry_point('driverless-common', 'console_scripts', 'display')())
  File "/home/alastair/QUTMS_Driverless/build/driverless_common/driverless_common/node_display.py", line 176, in main
    node = DisplayDetections()
  File "/home/alastair/QUTMS_Driverless/build/driverless_common/driverless_common/node_display.py", line 35, in __init__
    self.create_subscription(ConeDetectionStamped, "/vision/cone_detection", self.vision_callback, 1)
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rclpy/node.py", line 1367, in create_subscription
    check_is_valid_msg_type(msg_type)
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rclpy/type_support.py", line 35, in check_is_valid_msg_type
    check_for_type_support(msg_type)
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
    msg_or_srv_type.__class__.__import_type_support__()
  File "/home/alastair/QUTMS_Driverless/install/driverless_msgs/lib/python3.10/site-packages/driverless_msgs/msg/_cone_detection_stamped.py", line 29, in __import_type_support__
    module = import_type_support('driverless_msgs')
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'driverless_msgs'
[ros2run]: Process exited with failure 1
```

### Installed packages

```shell
# packages in environment at /home/alastair/mambaforge/envs/humble_env:
#
# Name                    Version                   Build  Channel
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-aarch64_curr_repodata_hack 4                   h57d6b7b_13    conda-forge
aiohttp                   3.8.4           py310hb89b984_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h4e544f5_0    conda-forge
aom                       3.5.0                headf329_0    conda-forge
argcomplete               3.0.8              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h3fd5656_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hf4e84e4_1    conda-forge
attr                      2.5.1                h4e544f5_1    conda-forge
attrs                     23.1.0             pyh71513ae_0    conda-forge
binutils                  2.39                 h64c2a2e_1    conda-forge
binutils_impl_linux-aarch64 2.39                 h48546ad_1    conda-forge
binutils_linux-aarch64    2.39                h489c705_13    conda-forge
boost                     1.78.0          py310hfd19f87_4    conda-forge
boost-cpp                 1.78.0               h484ac13_2    conda-forge
brotli                    1.0.9                h4e544f5_8    conda-forge
brotli-bin                1.0.9                h4e544f5_8    conda-forge
[... 639 lines cut ...]
          conda version : 22.9.0
    conda-build version : not installed
         python version : 3.10.6.final.0
       virtual packages : __cuda=10.2=0
                          __linux=4.9.299=0
                          __glibc=2.27=0
                          __unix=0=0
                          __archspec=1=aarch64
       base environment : /home/alastair/mambaforge  (writable)
      conda av data dir : /home/alastair/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-aarch64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-aarch64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/alastair/mambaforge/pkgs
                          /home/alastair/.conda/pkgs
       envs directories : /home/alastair/mambaforge/envs
                          /home/alastair/.conda/envs
               platform : linux-aarch64
             user-agent : conda/22.9.0 requests/2.28.1 CPython/3.10.6 Linux/4.9.299-tegra ubuntu/18.04.6 glibc/2.27
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-05-06:
Can you create a minimum example to replicate this issue, please? Does it happen on linux-64 as well?

b1n-ch1kn, 2023-05-06:
I've made a minimal node with one subscriber that uses one of the messages that failed:
https://github.com/QUT-Motorsport/QUTMS_Driverless/blob/jetson-humble-trt/src/common/minimal_debug/minimal_debug/test.py.

It works on linux-64 (Ubuntu 22.04):
```
(driverless_env) qev3d@Al-Desktop:~/QUTMS$ ros2 run minimal_debug test 
[INFO] [1683349847.394667060] [display_node]: ---Cone display node initialised---
```

But not on aarch-64:
```
(humble_env) alastair@nano:~/QUTMS_Driverless$ ros2 run minimal_debug test 
Traceback (most recent call last):
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'driverless_msgs.driverless_msgs_s__rosidl_typesupport_c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/alastair/QUTMS_Driverless/install/minimal_debug/lib/minimal_debug/test", line 33, in <module>
    sys.exit(load_entry_point('minimal-debug', 'console_scripts', 'test')())
  File "/home/alastair/QUTMS_Driverless/build/minimal_debug/minimal_debug/test.py", line 25, in main
    node = DisplayDetections()
  File "/home/alastair/QUTMS_Driverless/build/minimal_debug/minimal_debug/test.py", line 14, in __init__
    self.create_subscription(ConeDetectionStamped, "/vision/cone_detection", self.vision_callback, 1)
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rclpy/node.py", line 1367, in create_subscription
    check_is_valid_msg_type(msg_type)
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rclpy/type_support.py", line 35, in check_is_valid_msg_type
    check_for_type_support(msg_type)
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
    msg_or_srv_type.__class__.__import_type_support__()
  File "/home/alastair/QUTMS_Driverless/install/driverless_msgs/lib/python3.10/site-packages/driverless_msgs/msg/_cone_detection_stamped.py", line 29, in __import_type_support__
    module = import_type_support('driverless_msgs')
  File "/home/alastair/mambaforge/envs/humble_env/lib/python3.10/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'driverless_msgs'
[ros2run]: Process exited with failure 1
```

b1n-ch1kn, 2023-05-06:
### Linux-64 Mamba list:
```
# packages in environment at /home/qev3d/mambaforge/envs/driverless_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
abseil-cpp                20220623.0           h8cdb687_6    conda-forge
aiohttp                   3.8.4           py310h1fa729e_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
argcomplete               2.0.5              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
binutils                  2.39                 hdd6e379_1    conda-forge
binutils_impl_linux-64    2.39                 he00db2b_1    conda-forge
binutils_linux-64         2.39                h5fc0e48_11    conda-forge
blosc                     1.21.3               hafa529b_0    conda-forge
boost                     1.78.0          py310hc4a4660_4    conda-forge
boost-cpp                 1.78.0               h75c5d50_1    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
brotlipy                  0.7.0           py310h5764c6d_1005    conda-forge
build                     0.10.0                   pypi_0    pypi
bullet                    3.21                 hfdc917e_4    conda-forge
bullet-cpp                3.21                 h769672d_4    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
c-compiler                1.5.2                h0b41bf4_0    conda-forge
ca-certificates           2022.12.7            ha878542_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
cartographer              2.0.0           lua54h0ca914b_12    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
ceres-solver              2.1.0                hf302a74_1    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h255011f_3    conda-forge
cfitsio                   4.2.0                hd9d235c_0    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
chrpath                   0.16              h7f98852_1002    conda-forge
cmake                     3.21.3               h8897547_0    conda-forge
colcon-argcomplete        0.3.3                    pypi_0    pypi
colcon-bash               0.4.2                    pypi_0    pypi
colcon-cd                 0.1.1                    pypi_0    pypi
colcon-cmake              0.2.27                   pypi_0    pypi
colcon-common-extensions  0.3.0                    pypi_0    pypi
colcon-core               0.12.1                   pypi_0    pypi
colcon-defaults           0.2.8                    pypi_0    pypi
colcon-devtools           0.2.3                    pypi_0    pypi
colcon-library-path       0.2.1                    pypi_0    pypi
colcon-metadata           0.2.5                    pypi_0    pypi
colcon-notification       0.2.15                   pypi_0    pypi
colcon-output             0.2.13                   pypi_0    pypi
colcon-package-information 0.3.3                    pypi_0    pypi
colcon-package-selection  0.2.10                   pypi_0    pypi
colcon-parallel-executor  0.2.4                    pypi_0    pypi
colcon-pkg-config         0.1.0                    pypi_0    pypi
colcon-powershell         0.3.7                    pypi_0    pypi
colcon-python-setup-py    0.2.8                    pypi_0    pypi
colcon-recursive-crawl    0.2.1                    pypi_0    pypi
colcon-ros                0.3.23                   pypi_0    pypi
colcon-test-result        0.3.8                    pypi_0    pypi
colcon-zsh                0.4.0                    pypi_0    pypi
[... 781 lines cut ...]
          conda version : 22.11.1
    conda-build version : not installed
         python version : 3.10.8.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.1=0
                          __glibc=2.35=0
                          __linux=5.15.90.1=0
                          __unix=0=0
       base environment : /home/qev3d/mambaforge  (writable)
      conda av data dir : /home/qev3d/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
[... cut ...] onda-forge/noarch
          package cache : /home/qev3d/mambaforge/pkgs
                          /home/qev3d/.conda/pkgs
       envs directories : /home/qev3d/mambaforge/envs
                          /home/qev3d/.conda/envs
               platform : linux-64
             user-agent : conda/22.11.1 requests/2.28.2 CPython/3.10.8 Linux/5.15.90.1-microsoft-standard-WSL2 ubuntu/22.04.2 glibc/2.35
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-05-06:
Can you create a ws and zip it so I can replicate?

b1n-ch1kn, 2023-05-06:
I've got the message packages, minimal test, and environment yaml here:
[test_ws.zip](https://github.com/RoboStack/ros-humble/files/11411666/test_ws.zip)

Tobias-Fischer, 2023-05-06:
Is the jetson on the qut network by any chance? This would probably be the fastest way to debug if you grant me access.

b1n-ch1kn, 2023-05-06:
Ah unfortunately no, I'm testing on my personal jetson

Tobias-Fischer, 2023-05-08:
Hiya - I tried to replicate this issue this morning, but things work fine on my RaspBerry Pi.

Can you try whether it works with the attached environment file (rename to .yml)
[environment.txt](https://github.com/RoboStack/ros-humble/files/11417083/environment.txt)
? Also happy for you to grab my RPi if that's easier for your debugging.

Best, Tobi

Tobias-Fischer, 2023-05-08:
I noticed you have quite a few PyPi packages in your conda env, which is evil .. might cause this issue?

Tobias-Fischer, 2023-05-13:
Debugged this successfully offline.

closed 2023-05-13 by Tobias-Fischer