RoboStack/ros-humble #64: Problem building custom messages

james-p-foster, 2023-05-12:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I've been trying to convert pure message packages (only consisting of `.msg` files in the `/msg` dir) from ROS 1 to ROS 2 and have been running into multiple issues. To investigate, I made a minimal example:

package.xml:
```
<?xml version="1.0"?>
<package format="3">
  <name>blah_msgs</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="james.paul.foster@outlook.com">jfoster</maintainer>
  <license>TODO: License declaration</license>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <buildtool_depend>rosidl_default_generators</buildtool_depend>

  <exec_depend>rosidl_default_runtime</exec_depend>
  <member_of_group>rosidl_interface_packages</member_of_group>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
```

CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.8)
project(blah_msgs)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rosidl_default_generators REQUIRED)
rosidl_generate_interfaces(${PROJECT_NAME}
  "msg/DumbMessage.msg"
 )
ament_export_dependencies(rosidl_default_runtime)

ament_package()
```

My custom message, located in `msg/DumbMessage.msg`:
```
# This is a dumb message

float64 blam
float64 boom
```
When I run: `colcon build --packages-select blah_msgs --symlink-install`, I get:
```
Starting >>> blah_msgs
--- stderr: blah_msgs
Traceback (most recent call last):
  File "/home/jfoster/miniconda3/envs/ros2/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
  File "/home/jfoster/miniconda3/envs/ros2/lib/python3.10/site-packages/rosidl_generator_py/__init__.py", line 18, in <module>
    from .import_type_support_impl import import_type_support
  File "/home/jfoster/miniconda3/envs/ros2/lib/python3.10/site-packages/rosidl_generator_py/import_type_support_impl.py", line 17, in <module>
    from rpyutils import add_dll_directories_from_env
  File "/home/jfoster/miniconda3/envs/ros2/lib/python3.10/site-packages/rpyutils/__init__.py", line 16, in <module>
[... 682 lines cut ...]
          conda version : 22.9.0
    conda-build version : not installed
         python version : 3.7.10.final.0
       virtual packages : __cuda=11.8=0
                          __linux=5.4.0=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/jfoster/miniconda3  (writable)
      conda av data dir : /home/jfoster/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/jfoster/miniconda3/pkgs
                          /home/jfoster/.conda/pkgs
       envs directories : /home/jfoster/miniconda3/envs
                          /home/jfoster/.conda/envs
               platform : linux-64
             user-agent : conda/22.9.0 requests/2.29.0 CPython/3.7.10 Linux/5.4.0-148-generic ubuntu/20.04.5 glibc/2.31
                UID:GID : 1002:1002
             netrc file : /home/jfoster/.netrc
           offline mode : False
```

Tobias-Fischer, 2023-05-13:
Hi @b1n-ch1kn, this looks exactly like your problem where the wrong Python interpreter is picked up by cmake. Can you please write down the steps to fix it?

Basically, the issue seems to be with cmake. If you downgrade to cmake 3.18, delete all build files, and pass
https://github.com/RoboStack/vinca/blob/8be7ab86d554fa0c0e845f900b56b467e3092970/vinca/templates/build_catkin.sh.in#L72-L74
to your colcon call, it will work.

@traversaro - any clue what this could be?

Tobias-Fischer, 2023-05-13:
See https://colcon.readthedocs.io/en/released/reference/verb/build.html on how to pass cmake args.

traversaro, 2023-05-13:
See also https://github.com/RoboStack/ros-humble/issues/39#issuecomment-1399459318 . @james-p-foster can you are also:
* Output of `env` command (make sure not to include any sensitive env variable such as token or similar)
* Version of python in the system (i.e. outside the conda environement)?

Thanks!

b1n-ch1kn, 2023-05-13:
I also fixed my issue using the cmake args from https://github.com/RoboStack/ros-humble/issues/39#issuecomment-1399459318.
This is a little different to the issue I had on `aarch-linux64` where the message package was actually building successfully but not being imported correctly when implemented in another Python node

traversaro, 2023-05-13:
Probably as a first mitigation we could add https://github.com/RoboStack/ros-humble/issues/39#issuecomment-1399459318 to FAQs, I can also check if there is some way to influence FindPython's behaviour via env variables.

Tobias-Fischer, 2023-05-14:
Agreed! Could you please add instructions to https://github.com/RoboStack/robostack.github.io/pull/23 @traversaro? I’ll review and merge this next week.

traversaro, 2023-05-15:
> Agreed! Could you please add instructions to [RoboStack/robostack.github.io#23](https://github.com/RoboStack/robostack.github.io/pull/23) @traversaro? I’ll review and merge this next week.

Done: https://github.com/ruben-arts/robostack.github.io/pull/1 .

james-p-foster, 2023-05-16:
Thanks all for your help. As you suggest, CMake was finding the wrong python. I checked `CMakeCache.txt` and found that `PYTHON_EXECUTABLE` was my system python.

I'll note for any others arriving here however, that rather than doing e.g. `-DPYTHON_EXECUTABLE=$CONDA_PYTHON_EXE` in build arguments as suggested above, I instead chose to do `-DPYTHON_EXECUTABLE=/home/${USER}/miniconda3/envs/ros2/bin/python`, as I was working in a specific conda environment called `ros2`. The former instead gets you the python of the conda `base` environment (to my knowledge). I'll close this now.

Tobias-Fischer, 2023-05-16:
Good catch - `$CONDA_PYTHON_EXE` indeed points to the `base` env. Should we change it to `$PREFIX/bin/python`? Or is there a nicer way @traversaro @wolfv?

closed 2023-05-19 by Tobias-Fischer

traversaro, 2023-05-19:
> Good catch - `$CONDA_PYTHON_EXE` indeed points to the `base` env. Should we change it to `$PREFIX/bin/python`? Or is there a nicer way @traversaro @wolfv?

You are right, sorry! Indeed the correct way is  `$CONDA_PREFIX/bin/python` on Linux/macOS and `%CONDA_PREFIX%\python`  on Windows (`$PREFIX`  is defined just during conda-build I guess).