RoboStack/ros-humble #65: Does ros-humble support python 3.11 now?

songyuc, 2023-05-15:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi, guys!
Does ros-humble support python 3.11 now?

### Installed packages

```shell
Later.
```

### Environment info

```shell
Later.
```

Tobias-Fischer, 2023-05-15:
No - not yet. We'll consider it for the next rebuild (no ETA for that, though)

songyuc, 2023-05-15:
Thanks sincerely for the response!

songyuc, 2023-05-15:
Hi, @Tobias-Fischer!
Besides, is  vs2022_win-64 supported, currently?

Tobias-Fischer, 2023-05-15:
Not sure - @traversaro do you have any idea?

traversaro, 2023-05-15:
> Not sure - @traversaro do you have any idea?

Yes, all the packages have been built with VS2019 (if I recall correctly) and so you can use VS2022 downstream and link the libraries built with VS2019.

songyuc, 2023-05-22:
Could you please also provide this detailed information in the [doc](https://robostack.github.io/GettingStarted.html#:~:text=%23%20on%20Windows%2C%20install%20the%20Visual%20Studio%20command%20prompt%3A)?
Thanks very much!

Tobias-Fischer, 2023-05-22:
Can you please open a pull request @songyuc?

songyuc, 2023-05-22:
Hi, @Tobias-Fischer!
As a GitHub beginner, I want to try a PR of this doc.

Tobias-Fischer, 2023-05-24:
I've done this in https://github.com/RoboStack/robostack.github.io/pull/23

songyuc, 2023-05-29:
@Tobias-Fischer, thanks sincerely for your support!

songyuc, 2023-05-29:
Could you please add a requirement section in the doc, like:
> Requirements:
> python <= 3.10

Tobias-Fischer, 2023-05-29:
Please open a PR in https://github.com/RoboStack/robostack.github.io

closed 2023-05-29 by Tobias-Fischer

songyuc, 2023-07-04:
Hi, guys!
I am using Miniforge, but I found that the default command, such as `mamba create -n ros_humble`, creates a Python 3.11 environment. However, I remember it has been mentioned that ros-humble currently does not support Python 3.11. But the current official documentation uses the  default command, too, as [mamba create -n ros_env](https://robostack.github.io/GettingStarted.html#:~:text=mamba%20create%20%2Dn%20ros_env). I am wondering if this operation will bring some potential problems.

songyuc, 2023-07-04:
```bash
(base) C:\Users\songyucheng>conda info -e
# conda environments:
#
base                  *  C:\Users\songyucheng\mambaforge
mmcoursed3               C:\Users\songyucheng\mambaforge\envs\mmcoursed3
ros_env                  C:\Users\songyucheng\mambaforge\envs\ros_env

(base) C:\Users\songyucheng>mamba create -n ros_humble

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.4.1) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: []

Preparing transaction: done
Verifying transaction: done
Executing transaction: done

To activate this environment, use

     $ mamba activate ros_humble

To deactivate an active environment, use

     $ mamba deactivate

(base) C:\Users\songyucheng>mamba activate ros_humble

(ros_humble) C:\Users\songyucheng>python -V
Python 3.11.2

(ros_humble) C:\Users\songyucheng>
```

traversaro, 2023-07-04:
I think what you are finding when you execute the `python -V` is the `python` installed in the `base` environment. Once you do either `mamba install ros-noetic-desktop` or `mamba install ros-humble-desktop` a suitable python will be installed in the `ros_humble` environment.

songyuc, 2023-07-04:
```
(base) C:\Users\songyucheng>mamba activate ros_humble

(ros_humble) C:\Users\songyucheng>python -V
Python 3.11.2

(ros_humble) C:\Users\songyucheng>mamba deactivate

(base) C:\Users\songyucheng>python -V
Python 3.10.10

(base) C:\Users\songyucheng>
```
Hello, @traversaro, thank you for your reply and help. I checked the python version in the base environment and it was 3.10.10, while the previous output was 3.11.2, so the python version in the ros_humble environment is most likely 3.11.2.

songyuc, 2023-07-04:
Hello, @traversaro, thank you for your reply and help. I tested and found out that “Python 3.11.2” was actually the python environment that came with my Win11 system. I apologize for my lack of knowledge about python environments, and I appreciate your guidance very much.

songyuc, 2023-07-04:
I noticed that you mentioned that “a suitable python will be installed”. Does this mean that ros-humble does not support custom python version?

traversaro, 2023-07-04:
> I noticed that you mentioned that “a suitable python will be installed”. Does this mean that ros-humble does not support custom python version?

Yes, at the moment we are not building for multiple Python versions.

songyuc, 2023-07-05:
Thanks sincerely for @traversaro your response.