RoboStack/ros-humble #66: [Beginner Walkthrough] some questions about learning ROS2 on conda-ros-humble with ROS2 official Tutorials

songyuc, 2023-05-15:
Hi, guys!
I am a beginner at ROS2, and I plan to walk through the ROS2 official Tutorials with conda-ros-humble.
I hope to list my questions here and also write my tutorial blog for my teammates in the club of robots based on conda-ros-humble.
I don't know whether this is a good way to discuss it here.

Sincerely look forward to your guidance.

Tobias-Fischer, 2023-05-15:
Yes, here is a good place. It would be good to open separate GitHub issues for the problems you might encounter. Out of interest, which robotics club is it?

Tobias-Fischer, 2023-05-24:
As discussed, please open separate issues as they arise.

closed 2023-05-24 by Tobias-Fischer

songyuc, 2023-05-29:
Hi, guys!
The tutorial says: "As always, don’t forget to source ROS 2 in [every new terminal you open](https://docs.ros.org/en/humble/Tutorials/Beginner-CLI-Tools/Configuring-ROS2-Environment.html)."
According to my experience, the `conda activate ros_env` has contained the sourcing operation, am I right?

Tobias-Fischer, 2023-05-29:
Yes