RoboStack/ros-humble #68: Package request: ros-humble-ros-gz

icosac, 2023-05-23:
### Package name

ros-humble-ros-gz

### Operating System(s)

- [X] linux-64
- [X] linux-aarch64
- [X] osx-64
- [X] osx-arm64
- [X] win-64

### Additional context (optional)

Hi, I'm sorry if this is not the correct way of requesting the addition of a package, it's the first time I do this. 

I have to compile a ROS2 workspace that has as a dependency ros-humble-ros-gz (and also ros-humble-gazebo-ros2-control, but this I will request in another issue) and when running `rosdep`, it cannot find the package. 

Is it possible to add it to RoboStack? 

Thanks!

### Links and references (optional)

The repository of the ros-gz package can be found [here](https://index.ros.org/r/ros_gz/)

traversaro, 2023-05-23:
If you want to use gz-sim with robostack, pay attention that you may also hit the error: https://github.com/conda-forge/gz-sim-feedstock/issues/6 . The fix for that is available at https://github.com/gazebosim/gz-sim/pull/1968 . It is taking some time as it needs to be accepted upstream before being backported (as it introduces new ABI symbols), but hopefully we should be done with it soon.

icosac, 2023-05-23:
Thanks for pointing me to this issues! I will have them monitor and see if there is any progress.

mkarklins, 2023-06-05:
Awesome to see progress on this front!

It looks like I have some spare time available. Does it make sense to take a stab at adding the `ros-gz` package now before the `gz-sim` PR is merged? Since the `gz-sim*` things can be installed separately, relatively easy on mac with homebrew.

traversaro, 2023-06-05:
In general you can't mix homebrew and conda-forge packages, as they have different ABIs. Anyhow, even if https://github.com/conda-forge/gz-sim-feedstock/issues/6 is present, probably you can work around it by manually copying `$CONDA_PREFIX/share/gz/gz-sim7/gui/gui.config`  to `~/.gz/sim/7/gui.config` . Furthermore, as this problem is just at runtime, there should be no problem in already working on adding `ros-humble-ros-gz` to robostack, see https://github.com/RoboStack/ros-humble/blob/main/CONTRIBUTING.md#adding-new-packages-via-pull-requests for docs on that.

traversaro, 2023-06-05:
A more tricky issue is that apparently the Humble version of ros-gz packages depend on Gazebo (Ignition) Fortress (see https://github.com/gazebosim/ros_gz). However, all relocatability work is going on Gazebo Garden. While it may be possible to have some simpler workaround/fix to solve https://github.com/conda-forge/gz-sim-feedstock/issues/6, I am not sure if Gazebo Fortress packages work at all on macOS. Probably the easiest way to check is to first build from source ros-gz packages to check if they work.

mkarklins, 2023-06-05:
Is there an inherent requirement to support `Fortress` release ( a side from it being an LTS version )? It seems at least initially we can focus on `Garden`.

Tobias-Fischer, 2023-06-05:
Sounds good to me :)

traversaro, 2023-07-08:
> Thanks for pointing me to this issues! I will have them monitor and see if there is any progress.

For anyone just monitoring this issue and not that one, the problems on gz-sim on Linux were solved some weeks ago, see https://github.com/conda-forge/gz-sim-feedstock/issues/6#issuecomment-1599464051 .

traversaro, 2024-02-19:
> Hi @traversaro - I am a bit lost when it comes to Gazebo support for ROS2. I'm happy to look into it, but would appreciate a bit of guidance. What is the overall status of Classic-Gazebo and Ignition-Gazebo both standalone and in conjunction with RoboStack/ROS2? Which version of Classic-Gazebo / Ignition-Gazebo should we use?

Ok, let's try to recap. Sorry for the complexity, hopefully it will go away in the long term. In the following I will refer to "Classic gazebo" to refer to the "old" Gazebo that has been developed from ~2004 and will be not be supported upstream anymore in 2025, and with "Modern gz-sim" for the modern simulator (that used to be called "Ignition Gazebo") that is being actively developed and it is recommended for new developments. Unfortunately, both are being called "Gazebo", so I prefer to avoid using just "Gazebo" as it is ambiguous. It is important to distinguish the two, as even if they share the possibilities of loading models in URDF and SDF formats, the API for their plugins is completely different, so effectively two different simulators.

First of all, historically each version of ROS/ROS2 is associated to a corresponding release of both "Gazebo classic" and "Modern gz-sim". Using a different version may be working, but it is typically not supported in binary packages. For ROS2 Humble, the corresponding releases are:

| Gazebo version |  Version supported by ROS Humble | Package providing integration between ROS and Gazebo |  Documentation Link | OS supported |
|:--------------------:|:--------------------------:|:--------------------:|:--------------------:|:--------:|
| Gazebo Classic: | 11 | [`ros-humble-gazebo-ros-pkgs`](https://github.com/ros-simulation/gazebo_ros_pkgs/tree/ros2) | https://classic.gazebosim.org/tutorials?tut=ros2_overview | Linux, macOS and Windows. |
| Modern gz-sim: Gazebo Fortress | gz-sim6 (that is available in conda-forge as [libignition-gazebo6](https://github.com/conda-forge/libignition-gazebo-feedstock), as it was released before the Gazebo/Ignition renaming) . See https://github.com/gazebosim/ros_gz and https://github.com/gazebo-tooling/gazebodistro/blob/master/collection-fortress.yaml | [`ros-humble-ros-gz`](https://github.com/gazebosim/ros_gz) (it used to be called ros-ign, but it was retroactively renamed to ros-gz to simplify the life of users migrating to the future release of Modern gz-sim called "gz" instead of "ign"). | https://docs.ros.org/en/humble/Tutorials/Advanced/Simulators/Gazebo/Gazebo.html | Linux should work, I am not sure about macOS. For sure Windows it is not working, see https://github.com/gazebosim/gz-sim/issues/2089 |

Regarding robostack, it seems that `ros-humble-gazebo-ros-pkgs` was compiled recently, while in the latest rebuild, while `ros-humble-ros-gz` is missing, I will open a PR to build it.

Tobias-Fischer, 2024-02-19:
Many thanks @traversaro! I’ll reopen here to keep track of the Windows issue. I guess it would be resolved if we move to a newer ROS2 distribution (e.g. Iron) which supports a newer Modern go-sim?

traversaro, 2024-02-19:
> I guess it would be resolved if we move to a newer ROS2 distribution (e.g. Iron) which supports a newer Modern go-sim?

Kind of. The packages for gz-sim on Windows on conda-forge are available and basic simulations are working, but the GUI still does not work, but that's an upstream issue (see https://github.com/gazebosim/gz-sim/issues/2089).

Tobias-Fischer, 2026-10-18:
**Mostly covered** in the RoboStack monorepo, [RoboStack/robostack](https://github.com/RoboStack/robostack), where all ROS 2 distributions are now built: `ros_gz`, `ros_gz_bridge`, `ros_gz_image` and `ros_gz_interfaces` are now built for win-64. `ros-humble-ros-gz` for win-64 has been published since 2026-09-29, and the humble rebuild there ([RoboStack/robostack#62](https://github.com/RoboStack/robostack/pull/62)) updates it to 0.244.26. `ros_gz_sim` (and its demos) stays excluded on Windows, because Ignition Fortress (`libignition-gazebo6`) isn't installable on win-64 from conda-forge: its last win-64 build is 6.12.0 from 2023. That part would need fixing in conda-forge first.

🤖 Claude (AI assistant), on behalf of @Tobias-Fischer

closed 2026-10-18 by Tobias-Fischer