RoboStack/ros-humble #69: [ERROR:vcvars.bat] when installing vs2022_win-64

songyuc, 2023-05-29:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I saw an error message when installing vs2022_win-64 as shown in as https://gist.github.com/songyuc/dd00bfa9719f185e685e3d5c3d7ce55b

### Installed packages

```shell
(ros_env) C:\Users\songyucheng>mamba list
# packages in environment at C:\Users\songyucheng\mambaforge\envs\ros_env:
#
# Name                    Version                   Build  Channel
aiohttp                   3.8.4           py310h8d17308_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h63175ca_0    conda-forge
argcomplete               3.0.8              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
boost                     1.78.0          py310h220cb41_4    conda-forge
boost-cpp                 1.78.0               h9f4b32c_3    conda-forge
brotli                    1.0.9                hcfcfb64_8    conda-forge
brotli-bin                1.0.9                hcfcfb64_8    conda-forge
bullet                    3.24                 hbbfc1a7_0    conda-forge
bullet-cpp                3.24                 h1c4a608_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.5.2                hcfcfb64_0    conda-forge
ca-certificates           2023.5.7             h56e8100_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.8.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h628cb3f_3    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.26.4               h1537add_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py310h5588dad_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-notification       0.2.15          py310h5588dad_1    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compilers                 1.5.2                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.0.7           py310h232114e_0    conda-forge
coverage                  7.2.6           py310h8d17308_0    conda-forge
cppcheck                  2.10.3          py310hbf4f5aa_0    conda-forge
cryptography              40.0.2          py310h6e82f81_0    conda-forge
curl                      7.88.1               h68f0423_1    conda-forge
cxx-compiler              1.5.2                h91493d7_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.0                hcfcfb64_0    conda-forge
distlib                   0.3.6              pyhd8ed1ab_0    conda-forge
[... 543 lines cut ...]
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.10.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.1=0
                          __win=0=0
       base environment : C:\Users\songyucheng\mambaforge  (writable)
      conda av data dir : C:\Users\songyucheng\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\songyucheng\mambaforge\pkgs
                          C:\Users\songyucheng\.conda\pkgs
                          C:\Users\songyucheng\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\songyucheng\mambaforge\envs
                          C:\Users\songyucheng\.conda\envs
                          C:\Users\songyucheng\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.10 Windows/10 Windows/10.0.22621
          administrator : False
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-05-29:
Just checking that you have installed VS studio 2022?

songyuc, 2023-05-30:
Hi, @Tobias-Fischer.
Yes, I have installed it.
But I saw:
```
C:\Users\songyucheng>pushd C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\

C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools>CALL "VC\Auxiliary\Build\vcvars64.bat" -vcvars_ver=14.34 10.0.22621.0
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.6.2
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[ERROR:vcvars.bat] Toolset directory for version '14.34' was not found.
[ERROR:VsDevCmd.bat] *** VsDevCmd.bat encountered errors. Environment may be incomplete and/or incorrect. ***
[ERROR:VsDevCmd.bat] In an uninitialized command prompt, please 'set VSCMD_DEBUG=[value]' and then re-run
[ERROR:VsDevCmd.bat] vsdevcmd.bat [args] for additional details.
[ERROR:VsDevCmd.bat] Where [value] is:
[ERROR:VsDevCmd.bat]    1 : basic debug logging
[ERROR:VsDevCmd.bat]    2 : detailed debug logging
[ERROR:VsDevCmd.bat]    3 : trace level logging. Redirection of output to a file when using this level is recommended.
[ERROR:VsDevCmd.bat] Example: set VSCMD_DEBUG=3
[ERROR:VsDevCmd.bat]          vsdevcmd.bat > vsdevcmd.trace.txt 2>&1
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.6.2
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[ERROR:vcvars.bat] Version '14.29.30133' is not valid; directory does not exist
[ERROR:VsDevCmd.bat] *** VsDevCmd.bat encountered errors. Environment may be incomplete and/or incorrect. ***
[ERROR:VsDevCmd.bat] In an uninitialized command prompt, please 'set VSCMD_DEBUG=[value]' and then re-run
[ERROR:VsDevCmd.bat] vsdevcmd.bat [args] for additional details.
[ERROR:VsDevCmd.bat] Where [value] is:
[ERROR:VsDevCmd.bat]    1 : basic debug logging
[ERROR:VsDevCmd.bat]    2 : detailed debug logging
[ERROR:VsDevCmd.bat]    3 : trace level logging. Redirection of output to a file when using this level is recommended.
[ERROR:VsDevCmd.bat] Example: set VSCMD_DEBUG=3
[ERROR:VsDevCmd.bat]          vsdevcmd.bat > vsdevcmd.trace.txt 2>&1
```
I don't know how to solve it.

songyuc, 2023-05-30:
[Archive]:
According to the error "[ERROR:vcvars.bat] Toolset directory for version '14.34' was not found.", this error is most likely caused by the current system not having installed the build tools of version '14.34'.

closed 2023-05-30 by songyuc