RoboStack/ros-humble #70: no LC_RPATH load command

mrPuckk, 2023-06-05:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi there, 

I encountered an error while attempting to compile the project using the rclcpp package. 

### Preproduce: 

`colcon build --package-select my_package`

the output:
> Starting >>> my_package
--- stderr: my_package
error: /Users/touphung/mambaforge/envs/ros_env/bin/install_name_tool: no LC_RPATH load command with path: /Users/touphung/mambaforge/envs/ros_env/lib found in: /Users/touphung/myDev/ros2_ws/install/my_package/lib/my_package/simple_node (for architecture arm64), required for specified option "-delete_rpath /Users/touphung/mambaforge/envs/ros_env/lib"
--
Finished <<< my_package [0.25s]

`ros2 launch my_package my_package_launch_file.launch.py`

the output:
>  [INFO] [launch]: All log files can be found below /Users/touphung/.ros/log/2023-06-05-14-38-56-226404-QuangTuongs-MacBook-Pro.local-26932
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [simple_node-1]: process started with pid [26933]
[simple_node-1] dyld[26933]: Library not loaded: @rpath/librclcpp.dylib
[simple_node-1]   Referenced from: <C97886EB-2A80-38D9-B73E-BEB29FAF2FDE> /Users/touphung/myDev/ros2_ws/install/my_package/lib/my_package/simple_node
[simple_node-1]   Reason: tried: '/Users/touphung/mambaforge/envs/ros2/opt/rviz_ogre_vendor/lib/librclcpp.dylib' (no such file), '/usr/local/lib/librclcpp.dylib' (no such file), '/usr/lib/librclcpp.dylib' (no such file, not in dyld cache)
[ERROR] [simple_node-1]: process has died [pid 26933, exit code -6, cmd '/Users/touphung/myDev/ros2_ws/install/my_package/lib/my_package/simple_node --ros-args'].

I created the link from ~/mambaforge/envs/ros_env/lib to the '/usr/local/lib'
`sudo ln -s ~/mambaforge/envs/ros_env/lib /usr/local/`

Then compile, source and run build again, I got: 
> [INFO] [launch]: All log files can be found below /Users/touphung/.ros/log/2023-06-05-14-49-27-739340-QuangTuongs-MacBook-Pro.local-26989
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [simple_node-1]: process started with pid [26990]
[simple_node-1] dyld[26990]: symbol not found in flat namespace '_PyExc_RuntimeError'
[ERROR] [simple_node-1]: process has died [pid 26990, exit code -6, cmd '/Users/touphung/myDev/ros2_ws/install/my_package/lib/my_package/simple_node --ros-args'].

I was examining the CMakeLists.txt file and decided to remove a specific section: 
> install(TARGETS simple_node
DESTINATION lib/${PROJECT_NAME}
) 

The compilation process completed successfully without any errors. However, when I attempted to launch the build file, I encountered an error.  `[simple_node-1] dyld[26990]: symbol not found in flat namespace '_PyExc_RuntimeError'`

### Solution 
Not yet 

### Desire
Could you please provide more details about the specific error message or any relevant information related to the issue? This will help me understand the problem better. Thank you! 

 

### Installed packages

```shell
/Users/touphung/mambaforge/envs/ros_env/lib/python3.10/site-packages/conda_package_streaming/package_streaming.py:19: UserWarning: zstandard could not be imported. Running without .conda support.
  warnings.warn("zstandard could not be imported. Running without .conda support.")
/Users/touphung/mambaforge/envs/ros_env/lib/python3.10/site-packages/conda_package_handling/api.py:29: UserWarning: Install zstandard Python bindings for .conda support
  _warnings.warn("Install zstandard Python bindings for .conda support")
# packages in environment at /Users/touphung/mambaforge/envs/ros_env:
#
# Name                    Version                   Build  Channel
_r-mutex                  1.0.1               anacondar_1    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
argcomplete               1.12.3             pyhd3eb1b0_0    main
[... 762 lines cut ...]
       user config file : /Users/touphung/.condarc
 populated config files : /Users/touphung/mambaforge/envs/ros_env/.condarc
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.11.final.0
       virtual packages : __archspec=1=arm64
                          __osx=13.4=0
                          __unix=0=0
       base environment : /Users/touphung/mambaforge/envs/ros_env  (writable)
      conda av data dir : /Users/touphung/mambaforge/envs/ros_env/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/touphung/mambaforge/envs/ros_env/pkgs
                          /Users/touphung/.conda/pkgs
       envs directories : /Users/touphung/mambaforge/envs/ros_env/envs
                          /Users/touphung/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.11 Darwin/22.5.0 OSX/13.4
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2023-06-05:
I am not sure how it is possible as you are using mambaforge, but in your `mamba list` it is possible to see that you are mixing packages from the `main`  channel and the `conda-forge`  one, that is not supported. 

I suggest you to:
* Remove the symbolic link from `/usr/local`  to  `~/mambaforge/envs/ros_env/lib`, as it is not solving any problem and actual creates more
* Can you create a new clean environment following the instructions in https://robostack.github.io/GettingStarted.html 
* After you do so, you should do `mamba list` and check that all packages either come from `conda-forge`  or `robostack-staging` 

After you do this, I suggest to try again doing the compilation and check if you have still problems.

mrPuckk, 2023-06-05:
Thanks for your quick response. 

I followed your suggestion to create a new clean environment and made sure that there are no mixed packages from the main channel and other sources. However, the problem still persists with the error message: "Library not loaded: @rpath/librclcpp.dylib." when launching build file.

`colcon build --packages-select my_package`
> Starting >>> my_package
--- stderr: my_package
[cctools-port]: generating fake signature for '/Users/touphung/myDev/ros2_ws/install/my_package/lib/my_package/simple_node'
Finished <<< my_package [3.11s]
Summary: 1 package finished [3.34s]
  1 package had stderr output: my_package

`ros2 launch my_package my_package_launch_file.launch.py`

> [INFO] [launch]: All log files can be found below /Users/touphung/.ros/log/2023-06-06-09-42-28-682027-QuangTuongs-MacBook-Pro.local-50574
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [simple_node-1]: process started with pid [50575]
[simple_node-1] dyld[50575]: Library not loaded: @rpath/librclcpp.dylib
[simple_node-1]   Referenced from: <C97886EB-2A80-38D9-B73E-BEB29FAF2FDE> /Users/touphung/myDev/ros2_ws/install/my_package/lib/my_package/simple_node
[simple_node-1]   Reason: tried: '/Users/touphung/mambaforge/envs/ros_env/opt/rviz_ogre_vendor/lib/librclcpp.dylib' (no such file), '/usr/local/lib/librclcpp.dylib' (no such file), '/usr/lib/librclcpp.dylib' (no such file, not in dyld cache)
[ERROR] [simple_node-1]: process has died [pid 50575, exit code -6, cmd '/Users/touphung/myDev/ros2_ws/install/my_package/lib/my_package/simple_node --ros-args'].

The output from `mamba list`: 
>  packages in environment at /Users/touphung/mambaforge/envs/ros_env:
 Name                    Version                   Build  Channel
aiohttp                   3.8.4           py310h8e9501a_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
argcomplete               3.0.8              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
boost                     1.78.0          py310h629746b_4    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.0.9                h1a8c8d9_8    conda-forge
brotli-bin                1.0.9                h1a8c8d9_8    conda-forge
bullet                    3.24                 h69fb684_0    conda-forge
bullet-cpp                3.24            py310h2b830bf_0    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.1               hb547adb_0    conda-forge
c-compiler                1.5.2                h5008568_0    conda-forge
ca-certificates           2023.5.7             hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.3              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hcbb26d4_13    conda-forge
cctools_osx-arm64         973.0.1             hef52d2f_13    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h2399d43_3    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
clang                     14.0.6               hce30654_1    conda-forge
clang-14                  14.0.6          default_h5dc8d65_1    conda-forge
clang_osx-arm64           14.0.6               h15773ab_6    conda-forge
clangxx                   14.0.6          default_h610c423_1    conda-forge
clangxx_osx-arm64         14.0.6               he29aa18_6    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py310hbe9552e_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
[... 535 lines cut ...]
xorg-kbproto              1.0.7             h27ca646_1002    conda-forge
xorg-libice               1.0.10               h27ca646_0    conda-forge
xorg-libsm                1.2.3             h27ca646_1000    conda-forge
xorg-libx11               1.8.4                h1a8c8d9_0    conda-forge
xorg-libxau               1.0.11               hb547adb_0    conda-forge
[... cut ...]
yaml                      0.2.5                h3422bc3_2    conda-forge
yaml-cpp                  0.7.0                hb7217d7_2    conda-forge
yarl                      1.9.2           py310h2aa6e3c_0    conda-forge
zipp                      3.15.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h03a7124_4    conda-forge
zstd                      1.5.2                hf913c23_6    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge