RoboStack/ros-humble #76: ros-humble-vision-msgs  can not be installed on mac m1

yrik, 2023-06-14:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hello, I saw that `ros-humble-vision-msgs` should be supported on osx arm64 but when try to install it could not be found. Awesome work with the project, hope to have my ros setup working with it.

```
(ros_env) yura@Yuras-MacBook-Pro ros_ws % micromamba install ros-humble-vision-msgs       

Pinned packages:
  - python 3.10.*

error    libmamba Could not solve for environment specs
    The following package could not be installed
    └─ ros-humble-vision-msgs   does not exist (perhaps a typo or a missing channel).
critical libmamba Could not solve for environment specs
```

### Installed packages

```shell
https://gist.githubusercontent.com/yrik/3c5bca8c5003a4202b79bec58b21d782/raw/d49c7b767fd23af54cb4b373421d801d8e0c12ff/micormamba%2520list (output was longer than allowed by GitHub)
```

### Environment info

```shell
(ros_env) yura@Yuras-MacBook-Pro ros_ws % micromamba info

            environment : ros_env (active)
           env location : /Users/yura/micromamba/envs/ros_env
      user config files : /Users/yura/.mambarc
 populated config files : /Users/yura/.condarc
       libmamba version : 1.4.4
     micromamba version : 1.4.4
           curl version : libcurl/7.84.0 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.47.0
     libarchive version : libarchive 3.5.3 zlib/1.2.11 liblzma/5.0.5 bz2lib/1.0.8
       virtual packages : __unix=0=0
                          __osx=13.1=0
                          __archspec=1=arm64
               channels : https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
       base environment : /Users/yura/micromamba
               platform : osx-arm64
```

Tobias-Fischer, 2023-06-14:
I think you need to add the robostack-staging channel explicitly when using micromamba - could you please try?

yrik, 2023-06-15:
Thanks, it works.

closed 2023-06-15 by yrik