RoboStack/ros-humble #79: Update generate-parameter-library to version 0.3.3

olivier-stasse, 2023-06-26:
Dear Robostack maintainers, 
Some ros2_control packages are not available on ros-humble because the generate-parameter-library binaries are on the 0.3.0 version and not the 0.3.3 currently available on linux. 
I tried to see how to update the version number from the documentation but did not understand what to do...

Tobias-Fischer, 2023-06-26:
Hi Olivier, Which packages are missing, and which platforms are you interested in?

olivier-stasse, 2023-06-26:
Dear @Tobias-Fischer, 
I tried to update and add several packages: 
  - generate-parameter-library (update)
  - control_msgs (update)
  - steering-controllers-library (new)
  - ackermann-steering-controller (new)
  - forward-command-controller (new)
using the following vinca.yaml file on the branch ```buildbranch_osx```:
```
ros_distro: humble

# mapping for package keys
conda_index:
  - robostack.yaml
  - packages-ignore.yaml

build_number: 3

mutex_package: ros2-distro-mutex 0.3 humble

skip_all_deps: true

# If full rebuild, the build number of the existing package has
# to match the selected build number for skipping
full_rebuild: true

# build_in_own_azure_stage:
# - ros-galactic-rviz-rendering

packages_skip_by_deps:
  # - rttest
  - cartographer
  - octomap

packages_remove_from_deps:
  - cartographer
  - octomap
  - tlsf
  - tlsf_cpp
  - pendulum_control
  - rttest
  # Not available for macOS and Windows!
  - gripper_controllers

skip_existing:
  # - https://conda.anaconda.org/robostack-humble/
  - https://conda.anaconda.org/robostack-staging/

packages_select_by_deps:
  - ros_workspace
  - ros_environment
  - ros_base
  - desktop
  - moveit
  - moveit_servo
  - desktop_full
  #- navigation2
  - rosbridge_suite
  - vision_msgs
  #- rosbag2_storage_mcap
  #- foxglove_bridge
  - turtlebot3
  - generate-parameter-library
  - control_msgs
  - steering-controllers-library
  - ackermann-steering-controller
  - forward-command-controller
  - joint-state-broadcaster
  - joint-state-publisher
  - joint-trajectory-controller
  - joint-limits
  - xacro
  # - robot-localization
  # - velodyne
  # - sbg_driver
  # - vision-opencv
  - ackermann-msgs

  - gazebo_msgs
  - gazebo_dev
  - gazebo_ros
  - gazebo_plugins
  - gazebo_ros_control
  - gazebo_ros_pkgs

  - turtlebot3_gazebo
  - plotjuggler-ros
  #- plotjuggler

patch_dir: patch
```
  
with ```skip_all_deps: true```. The recipes were generated with ```vinca -m -n```.

The comments on the packages were used to avoid compiling the packages that outside the scope of the ros2_controllers.

It was tried using ```boa``` and it seems to work on ```buildbranch_osx```. 

I am not sure how this translate in a useful PR for [ros-humble](https://github.com/RoboStack/ros-humble) wrt the updates. If we only need to add the new packages it is easy to submit one.

Thanks again for the support.

Tobias-Fischer, 2023-06-26:
You could submit a pull request with the `buildbranch_osx` as upstream branch, and I'd be happy to merge that.

Tobias-Fischer, 2023-06-26:
(big hack, but it's ok ..)

closed 2023-06-27 by Tobias-Fischer

olivier-stasse, 2023-06-27:
Sorry thanks !