RoboStack/ros-humble #84: Unable to teleop turtlesim with a MacBook Pro M2

cmastalli, 2023-06-30:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Thank you again for supporting this!

I installed ROS2 humble and tested it as explained on the [Robostack website](https://robostack.github.io/GettingStarted.html). Everything looks fine as I can launch Rviz.

However, I am unable to teleop turtlesim as explained in the [RO2 tutorial website](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Introducing-Turtlesim/Introducing-Turtlesim.html). I can run each node independently but they do not communicate with each other. I have defined the ROS domain ID as 1.

Since this was not working, I tried to inspect the nodes list. However, when doing so, I encountered the same issue reported for Windows OS at https://github.com/ros2/ros2cli/issues/713.

At this point, I am unsure where it is a bug, but I thought it was useful to report this here.

Thanks,
Carlos

### Installed packages

```shell
# packages in environment at /Users/cm2099/mambaforge/envs/ros2:
#
# Name                    Version                   Build  Channel
aiohttp                   3.8.4           py310h2aa6e3c_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
argcomplete               3.1.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
boost                     1.78.0          py310h629746b_4    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.0.9                h1a8c8d9_9    conda-forge
brotli-bin                1.0.9                h1a8c8d9_9    conda-forge
bullet                    3.24                 h69fb684_0    conda-forge
bullet-cpp                3.24            py310h2b830bf_0    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.1               hb547adb_0    conda-forge
ca-certificates           2023.5.7             hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h2399d43_3    conda-forge
charset-normalizer        3.1.0              pyhd8ed1ab_0    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.1.0           py310h38f39d4_0    conda-forge
cppcheck                  2.7.5           py310hf3ebaa5_1    conda-forge
cryptography              41.0.1          py310hdd3b5e7_0    conda-forge
curl                      7.88.1               h9049daf_1    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1          py310hbe9552e_0    conda-forge
double-conversion         3.2.0                hb7217d7_1    conda-forge
eigen                     3.4.0                hc021e02_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.1.1              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hb7217d7_1    conda-forge
ffmpeg                    5.1.2           gpl_hf318d42_106    conda-forge
flake8                    6.0.0              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             hd3e9afc_1011    conda-forge
fmt                       9.1.0                hffc8910_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
[... 511 lines cut ...]
 populated config files : /Users/cm2099/mambaforge/.condarc
                          /Users/cm2099/mambaforge/envs/ros2/.condarc
          conda version : 23.1.0
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=arm64
                          __osx=13.0=0
                          __unix=0=0
       base environment : /Users/cm2099/mambaforge  (writable)
      conda av data dir : /Users/cm2099/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/cm2099/mambaforge/pkgs
                          /Users/cm2099/.conda/pkgs
       envs directories : /Users/cm2099/mambaforge/envs
                          /Users/cm2099/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.1.0 requests/2.31.0 CPython/3.10.12 Darwin/22.1.0 OSX/13.0
                UID:GID : 503:20
             netrc file : None
           offline mode : False
```

fodaysahid, 2023-09-24:
I am using a Macbook Air M2 and I just tried and it works following the guide here [Ros 2 Documentation](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Introducing-Turtlesim/Introducing-Turtlesim.html)

Please see screenshot below
[image]

Tobias-Fischer, 2023-11-06:
Were you able to resolve this issue @cmastalli? I'll close here in the meantime, let us know if you are still having issues.

closed 2023-11-06 by Tobias-Fischer

cmastalli, 2023-11-07:
> Were you able to resolve this issue @cmastalli? I'll close here in the meantime, let us know if you are still having issues.

@Tobias-Fischer I haven't been able to resolve this issue. I am unable to debug this, unfortunately.