RoboStack/ros-humble #88: RViz2 launch failed on MacBook Pro M1

huma0605, 2023-07-06:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I have a MacBook Pro 2021 with M1 Pro. I followed the instructions to install humble. But when launching rviz2, I am seeing the following error message
```
dyld[95562]: Library not loaded: /usr/local/opt/freetype/lib/libfreetype.6.dylib
  Referenced from: <4C82972E-B20F-33C3-A22D-6B0C85060527> /Users/<username>/miniconda/envs/ros_env_humble/opt/rviz_ogre_vendor/lib/libOgreOverlay.1.12.13.dylib
  Reason: tried: '/Users/<username>/miniconda/envs/ros_env_humble/opt/rviz_ogre_vendor/lib/libfreetype.6.dylib' (no such file), '/usr/local/opt/freetype/lib/libfreetype.6.dylib' (no such file), '/usr/local/opt/freetype/lib/libfreetype.6.dylib' (no such file), '/usr/local/lib/libfreetype.6.dylib' (no such file), '/usr/lib/libfreetype.6.dylib' (no such file, not in dyld cache)
```

I try to `brew install freetype` and force link it to `/Users/<username>/miniconda/envs/ros_env_humble/opt/rviz_ogre_vendor/lib/libfreetype.6.dylib`. But then the error message changed to
```
dyld[95562]: Library not loaded: /usr/local/opt/freetype/lib/libfreetype.6.dylib
  Referenced from: <4C82972E-B20F-33C3-A22D-6B0C85060527> /Users/weichengkuo/miniconda/envs/ros_env_humble/opt/rviz_ogre_vendor/lib/libOgreOverlay.1.12.13.dylib
  Reason: tried: '/Users/weichengkuo/miniconda/envs/ros_env_humble/opt/rviz_ogre_vendor/lib/libfreetype.6.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/opt/freetype/lib/libfreetype.6.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/freetype/lib/libfreetype.6.dylib' (no such file), '/usr/local/opt/freetype/lib/libfreetype.6.dylib' (no such file), '/usr/local/lib/libfreetype.6.dylib' (no such file), '/usr/lib/libfreetype.6.dylib' (no such file, not in dyld cache)
```

That is, it complained that the incompatible architecture ('arm64' vs 'x86_64').

### Installed packages

```shell
# Name                    Version                   Build  Channel
aiohttp                   3.8.4           py310h6729b98_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                hf0c8a7f_0    conda-forge
argcomplete               3.1.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hd9e13b6_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
boost                     1.78.0          py310h3e792ce_4    conda-forge
boost-cpp                 1.78.0               h31500c2_2    conda-forge
brotli                    1.0.9                hb7f2c08_9    conda-forge
brotli-bin                1.0.9                hb7f2c08_9    conda-forge
bullet                    3.24                 ha188af9_0    conda-forge
bullet-cpp                3.24                 hcd8b382_0    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.19.1               h0dc2134_0    conda-forge
c-compiler                1.5.2                hbf74d83_0    conda-forge
ca-certificates           2023.5.7             h8857fd0_0    conda-forge
cairo                     1.16.0            h904041c_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.3              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             h76f1dac_13    conda-forge
cctools_osx-64            973.0.1             hcc6d90d_13    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310ha78151a_3    conda-forge
charset-normalizer        3.1.0              pyhd8ed1ab_0    conda-forge
clang                     14.0.6               h694c41f_1    conda-forge
clang-14                  14.0.6          default_hdb78580_1    conda-forge
clang_osx-64              14.0.6               h3113cd8_6    conda-forge
clangxx                   14.0.6          default_hdb78580_1    conda-forge
clangxx_osx-64            14.0.6               h6f97653_6    conda-forge
cmake                     3.26.3               h4032537_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py310h2ec42d9_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
[... 569 lines cut ...]
       user config file : /Users/<username>/.condarc
 populated config files : /Users/<username>/miniconda/envs/ros_env_humble/.condarc
          conda version : 23.1.0
    conda-build version : not installed
         python version : 3.10.9.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=13.3=0
                          __unix=0=0
       base environment : /Users/<username>/miniconda  (writable)
      conda av data dir : /Users/<username>/miniconda/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/<username>/miniconda/pkgs
                          /Users/<username>/.conda/pkgs
       envs directories : /Users/<username>/miniconda/envs
                          /Users/<username>/.conda/envs
               platform : osx-64
             user-agent : conda/23.1.0 requests/2.28.1 CPython/3.10.9 Darwin/22.4.0 OSX/13.3
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-07-07:
There should be a libfreetype.dylib in `/Users/<username>/miniconda/envs/ros_env_humble/lib` - can you try to force link to this one instead? Try to avoid mixing homebrew and conda as much ass possible.

Also, why are you using the osx-64 (Intel) packages? I would recommend setting up your conda afresh and download the osx-arm64 version.

huma0605, 2023-07-14:
Thanks for the pointer! Confirmed that the following link could resolve the issue.
```
sudo ln -s /Users/<username>/miniconda3/envs/ros_env_humble/lib/libfreetype.6.dylib /Users/<username>/miniconda3/envs/ros_env_humble/opt/rviz_ogre_vendor/lib/libfreetype.6.dylib
```

Tobias-Fischer, 2024-02-09:
Note: This does not happen with the osx-arm64 packages, and I think it does not happen on osx-64 either. So it only happens on M1 machines with osx-64 packages installed. Therefore, I don't think it's worth fixing.

Tobias-Fischer, 2026-10-18:
**Real bug on osx-64, and the rebuild doesn't fix it.** The published osx-64 `rviz_ogre_vendor` (11.2.26, build 20) has `libOgreOverlay.1.12.13.dylib` linking `/usr/local/opt/freetype/lib/libfreetype.6.dylib`, from Homebrew on the CI runner; the osx-arm64 build correctly links `@rpath/libfreetype.6.dylib`. So this hits any osx-64 user without Homebrew's freetype, not just Rosetta setups. The humble rebuild in the RoboStack monorepo, [RoboStack/robostack](https://github.com/RoboStack/robostack), where all ROS 2 distributions are now built ([RoboStack/robostack#62](https://github.com/RoboStack/robostack/pull/62)) still logs `Found FREETYPE: /usr/local/lib/libfreetype.dylib` on osx-64. **Candidate for moving** there, where it can be fixed by pointing Ogre's FreeType lookup at the conda prefix: if a maintainer replies "go ahead", I'll transfer this issue there.

🤖 Claude (AI assistant), on behalf of @Tobias-Fischer

closed 2026-10-18 by Tobias-Fischer