RoboStack/ros-humble #89: Met an error when running `colcon build --symlink-install --merge-install` in the Humble tutorial

songyuc, 2023-07-14:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi, guys!
I met an error on Win11 when running [colcon build --symlink-install --merge-install](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html#:~:text=%E5%8A%A0%E5%BF%AB%E8%BF%AD%E4%BB%A3%E9%80%9F%E5%BA%A6%E3%80%82-,Linux,colcon%20build%20%2D%2Dsymlink%2Dinstall%20%2D%2Dmerge%2Dinstall,-Windows%20doesn%E2%80%99t%20allow)  in [the Humble tutorial](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html).

The command and error output is like the following:
```
(ros_env) D:\dev\ros2_ws>colcon build --symlink-install --merge-install
Starting >>> examples_rclcpp_async_client
Starting >>> examples_rclcpp_cbg_executor
Starting >>> examples_rclcpp_minimal_action_client
Starting >>> examples_rclcpp_minimal_action_server
Starting >>> examples_rclcpp_minimal_client
Starting >>> examples_rclcpp_minimal_composition
Starting >>> examples_rclcpp_minimal_publisher
Starting >>> examples_rclcpp_minimal_service
Starting >>> examples_rclcpp_minimal_subscriber
Starting >>> examples_rclcpp_minimal_timer
Starting >>> examples_rclcpp_multithreaded_executor
Starting >>> examples_rclcpp_wait_set
Starting >>> examples_rclpy_executors
Starting >>> examples_rclpy_guard_conditions
Starting >>> examples_rclpy_minimal_action_client
Starting >>> examples_rclpy_minimal_action_server
--- stderr: examples_rclpy_minimal_action_client
C:\Users\songyucheng\mambaforge\envs\ros_env\Lib\site-packages\setuptools\command\easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
C:\Users\songyucheng\mambaforge\envs\ros_env\Lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< examples_rclpy_minimal_action_client [18.1s]
Starting >>> examples_rclpy_minimal_client
--- stderr: examples_rclpy_minimal_action_server
C:\Users\songyucheng\mambaforge\envs\ros_env\Lib\site-packages\setuptools\command\easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
C:\Users\songyucheng\mambaforge\envs\ros_env\Lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< examples_rclpy_minimal_action_server [18.3s]
--- stderr: examples_rclpy_guard_conditions
C:\Users\songyucheng\mambaforge\envs\ros_env\Lib\site-packages\setuptools\command\easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
C:\Users\songyucheng\mambaforge\envs\ros_env\Lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< examples_rclpy_guard_conditions [18.5s]
Starting >>> examples_rclpy_minimal_publisher
Starting >>> examples_rclpy_minimal_service
--- stderr: examples_rclpy_executors
C:\Users\songyucheng\mambaforge\envs\ros_env\Lib\site-packages\setuptools\command\easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
C:\Users\songyucheng\mambaforge\envs\ros_env\Lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< examples_rclpy_executors [18.8s]
Starting >>> examples_rclpy_minimal_subscriber
Finished <<< examples_rclcpp_minimal_client [29.9s]
Starting >>> examples_rclpy_pointcloud_publisher
Finished <<< examples_rclcpp_async_client [30.3s]
Starting >>> launch_testing_examples
Finished <<< examples_rclcpp_multithreaded_executor [30.3s]
Finished <<< examples_rclcpp_minimal_timer [36.9s]
Finished <<< examples_rclcpp_minimal_service [39.0s]
Finished <<< examples_rclcpp_minimal_action_server [39.3s]
--- stderr: examples_rclpy_minimal_service
C:\Users\songyucheng\mambaforge\envs\ros_env\Lib\site-packages\setuptools\command\easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
[... 670 lines cut ...]
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.10.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.1=0
                          __win=0=0
       base environment : C:\Users\songyucheng\mambaforge  (writable)
      conda av data dir : C:\Users\songyucheng\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\songyucheng\mambaforge\pkgs
                          C:\Users\songyucheng\.conda\pkgs
                          C:\Users\songyucheng\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\songyucheng\mambaforge\envs
                          C:\Users\songyucheng\.conda\envs
                          C:\Users\songyucheng\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.10 Windows/10 Windows/10.0.22621
          administrator : True
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-08-13:
Seems to be the same issue as https://github.com/ros2/examples/issues/344 and not related to RoboStack specifically but to ROS2 in general.

Tobias-Fischer, 2023-08-13:
Suggested solution is to change
https://github.com/ros2/examples/blob/1d97c4fc7445554f6f85f63305d424fc017212a0/rclcpp/topics/minimal_subscriber/CMakeLists.txt#L40
to `STATIC` instead of `SHARED`.

songyuc, 2023-08-14:
Hi, @Tobias-Fischer! Thanks sincerely for the suggested solution.

Tobias-Fischer, 2023-08-14:
Can we close here?

songyuc, 2023-08-14:
I am trying validate this solution. So, could you please give a little more time? And I will list my validation results here.

songyuc, 2023-08-14:
This issue would also appear on a ROS2 Iron [Official Installation](https://docs.ros.org/en/iron/Installation/Windows-Install-Binary.html).