RoboStack/ros-humble #91: Package conflict between ros-humble and ros-noetic

creaton0305, 2023-08-03:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi, I am trying to install ros-noetic and ros-humble by following the getting-started guide (https://robostack.github.io/GettingStarted.html). However, there seems to be a python version conflict in the packages which makes it unable to install the packages. (ros-noetic install python 3.9, while ros-humble installs python 3.10)

Error:
Looking for: ['ros-humble-desktop']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
robostack-staging/noarch                                      No change
robostack-staging/linux-64                                    No change

Pinned packages:
  - python 3.9.*

Could not solve for environment specs
The following package could not be installed
└─ ros-humble-desktop   is installable and it requires
   ├─ python  , which can be installed;
   └─ python_abi 3.10.* *_cp310 with the potential options
      ├─ python_abi 3.10 would require
      │  └─ python 3.10.* *_cpython, which conflicts with any installable versions previously reported;
      └─ python_abi 3.10 would require
         └─ python 3.10.* , which can be installed.

### Installed packages

```shell
# packages in environment at /home/ws2/mambaforge/envs/ros_env2:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bcrypt                    3.2.2            py39hb9d737c_1    conda-forge
boost                     1.78.0           py39h7c9e3ff_4    conda-forge
boost-cpp                 1.78.0               h5adbc97_2    conda-forge
brotli                    1.0.9                h166bdaf_9    conda-forge
brotli-bin                1.0.9                h166bdaf_9    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.19.1               hd590300_0    conda-forge
ca-certificates           2023.7.22            hbcca054_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_3    conda-forge
cmake                     3.26.3               h077f3f9_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.1.0            py39h7633fee_0    conda-forge
cryptography              41.0.2           py39hd4f0224_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1           py39hf3d152e_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
ffmpeg                    5.1.2           gpl_h8dda1f0_106    conda-forge
fftw                      3.3.10          nompi_hc118613_108    conda-forge
[... 437 lines cut ...]
       virtual packages : __archspec=1=x86_64
                          __glibc=2.35=0
                          __linux=5.19.0=0
                          __unix=0=0
       base environment : /home/ws2/mambaforge  (writable)
      conda av data dir : /home/ws2/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-experimental/linux-64
                          https://conda.anaconda.org/robostack-experimental/noarch
                          https://conda.anaconda.org/robostack-humble/linux-64
                          https://conda.anaconda.org/robostack-humble/noarch
                          https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/ws2/mambaforge/pkgs
                          /home/ws2/.conda/pkgs
       envs directories : /home/ws2/mambaforge/envs
                          /home/ws2/.conda/envs
               platform : linux-64
             user-agent : conda/23.1.0 requests/2.31.0 CPython/3.10.12 Linux/5.19.0-50-generic ubuntu/22.04.2 glibc/2.35
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

traversaro, 2023-08-03:
Do you have any specific use case for installing ros-humble and ros-noetic in the same environment, as opposed to install them in separate environments?

creaton0305, 2023-08-03:
> Do you have any specific use case for installing ros-humble and ros-noetic in the same environment, as opposed to install them in separate environments?

Thanks for the quick reply. I'm trying to use MAVROS (built on ROS 1) and ROS 2 together for one of my projects, so I thought it would be better to build both versions inside 1 environment

traversaro, 2023-08-03:
> Thanks for the quick reply. I'm trying to use MAVROS (built on ROS 1) and ROS 2 together for one of my projects, so I thought it would be better to build both versions inside 1 environment

Together in which way? If you do not need to used them as part of the same Python script or C++ application, you probably should be able to use two different environments, unless I am missing something specific of your use case.

creaton0305, 2023-08-03:
> Together in which way? If you do not need to used them as part of the same Python script or C++ application, you probably should be able to use two different environments, unless I am missing [...]

Actually come to think of it, i think you are right. I think i had previously misunderstood on how the bridge between ROS 1 and 2 work. I will try doing it in 2 separate environments. Thanks for the advice and support!

closed 2023-08-03 by creaton0305

traversaro, 2023-08-03:
Feel free to reopen if installing both in the same environment is actually necessary, thanks!