RoboStack/ros-humble #94: Can someone explain this?

Thomas-Yip, 2023-08-12:
[image]

The colcon build of the package is stated successful but I can't run the package. What's wrong with it?

traversaro, 2023-08-12:
Hello @Thomas-Yip , can you share more info on the conda environment you are working in (i.e. report the output of the  mamba list command) and the colcon workspace you are trying to build? 
Furthermore, are you sure that you sourced the colcon workspace?

Thomas-Yip, 2023-08-13:
Here is the mamba list command:
***
# Name                    Version                   Build  Channel
brotli-python             1.0.9           py310h0f1eb42_9    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.1               hb547adb_0    conda-forge
ca-certificates           2023.7.22            hf0a4a13_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h2399d43_3    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
conda                     23.1.0          py310hbe9552e_0    conda-forge
conda-package-handling    2.0.2              pyh38be061_0    conda-forge
conda-package-streaming   0.8.0              pyhd8ed1ab_0    conda-forge
cryptography              41.0.2          py310hdd3b5e7_0    conda-forge
fmt                       9.1.0                hffc8910_0    conda-forge
icu                       72.1                 he12128b_0    conda-forge
idna                      3.4                pyhd8ed1ab_0    conda-forge
krb5                      1.21.1               h92f50d5_0    conda-forge
libarchive                3.6.2                h82b9b87_1    conda-forge
libcurl                   8.1.2                hc52a3a8_1    conda-forge
libcxx                    16.0.6               h4653b0c_0    conda-forge
libedit                   3.1.20191231         hc8eb9b7_2    conda-forge
libev                     4.33                 h642e427_1    conda-forge
libffi                    3.4.2                h3422bc3_5    conda-forge
libiconv                  1.17                 he4db4b2_0    conda-forge
libmamba                  1.4.2                h7d1d596_0    conda-forge
libmambapy                1.4.2           py310h34b6e76_0    conda-forge
libnghttp2                1.52.0               hae82a92_0    conda-forge
libsolv                   0.7.24               ha614eb4_1    conda-forge
libsqlite                 3.42.0               hb31c410_0    conda-forge
libssh2                   1.11.0               h7a5bd25_0    conda-forge
libxml2                   2.11.4               he3bdae6_0    conda-forge
libzlib                   1.2.13               h53f4e23_5    conda-forge
lz4-c                     1.9.4                hb7217d7_0    conda-forge
lzo                       2.10              h642e427_1000    conda-forge
mamba                     1.4.2           py310ha5d4528_0    conda-forge
ncurses                   6.4                  h7ea286d_0    conda-forge
openssl                   3.1.2                h53f4e23_0    conda-forge
pip                       23.1.2             pyhd8ed1ab_0    conda-forge
pluggy                    1.2.0              pyhd8ed1ab_0    conda-forge
pybind11-abi              4                    hd8ed1ab_3    conda-forge
pycosat                   0.6.4           py310h8e9501a_1    conda-forge
pycparser                 2.21               pyhd8ed1ab_0    conda-forge
pyopenssl                 23.2.0             pyhd8ed1ab_1    conda-forge
pysocks                   1.7.1              pyha2e5f31_6    conda-forge
python                    3.10.12         h01493a6_0_cpython    conda-forge
python_abi                3.10                    3_cp310    conda-forge
readline                  8.2                  h92ec313_1    conda-forge
reproc                    14.2.4               h1a8c8d9_0    conda-forge
reproc-cpp                14.2.4               hb7217d7_0    conda-forge
requests                  2.31.0             pyhd8ed1ab_0    conda-forge
ruamel.yaml               0.17.32         py310h2aa6e3c_0    conda-forge
ruamel.yaml.clib          0.2.7           py310h8e9501a_1    conda-forge
setuptools                68.0.0             pyhd8ed1ab_0    conda-forge
tk                        8.6.12               he1e0b03_0    conda-forge
toolz                     0.12.0             pyhd8ed1ab_0    conda-forge
tqdm                      4.65.0             pyhd8ed1ab_1    conda-forge
tzdata                    2023c                h71feb2d_0    conda-forge
urllib3                   2.0.3              pyhd8ed1ab_1    conda-forge
wheel                     0.40.0             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
yaml-cpp                  0.7.0                hb7217d7_2    conda-forge
zstandard                 0.19.0          py310h07fcf8c_2    conda-forge
zstd                      1.5.2                h4f39d0f_7    conda-forge

***
I have problem with following command. "

bash-3.2$ . ~/install/ros2-osx/setup.bash
bash: /Users/yipchiho/ros2_install/ros2-osx/setup.bash: No such file or directory "

Thomas-Yip, 2023-08-14:
I finally figured out the mistakes I made, I should source the workspace in .zsh instead of .bash

closed 2023-08-14 by Thomas-Yip