RoboStack/ros-humble #96: failed to create process when running ros2 on W11

RicardoSPeres, 2023-08-26:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Followed the getting started documentation and managed to get rviz2 up and running (despite getting multiple errors stating "could not load pixmap package").
However when I try to run ros2, or for instance ros2 launch for a launch file, it only prints "failed to create process.". Sourcing the workspace with "call install/local_setup.bat" didn't help either.
Any help would be appreciated, thank you

### Installed packages

```shell
# Name                    Version                   Build  Channel
aiohttp                   3.8.5           py310h8d17308_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h63175ca_0    conda-forge
argcomplete               3.1.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
boost                     1.78.0          py310h220cb41_4    conda-forge
boost-cpp                 1.78.0               h9f4b32c_3    conda-forge
brotli                    1.0.9                hcfcfb64_9    conda-forge
brotli-bin                1.0.9                hcfcfb64_9    conda-forge
bullet                    3.24                 hbbfc1a7_0    conda-forge
bullet-cpp                3.24                 h1c4a608_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h628cb3f_3    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
cmake                     3.26.4               h1537add_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.1.0           py310h232114e_0    conda-forge
cppcheck                  2.10.3          py310hbf4f5aa_0    conda-forge
cryptography              41.0.3          py310h6e82f81_0    conda-forge
curl                      7.88.1               h68f0423_1    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.0                hcfcfb64_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1          py310h5588dad_0    conda-forge
double-conversion         3.2.0                h63175ca_1    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
ffmpeg                    5.1.2           gpl_h5037a79_109    conda-forge
flake8                    6.1.0              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             h23a95e4_1011    conda-forge
fmt                       9.1.0                h181d51b_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.42.1          py310h8d17308_0    conda-forge
foonathan-memory          0.7.2                h57928b3_1    conda-forge
freeglut                  3.2.2                h63175ca_2    conda-forge
freeimage                 3.18.0              h064cf08_10    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
frozenlist                1.4.0           py310h8d17308_0    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
[... 476 lines cut ...]
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.2=0
                          __win=0=0
       base environment : C:\mambaforge  (read only)
      conda av data dir : C:\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\mambaforge\pkgs
                          C:\Users\Ricardo Peres\.conda\pkgs
                          C:\Users\Ricardo Peres\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\Ricardo Peres\.conda\envs
                          C:\mambaforge\envs
                          C:\Users\Ricardo Peres\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Windows/10 Windows/10.0.22621
          administrator : False
             netrc file : None
           offline mode : False
```

traversaro, 2023-08-26:
I tried a few commands and they are working fine. Can you report exactly the command you are executing that is resulting in "failed to create process."  error? Thanks!

RicardoSPeres, 2023-08-27:
After following the "getting started" instructions, testing works fine and rviz2 opens the rviz window as expected. However any command starting with "ros2" results in the "failed to create process" output, it can be anything from just running "ros2", to "ros2 run <package_name> <node>", "ros2 launch <package_name> <launch_file>" etc.

I noticed there are multiple errors when I activate the ros2 env, might be related. Here's the output after running "mamba activate ros_humble":
```
C:\>mamba activate ros_humble

C:\>SET DISTUTILS_USE_SDK=1

C:\>SET MSSdk=1

C:\>SET "VS_VERSION=16.0"

C:\>SET "VS_MAJOR=16"

C:\>SET "VS_YEAR=2019"

C:\>set "MSYS2_ARG_CONV_EXCL=/AI;/AL;/OUT;/out"

C:\>set "MSYS2_ENV_CONV_EXCL=CL"

C:\>set "PY_VCRUNTIME_REDIST=\bin\vcruntime140.dll"

C:\>set "CXX=cl.exe"

C:\>set "CC=cl.exe"

C:\>set "VSINSTALLDIR="

C:\>set "NEWER_VS_WITH_OLDER_VC=0"

C:\>for /F "usebackq tokens=*" %i in (`vswhere.exe -nologo -products * -version [16.0,17.0) -property installationPath`) do (set "VSINSTALLDIR=%i\" )

C:\>(set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\" )

C:\>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\" (for /F "usebackq tokens=*" %i in (`vswhere.exe -nologo -products * -requires Microsoft.VisualStudio.ComponentGroup.VC.Tools.142.x86.x64 -property installationPath`) do (
set "VSINSTALLDIR=%i\"
 set "NEWER_VS_WITH_OLDER_VC=1"
) )

C:\>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\" (for /F "usebackq tokens=*" %i in (`vswhere.exe -nologo -products * -requires Microsoft.VisualStudio.Component.VC.v142.x86.x64 -property installationPath`) do (
set "VSINSTALLDIR=%i\"
 set "NEWER_VS_WITH_OLDER_VC=1"
) )

C:\>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\" (set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\" )

C:\>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\" (set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\" )

C:\>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\" (set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\" )

C:\>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\" (set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\" )

C:\>IF NOT "" == "" (
set "INCLUDE=;"
 set "LIB=;"
 set "CMAKE_PREFIX_PATH=;"
)

C:\>call :GetWin10SdkDir

C:\>call :GetWin10SdkDirHelper HKLM\SOFTWARE\Wow6432Node  1>nul 2>&1

C:\>if errorlevel 1 call :GetWin10SdkDirHelper HKCU\SOFTWARE\Wow6432Node  1>nul 2>&1

C:\>if errorlevel 1 call :GetWin10SdkDirHelper HKLM\SOFTWARE  1>nul 2>&1

C:\>if errorlevel 1 call :GetWin10SdkDirHelper HKCU\SOFTWARE  1>nul 2>&1

C:\>if errorlevel 1 exit /B 1

[... 64 lines cut ...]
C:\>pushd C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>CALL "VC\Auxiliary\Build\vcvars64.bat" -vcvars_ver=14.29 10.0.18362.0
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.8.4
** Copyright (c) 2020 Microsoft Corporation
**********************************************************************
[ERROR:vcvars.bat] Toolset directory for version '14.29' was not found.
[ERROR:VsDevCmd.bat] *** VsDevCmd.bat encountered errors. Environment may be incomplete and/or incorrect. ***
[ERROR:VsDevCmd.bat] In an uninitialized command prompt, please 'set VSCMD_DEBUG=[value]' and then re-run
[ERROR:VsDevCmd.bat] vsdevcmd.bat [args] for additional details.
[ERROR:VsDevCmd.bat] Where [value] is:
[ERROR:VsDevCmd.bat]    1 : basic debug logging
[ERROR:VsDevCmd.bat]    2 : detailed debug logging
[ERROR:VsDevCmd.bat]    3 : trace level logging. Redirection of output to a file when using this level is recommended.
[ERROR:VsDevCmd.bat] Example: set VSCMD_DEBUG=3
[ERROR:VsDevCmd.bat]          vsdevcmd.bat > vsdevcmd.trace.txt 2>&1
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.8.4
** Copyright (c) 2020 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

(ros_humble) C:\>
```

traversaro, 2023-08-27:
I just tried `ros2 launch turtlesim multisim.launch.py` from https://docs.ros.org/en/rolling/Tutorials/Beginner-CLI-Tools/Launching-Multiple-Nodes/Launching-Multiple-Nodes.html and it is working fine? 
Can you tell **exactly** a command that this given the error? Thanks!

RicardoSPeres, 2023-08-30:
The issue happens with anything starting with `ros2`, I just tried running `ros2 launch turtlesim multisim.launch.py` and the output is still "failed to create process.". Thank you

traversaro, 2023-08-30:
Can you try to install the environment in a directory without spaces?

traversaro, 2023-08-30:
The problem does not occurs with entry_points in conda-forge recipes, so it is either a problem of boa or of how recipes with Python entry_points are generated in robostack.

Tobias-Fischer, 2023-08-30:
I don't think we take care of the specifying Python entry points in the meta.yaml/recipe.yaml (as per https://docs.conda.io/projects/conda-build/en/stable/resources/define-metadata.html#python-entry-points). We might be able to steal some code from grayskull to generate them in vinca automatically.

Tobias-Fischer, 2026-10-18:
**Likely still present** (not verified on Windows), **and the fix belongs in vinca.** Its Windows template for Python packages still installs scripts via `setup.py install`, which creates the setuptools `.exe` launchers identified above as the cause; the humble rebuild in the RoboStack monorepo, [RoboStack/robostack](https://github.com/RoboStack/robostack), where all ROS 2 distributions are now built doesn't change this. **Candidate for moving** to [RoboStack/vinca](https://github.com/RoboStack/vinca): if a maintainer replies "go ahead", I'll transfer it there.

🤖 Claude (AI assistant), on behalf of @Tobias-Fischer

traversaro, 2026-10-18:
go ahead