RoboStack/ros-humble #97: Python 3.8 compatibility with ROS2 humble robostack

hanschanhs, 2023-08-28:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I expect ROS2 humble to be compatible with python 3.8. Yet when I follow the instructions to install Robostack on a mamba env with python 3.8, mamba could not solve for environment specs and asks for python 3.10. I require python 3.8 for other python libraries to function on an nvidia jetson orin (I am aware that nvidia offers docker containers for ros, but honestly Robostack has worked very well for me so far when mamba env with python=3.10 was used).

Platfrom: Nvidia Jetson Orin dev board, Jetpack 5.1.2 (Ubuntu 20.04)

Issue: after adding all the necessary channels to the environment, running
```shell
mamba install ros-humble-desktop
```
on a mamba 1.4.2 environment with python=3.8.17 installed, returns the following:
```
Could not solve for environment specs
The following package could not be installed
└─ ros-humble-desktop   is installable and it requires
   ├─ python_abi 3.10.* *_cp310 with the potential options
   │  ├─ python_abi 3.10, which can be installed;
   │  └─ python_abi 3.10 would require
   │     └─ python 3.10.* , which can be installed;
   └─ ros2-distro-mutex [0.2.* humble|0.3.* ], which requires
      └─ setuptools 61.0.0* , which can be installed.
```

Expected result: backwards compatibility of ros-humble-desktop in mamba, at least to python 3.8.

Thanks a lot in advance!

### Installed packages

```shell
_openmp_mutex             4.5                       2_gnu    conda-forge
aiofiles                  23.2.1                   pypi_0    pypi
aiohttp                   3.8.5                    pypi_0    pypi
aiosignal                 1.3.1                    pypi_0    pypi
altair                    5.0.1                    pypi_0    pypi
annotated-types           0.5.0                    pypi_0    pypi
anyio                     3.7.1                    pypi_0    pypi
asttokens                 2.2.1                    pypi_0    pypi
async-timeout             4.0.3                    pypi_0    pypi
attrs                     23.1.0                   pypi_0    pypi
backcall                  0.2.0                    pypi_0    pypi
bzip2                     1.0.8                hf897c2e_4    conda-forge
ca-certificates           2023.7.22            hcefe29a_0    conda-forge
certifi                   2023.7.22                pypi_0    pypi
charset-normalizer        3.2.0                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
clip                      1.0                      pypi_0    pypi
comm                      0.1.4                    pypi_0    pypi
contourpy                 1.1.0                    pypi_0    pypi
cycler                    0.11.0                   pypi_0    pypi
debugpy                   1.6.7.post1              pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
executing                 1.2.0                    pypi_0    pypi
fastapi                   0.101.1                  pypi_0    pypi
ffmpy                     0.3.1                    pypi_0    pypi
filelock                  3.12.2                   pypi_0    pypi
fonttools                 4.42.1                   pypi_0    pypi
frozenlist                1.4.0                    pypi_0    pypi
fsspec                    2023.6.0                 pypi_0    pypi
ftfy                      6.1.1                    pypi_0    pypi
gradio                    3.35.2                   pypi_0    pypi
gradio-client             0.5.0                    pypi_0    pypi
h11                       0.14.0                   pypi_0    pypi
httpcore                  0.17.3                   pypi_0    pypi
httpx                     0.24.1                   pypi_0    pypi
huggingface-hub           0.16.4                   pypi_0    pypi
[... 107 lines cut ...]
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=aarch64
                          __cuda=11.4=0
                          __glibc=2.31=0
                          __linux=5.10.104=0
                          __unix=0=0
       base environment : <home>/mambaforge  (writable)
      conda av data dir : <home>/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-aarch64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-aarch64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : <home>/mambaforge/pkgs
                          <home>/.conda/pkgs
       envs directories : <home>/mambaforge/envs
                          <home>/.conda/envs
               platform : linux-aarch64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Linux/5.10.104-tegra ubuntu/20.04.6 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

traversaro, 2023-08-28:
At the moment, we build for a single version of Python for each distribution to keep the mantainance burden managiable. It was Python 3.9 before https://github.com/RoboStack/ros-humble/pull/21, and now it is Python 3.10 . See also https://github.com/RoboStack/ros-noetic/issues/386#issuecomment-1695530200 for a similar issue with ROS Noetic.

hanschanhs, 2023-08-28:
Thanks for the prompt reply - I see. Given the almost simultaneous request from https://github.com/RoboStack/ros-noetic/issues/386#issuecomment-1695530200 are there plans for backwards compatibility for the ros versions on robostack to at least python 3.8? I understand it is a lot of work but certainly many robotics-relevant  e.g. computer vision, ML python packages share compatibility to latest python 3.8 so I am sure many will find this useful. A big appeal of robostack has been that we didn't have to mess around with e.g. docker to get LTS ros2 humble working, so would be a critical differentiating feature/unique selling point of robostack if enabled.

In the meantime, do you have any suggested workarounds? E.g. having two mamba environments one with python3.10 + robostack and one with python3.8 + my other dependencies will probably struggle to work.

Tobias-Fischer, 2023-08-28:
Hi @hanschanhs, great to see you are interested in RoboStack and giving it a go :).

Unfortunately, as @traversaro said above, the maintenance burden is too high for us to support multiple Python versions (the build time multiplies by the number of Python versions).

That said, there are multiple ways to potentially achieve what you like:
- You could build RoboStack for a particular Python locally. This should be quite straightforward, see our contribution guide. In fact, our users have done this for Python 3.11 (ROS noetic, see https://github.com/RoboStack/ros-noetic/issues/339).
- You could help us bring the computer vision / ML packages that are still missing to conda-forge, which supports python 3.9 (and underlies RoboStack). We are happy to help out to get you started; the RoboStack team probably maintains >100 packages in robotics/ML/CV. Let us know which particular packages require Python 3.8.
- If you work for a company, you could ask them to sponsor RoboStack. If we had money for more powerful CI, we could support multiple Python versions.

Tobias-Fischer, 2025-01-16:
Note - it is now trivial to build locally for different python versions. Simply:
1. `git clone https://github.com/RoboStack/ros-humble.git`
2. Change https://github.com/RoboStack/ros-humble/blob/0651465388131155883cf9e0ba3e6ab3c6a227bf/conda_build_config.yaml#L15-L16 to the desired version
3. Increase the `build_number` in the `vinca_*` files
4. `pixi run build`

Tobias-Fischer, 2026-10-18:
**Won't be possible with current builds:** the RoboStack monorepo, [RoboStack/robostack](https://github.com/RoboStack/robostack), where all ROS 2 distributions are now built builds humble for Python 3.14, and conda-forge no longer builds for Python 3.8. Building several Python versions is tracked in #424 (RoboStack/vinca#153), but 3.8 wouldn't be among them.

🤖 Claude (AI assistant), on behalf of @Tobias-Fischer

closed 2026-10-18 by Tobias-Fischer