RoboStack/ros-humble #98: Packages uploaded to robostack-humble instead of robostack-staging

traversaro, 2023-08-29:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Since https://github.com/RoboStack/ros-humble/pull/21#issuecomment-1295861490, the `ANACONDA_API_TOKEN` of this repo pointed to the `robostack-staging`, that after some time has been adopted as the official channel (despite its name).

However, recently there have been a commit (see https://github.com/RoboStack/ros-humble/commit/3205c8c7ff58866831718990f65cef0f579b20d4 and https://github.com/RoboStack/ros-humble/actions/runs/5987103314/job/16240943279), that instead uploaded the package at `robostack-humble` : 

~~~
2023-08-26T21:49:45.0779993Z + anaconda -t *** upload /opt/conda/build_artifacts/linux-aarch64/ros-humble-slam-toolbox-2.6.5-py310hda37a7e_3.tar.bz2 --force
2023-08-26T21:49:47.9208085Z Using Anaconda API: https://api.anaconda.org
2023-08-26T21:49:48.5051538Z Using "robostack-humble" as upload username
2023-08-26T21:49:48.5053908Z Processing "/opt/conda/build_artifacts/linux-aarch64/ros-humble-slam-toolbox-2.6.5-py310hda37a7e_3.tar.bz2"
2023-08-26T21:49:48.5055380Z Detecting file type...
2023-08-26T21:49:49.6481091Z File type is "Conda"
2023-08-26T21:49:49.6484364Z Extracting conda attributes for upload
2023-08-26T21:49:51.2181067Z Creating package "ros-humble-slam-toolbox"
2023-08-26T21:49:51.4249472Z Creating release "2.6.5"
2023-08-26T21:49:51.7195312Z Uploading file "robostack-humble/ros-humble-slam-toolbox/2.6.5/linux-aarch64/ros-humble-slam-toolbox-2.6.5-py310hda37a7e_3.tar.bz2"
2023-08-26T21:49:51.9749503Z 
2023-08-26T21:49:52.2305844Z   0%|          | 0.00/3.04M [00:00<?, ?B/s]
2023-08-26T21:49:52.3663851Z   0%|          | 8.00k/3.04M [00:00<01:38, 32.1kB/s]
2023-08-26T21:49:52.4852782Z   4%|▎         | 112k/3.04M [00:00<00:08, 357kB/s]  
2023-08-26T21:49:52.6205387Z  10%|█         | 312k/3.04M [00:00<00:03, 836kB/s]
2023-08-26T21:49:52.7205757Z  26%|██▌       | 808k/3.04M [00:00<00:01, 1.90MB/s]
2023-08-26T21:49:52.9625213Z  92%|█████████▏| 2.80M/3.04M [00:00<00:00, 7.23MB/s]
2023-08-26T21:49:52.9626124Z 3.04MB [00:00, 3.23MB/s]                            
2023-08-26T21:49:53.2049034Z Upload complete
2023-08-26T21:49:53.2049508Z 
2023-08-26T21:49:53.2051052Z conda located at:
2023-08-26T21:49:53.2052303Z   https://anaconda.org/robostack-humble/ros-humble-slam-toolbox
~~~

was that expected? fyi @Tobias-Fischer @wolfv 

### Installed packages

```shell
Not relevant.
```

### Environment info

```shell
Not relevant.
```

traversaro, 2023-08-29:
The latest upload job done before the linked one that I can find in the GitHub Actions history is https://github.com/RoboStack/ros-humble/actions/runs/5434084673 , and there the package was uploaded to:
~~~
2023-07-02T01:52:52.1507750Z Using Anaconda API: https://api.anaconda.org
2023-07-02T01:52:52.3443240Z Using "robostack-staging" as upload username
2023-07-02T01:52:52.3545690Z Processing '/Users/runner/mambaforge/conda-bld//osx-64/ros-humble-force-torque-sensor-broadcaster-2.23.0-py310h5aa156f_4.tar.bz2'
2023-07-02T01:52:52.3645320Z Detecting file type...
2023-07-02T01:52:52.3747800Z File type is "Conda"
2023-07-02T01:52:52.3849300Z Extracting conda attributes for upload
2023-07-02T01:52:52.4457360Z Creating package "ros-humble-force-torque-sensor-broadcaster"
2023-07-02T01:52:52.5008040Z Creating release "2.23.0"
2023-07-02T01:52:52.6358630Z Uploading file "robostack-staging/ros-humble-force-torque-sensor-broadcaster/2.23.0/osx-64/ros-humble-force-torque-sensor-broadcaster-2.23.0-py310h5aa156f_4.tar.bz2"
2023-07-02T01:52:52.8796010Z 
2023-07-02T01:52:53.0504170Z   0%|          | 0.00/98.1k [00:00<?, ?B/s]
2023-07-02T01:52:53.0606210Z 104kB [00:00, 608kB/s]                     
2023-07-02T01:52:53.2523060Z Upload complete
2023-07-02T01:52:53.2523600Z 
2023-07-02T01:52:53.2527780Z Processing '/Users/runner/mambaforge/conda-bld//osx-64/ros-humble-imu-sensor-broadcaster-2.23.0-py310h5aa156f_4.tar.bz2'
2023-07-02T01:52:53.2529420Z Detecting file type...
2023-07-02T01:52:53.2699360Z File type is "Conda"
2023-07-02T01:52:53.2700330Z Extracting conda attributes for upload
2023-07-02T01:52:53.3312740Z Creating package "ros-humble-imu-sensor-broadcaster"
2023-07-02T01:52:53.3907370Z Creating release "2.23.0"
2023-07-02T01:52:53.5520180Z Uploading file "robostack-staging/ros-humble-imu-sensor-broadcaster/2.23.0/osx-64/ros-humble-imu-sensor-broadcaster-2.23.0-py310h5aa156f_4.tar.bz2"
2023-07-02T01:52:53.7772450Z 
2023-07-02T01:52:54.0565120Z   0%|          | 0.00/97.5k [00:00<?, ?B/s]
2023-07-02T01:52:54.0617450Z 103kB [00:00, 372kB/s]                     
2023-07-02T01:52:54.2188270Z Upload complete
2023-07-02T01:52:54.2204630Z 
2023-07-02T01:52:54.2219710Z Processing '/Users/runner/mambaforge/conda-bld//osx-64/ros-humble-position-controllers-2.23.0-py310h5aa156f_4.tar.bz2'
2023-07-02T01:52:54.2227670Z Detecting file type...
2023-07-02T01:52:54.2231110Z File type is "Conda"
2023-07-02T01:52:54.2234610Z Extracting conda attributes for upload
2023-07-02T01:52:54.2660600Z Creating package "ros-humble-position-controllers"
2023-07-02T01:52:54.3279270Z Creating release "2.23.0"
2023-07-02T01:52:54.4645260Z Uploading file "robostack-staging/ros-humble-position-controllers/2.23.0/osx-64/ros-humble-position-controllers-2.23.0-py310h5aa156f_4.tar.bz2"
2023-07-02T01:52:54.6914020Z 
2023-07-02T01:52:54.8668640Z   0%|          | 0.00/25.4k [00:00<?, ?B/s]
2023-07-02T01:52:54.8668930Z 31.2kB [00:00, 189kB/s]                    
2023-07-02T01:52:55.0418940Z Upload complete
2023-07-02T01:52:55.0476720Z 
2023-07-02T01:52:55.0527000Z conda located at:
2023-07-02T01:52:55.0611020Z https://anaconda.org/robostack-staging/ros-humble-force-torque-sensor-broadcaster
2023-07-02T01:52:55.0711650Z 
2023-07-02T01:52:55.0772550Z conda located at:
2023-07-02T01:52:55.0874290Z https://anaconda.org/robostack-staging/ros-humble-imu-sensor-broadcaster
2023-07-02T01:52:55.0975360Z 
2023-07-02T01:52:55.1010960Z conda located at:
2023-07-02T01:52:55.1041060Z https://anaconda.org/robostack-staging/ros-humble-position-controllers
2023-07-02T01:52:55.1117710Z 
2023-07-02T01:52:55.1512020Z Post job cleanup.
2023-07-02T01:52:55.3480210Z [command]/usr/local/bin/git version
2023-07-02T01:52:55.3933460Z git version 2.41.0
2023-07-02T01:52:55.4004490Z Copying '/Users/runner/.gitconfig' to '/Users/runner/work/_temp/4d8ef5a8-85e3-49ed-b51b-0e3190d4fb33/.gitconfig'
2023-07-02T01:52:55.4031530Z Temporarily overriding HOME='/Users/runner/work/_temp/4d8ef5a8-85e3-49ed-b51b-0e3190d4fb33' before making global git config changes
2023-07-02T01:52:55.4032840Z Adding repository directory to the temporary git global config as a safe directory
2023-07-02T01:52:55.4043590Z [command]/usr/local/bin/git config --global --add safe.directory /Users/runner/work/ros-humble/ros-humble
2023-07-02T01:52:55.4221160Z [command]/usr/local/bin/git config --local --name-only --get-regexp core\.sshCommand
2023-07-02T01:52:55.4359590Z [command]/usr/local/bin/git submodule foreach --recursive sh -c "git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :"
2023-07-02T01:52:55.6280510Z [command]/usr/local/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2023-07-02T01:52:55.6402090Z http.https://github.com/.extraheader
2023-07-02T01:52:55.6441220Z [command]/usr/local/bin/git config --local --unset-all http.https://github.com/.extraheader
2023-07-02T01:52:55.6781100Z [command]/usr/local/bin/git submodule foreach --recursive sh -c "git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :"
2023-07-02T01:52:56.1141720Z Cleaning up orphan processes
~~~

Tobias-Fischer, 2023-08-29:
/cc @wolfv  something must have gone wrong with the new API key. We should check noetic too.

Tobias-Fischer, 2023-08-29:
I worked it out. Noetic is fine.

Tobias-Fischer, 2023-08-29:
Re-run jobs to upload to correct channel now. Thanks for reporting this @traversaro!

closed 2023-08-29 by Tobias-Fischer