RoboStack/ros-humble #100: On Windows AMENT_PYTHON_EXECUTABLE does not point to a python executable

tony-p, 2023-09-08:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

When creating a shell get an error that AMENT_PYTHON_EXECUTABLE  does not point to a valid executable

```sh
error: AMENT_PYTHON_EXECUTABLE 'C:\Users\me\wa-ros-test\.pixi/env\Library\python.exe' doesn't exist
```

Python executables available in the env
```sh
.pixi/env/Lib/venv/scripts/nt/python.exe
.pixi/env/python.exe
```

### Installed packages

```shell
pixi.toml

[project]
name = "wa-ros-test"
version = "0.1.0"
description = "Proof of concept automation core"
authors = ["me"]
channels = ["conda-forge", "robostack-staging"]
platforms = ["win-64", "linux-64"]

[target.win-64.activation]
scripts = [".pixi/install/setup.bat"]

[target.linux-64.activation]
scripts = [".pixi/install/setup.bash"]

[tasks]
clean = {cmd="rm -rf .pixi/build .pixi/install log/*"}
build = {cmd="PATHEXT=\"$PATHEXT;.py\" colcon build --merge-install --build-base .pixi/build --install-base .pixi/install"}

[dependencies]
ros-humble-desktop = "0.10.*"
cmake = "*"
pkg-config = "*"
make = "*"
ninja = "*"
colcon-common-extensions = "*"
catkin_tools = "*"
zeromq = "4.3.4"
sqlite = "3.40.*"
python = "3.10.*"

[target.win-64.dependencies]
vs2022_win-64 = "19.*"

[target.linux-64.dependencies]
gxx = "12.2.*"
```

### Environment info

```shell
$ pixi info
pixi 0.2.0

Platform            : win-64
Virtual packages    : __win=0=0
                    : __cuda=12.0=0
                    : __archspec=1=x86_64
Cache dir           : C:\Users\me\AppData\Local\rattler/cache
Auth storage        : C:\Users\me\.rattler

Project
------------

Manifest file       : C:\Users\me\wa-ros-test\pixi.toml
Dependency count    : 14
Last updated        : 30-08-2023 13:43:40
Target platforms    : win-64
                    : linux-64
Tasks               : test
                    : build
                    : clean
```

Tobias-Fischer, 2023-09-08:
/cc @wolfv @baszalmstra

traversaro, 2023-09-09:
For full alignment, this was caused by a wrong comment by me in https://github.com/RoboStack/ros-galactic/issues/53#issuecomment-922688831 , see also the related comment by @danzimmerman in https://github.com/RoboStack/ros-galactic/issues/53#issuecomment-1164986782 (that probably went unnoticed as it was in a closed issue).

Tobias-Fischer, 2023-09-09:
Opening till proper package is rebuilt

Tobias-Fischer, 2023-09-14:
Just rebuild it.

closed 2023-09-14 by Tobias-Fischer