RoboStack/ros-humble #101: Package request: PACKAGE_NAME

thinkabout4x, 2023-09-10:
### Package name

ros2_control

### Operating System(s)

- [X] linux-64
- [X] linux-aarch64
- [X] osx-64
- [X] osx-arm64
- [X] win-64

### Additional context (optional)

Hello!
I am currently working on my personal project in ROS2 Humble and highly appreciate your effort of doing  Robostack project!

I want to ask you about the ability of addition of ros2_control package. For my opinion it is quite base thing in ROS that is very widely used. Do you have plans to add it in the nerest future? I also will be happy to hear any advices about building this package on mac m1 from source to try to dig in into the problem!

Thank you in advance!

### Links and references (optional)

https://github.com/ros-controls/ros2_control

Tobias-Fischer, 2023-09-10:
Could you please open a pull request? It's as simple as adding one line to a yaml file, see e.g. https://github.com/RoboStack/ros-humble/pull/56

We already have it on linux-64: https://anaconda.org/robostack-staging/ros-humble-ros2-control

thinkabout4x, 2023-09-11:
Ok! Thank you!
I will do it!

olivier-stasse, 2023-09-11:
Dear @thinkabout4x ros2_control is a meta package for several packages.
A significant amount are already available on robostack.

thinkabout4x, 2023-09-11:
I checked it, thanks! For my case I am working on mac m1 and a lot of this packages such as ros-humble-ros2-controllers, ros-humble-gazebo-ros2-control are available only on linux or not available at all.

My goal is to make simple drone simulator with teleop control. I already set up model for gazebo and now looking for a solution to have a plugin for gazebo to actually make a drone control. The most straightforward solution for my opinion is using ros2_control package, but I also thought about of writing a plugin from scratch.
If you can give me any advices about what packages is better to use for my project I will be very happy!

Tobias-Fischer, 2023-09-14:
See #102

thinkabout4x, 2023-09-14:
Thank you very much!!!