RoboStack/ros-humble #103: Windows - Tutorial interfaces package build doesn't complete

PeterS-mBryonics, 2023-09-15:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

In a fresh ros environment creating the tutorial-interfaces package from https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html results in a build that does not seem to complete - I left it running 40 mins and it hadn't finished. You also can't kill the process with CTRL-C for some reason. I've tried it on two different computers. The same package does build in WSL2 (slower than pure python but it still builds).

Attached is my minimum example (I've removed my email address). Hopefully I've just installed something poorly!

[tutorial_interfaces_2.zip](https://github.com/RoboStack/ros-humble/files/12621629/tutorial_interfaces_2.zip)

### Installed packages

```shell
# Name                    Version                   Build  Channel
aiohttp                   3.8.5           py310h8d17308_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h63175ca_0    conda-forge
argcomplete               3.1.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
boost                     1.78.0          py310h220cb41_4    conda-forge
boost-cpp                 1.78.0               h9f4b32c_4    conda-forge
brotli                    1.1.0                hcfcfb64_0    conda-forge
brotli-bin                1.1.0                hcfcfb64_0    conda-forge
bullet                    3.24                 hbbfc1a7_0    conda-forge
bullet-cpp                3.24                 h1c4a608_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.6.0                hcfcfb64_0    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.8.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h628cb3f_3    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.27.4               h1537add_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py310h5588dad_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-notification       0.2.15          py310h5588dad_1    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compilers                 1.6.0                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.1.0           py310h232114e_0    conda-forge
coverage                  7.3.1           py310h8d17308_0    conda-forge
cppcheck                  2.10.3          py310hbf4f5aa_0    conda-forge
cryptography              41.0.3          py310h6e82f81_0    conda-forge
curl                      7.88.1               h68f0423_1    conda-forge
cxx-compiler              1.6.0                h91493d7_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.0                hcfcfb64_0    conda-forge
[... 520 lines cut ...]
                          C:\Users\peter\mambaforge\envs\ros\.condarc
          conda version : 23.7.4
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=x86_64
                          __win=0=0
       base environment : C:\Users\peter\mambaforge  (writable)
      conda av data dir : C:\Users\peter\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\peter\mambaforge\pkgs
                          C:\Users\peter\.conda\pkgs
                          C:\Users\peter\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\peter\mambaforge\envs
                          C:\Users\peter\.conda\envs
                          C:\Users\peter\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.7.4 requests/2.31.0 CPython/3.10.12 Windows/10 Windows/10.0.22621
          administrator : False
             netrc file : None
           offline mode : False
```

traversaro, 2023-09-15:
Which command is hanging? `colcon build --merge-install --packages-select tutorial_interfaces` ? In that case, can you upload the `ros2_ws/logs` directory?

PeterS-mBryonics, 2023-09-15:
[build_2023-09-15_17-16-08.zip](https://github.com/RoboStack/ros-humble/files/12621702/build_2023-09-15_17-16-08.zip)

See attached. Yes,  that's the command that hangs

traversaro, 2023-09-15:
Can you try to run in a tutorial that does not have a space in the path, i.e. not in `G:\My Drive` ? So we understand if that is the problem.

PeterS-mBryonics, 2023-09-15:
Yup! Horay, different output:
colcon build --merge-install --packages-select tutorial_interfaces_2
Starting >>> tutorial_interfaces_2
[Processing: tutorial_interfaces_2]
[Processing: tutorial_interfaces_2]
[Processing: tutorial_interfaces_2]
[Processing: tutorial_interfaces_2]
--- stderr: tutorial_interfaces_2
CMake Warning (dev) at C:/Users/peter/mambaforge/envs/ros/Library/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:20 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  C:/Users/peter/mambaforge/envs/ros/Library/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  C:/Users/peter/mambaforge/envs/ros/Library/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:16 (rosidl_generate_interfaces)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Users/peter/mambaforge/envs/ros/Library/share/python_cmake_module/cmake/Modules/FindPythonExtra.cmake:52 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  C:/Users/peter/mambaforge/envs/ros/Library/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:23 (find_package)
  C:/Users/peter/mambaforge/envs/ros/Library/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  C:/Users/peter/mambaforge/envs/ros/Library/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:16 (rosidl_generate_interfaces)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Users/peter/mambaforge/envs/ros/Library/share/python_cmake_module/cmake/Modules/FindPythonExtra.cmake:140 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  C:/Users/peter/mambaforge/envs/ros/Library/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:23 (find_package)
  C:/Users/peter/mambaforge/envs/ros/Library/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  C:/Users/peter/mambaforge/envs/ros/Library/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:16 (rosidl_generate_interfaces)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< tutorial_interfaces_2 [2min 14s]

Summary: 1 package finished [2min 15s]
  1 package had stderr output: tutorial_interfaces_2

PeterS-mBryonics, 2023-09-15:
That seems to have got something that runs, I'll take a closer look on Monday. Thank you!

Tobias-Fischer, 2023-09-15:
Great! Thanks for the debugging @traversaro!

closed 2023-09-15 by Tobias-Fischer