RoboStack/ros-humble #104: Cannot run robot_state_publisher / library 'sdformat_urdf_plugin' reported missing

strongly-typed, 2023-09-17:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Great to have robostack available! I'm on macOS 13.5.2 (22G91). rviz2 and simple pub/sub works. I tried to run a URDF demo. I broke it down to that robot_state_publisher does not work.

```
(ros_env) ➜  ros2_ws ros2 launch robot_state_publisher rsp-launch-urdf-inline.py
[INFO] [launch]: All log files can be found below /Users/sascha/.ros/log/2023-09-17-11-12-08-521308-Saschas-MacBook-Pro.local-47448
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [47449]
[robot_state_publisher-1] libc++abi: terminating due to uncaught exception of type pluginlib::LibraryLoadException: Could not find library corresponding to plugin sdformat_urdf_plugin/SDFormatURDFParser. Make sure that the library 'sdformat_urdf_plugin' actually exists.
[ERROR] [robot_state_publisher-1]: process has died [pid 47449, exit code -6, cmd '/usr/local/Caskroom/mambaforge/base/envs/ros_env/lib/robot_state_publisher/robot_state_publisher --ros-args --params-file /var/folders/2s/vmgn6q6d4mdgtz8x0dn57wgw0000gn/T/launch_params_fmxnjock'].
```

Something named `sdformat_urdf_plugin` seems to be installed:
```
(ros_env) ➜  ros2_ws find /usr/local/Caskroom -iname "sdformat_urdf_plugin*"
/usr/local/Caskroom/mambaforge/base/pkgs/ros-humble-sdformat-urdf-1.0.1-py310h5aa156f_3/share/sdformat_urdf/sdformat_urdf_plugin.xml
/usr/local/Caskroom/mambaforge/base/envs/ros_env/share/sdformat_urdf/sdformat_urdf_plugin.xml
```

Would be nice to get this running!

### Installed packages

```shell
(ros_env) ➜  ros2_ws mamba list
# packages in environment at /usr/local/Caskroom/mambaforge/base/envs/ros_env:
#
# Name                    Version                   Build  Channel
aiohttp                   3.8.5           py310h6729b98_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                hf0c8a7f_0    conda-forge
argcomplete               3.1.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hd9e13b6_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
boost                     1.78.0          py310h3e792ce_4    conda-forge
boost-cpp                 1.78.0               h31500c2_2    conda-forge
brotli                    1.1.0                h0dc2134_0    conda-forge
brotli-bin                1.1.0                h0dc2134_0    conda-forge
bullet                    3.21                 ha188af9_4    conda-forge
bullet-cpp                3.21                 hcd8b382_4    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.19.1               h0dc2134_0    conda-forge
c-compiler                1.6.0                h63c33a9_0    conda-forge
ca-certificates           2023.7.22            h8857fd0_0    conda-forge
cairo                     1.16.0            h904041c_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hd9ad811_14    conda-forge
cctools_osx-64            973.0.1             habff3f6_14    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310ha78151a_3    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
clang                     15.0.7               h694c41f_3    conda-forge
clang-15                  15.0.7          default_hdb78580_3    conda-forge
clang_osx-64              15.0.7               h03d6864_3    conda-forge
clangxx                   15.0.7          default_hdb78580_3    conda-forge
clangxx_osx-64            15.0.7               h2133e9c_3    conda-forge
cli11                     2.3.2                hf0c8a7f_0    conda-forge
cmake                     3.26.3               h4032537_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
[... 671 lines cut ...]
 populated config files : /usr/local/Caskroom/mambaforge/base/.condarc
                          /usr/local/Caskroom/mambaforge/base/envs/ros_env/.condarc
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=13.5.2=0
                          __unix=0=0
       base environment : /usr/local/Caskroom/mambaforge/base  (writable)
      conda av data dir : /usr/local/Caskroom/mambaforge/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /usr/local/Caskroom/mambaforge/base/pkgs
                          /Users/sascha/.conda/pkgs
       envs directories : /usr/local/Caskroom/mambaforge/base/envs
                          /Users/sascha/.conda/envs
               platform : osx-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Darwin/22.6.0 OSX/13.5.2
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-09-17:
/cc @traversaro - I remember a similar issue previously, can you remember what the solution was?

traversaro, 2023-09-17:
> /cc @traversaro - I remember a similar issue previously, can you remember what the solution was?

I can't remember any similar issue, sorry!

strongly-typed, 2023-09-18:
I tried on another MacBook

```
(ros_env) ➜  ~ ros2 launch robot_state_publisher rsp-launch-urdf-inline.py
[INFO] [launch]: All log files can be found below /Users/sascha/.ros/log/2023-09-18-21-09-02-749612-Saschas-MacBook-Pro-Early-2013.local-93358
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [93360]
[robot_state_publisher-1] Warning: link 'base_link' material 'blue' undefined.
[robot_state_publisher-1]          at line 84 in /Users/runner/mambaforge/conda-bld/ros-humble-urdfdom-0_1675646167591/work/ros-humble-urdfdom/src/work/urdf_parser/src/model.cpp
[robot_state_publisher-1] Warning: link 'base_link' material 'blue' undefined.
[robot_state_publisher-1]          at line 84 in /Users/runner/mambaforge/conda-bld/ros-humble-urdfdom-0_1675646167591/work/ros-humble-urdfdom/src/work/urdf_parser/src/model.cpp
[robot_state_publisher-1] [INFO] [1695064143.029658101] [robot_state_publisher]: got segment base_link
```

and it seems to work there.

It is running macOS Big Sur 11.7.9 (20G1426).

Tobias-Fischer, 2023-09-18:
Is one of them an intel, and the other one apple silicon?

strongly-typed, 2023-09-18:
No, both Intel.

strongly-typed, 2023-10-22:
I can run it on the new machine. So I close this until it bugs someone else.

strongly-typed, 2023-10-22:
Sorry to reopen the issue so quickly again ... It started bugging me again and now I can reproduce it!

The bug appears after installing `ros-humble-desktop-full`. It works fine if I only install `ros-humble-desktop`.

To start from scratch, I completely deleted `miniconda` and deleted `build`, `log` and `install` in the workspace.

I installed `ros-humble-desktop` and `compilers cmake pkg-config make ninja colcon-common-extensions catkin_tools` with `mamba`.

Then, it was possible to compile and run the example.

I then again removed `build`, `log` and `install` in the workspace, installed `ros-humble-desktop-full`, did a `mamba deactivate && mamba activate ros_env`. After that, I am back at the error mentioned.

Any idea how the installation of `ros-humble-desktop-full` can break it?

```
(ros_env) ➜  ros2_ws mamba install ros-humble-desktop-full

Looking for: ['ros-humble-desktop-full']

conda-forge/osx-64                                          Using cache
conda-forge/noarch                                          Using cache
robostack-staging/osx-64                                      No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.10.*

Transaction

  Prefix: /usr/local/Caskroom/miniconda/base/envs/ros_env

  Updating specs:

   - ros-humble-desktop-full
   - ca-certificates
   - certifi
   - openssl

  Package                                           Version  Build             Channel                 Size
─────────────────────────────────────────────────────────────────────────────────────────────────────────────
  Install:
─────────────────────────────────────────────────────────────────────────────────────────────────────────────

  + octomap                                           1.9.8  hb8565cd_0        conda-forge            205kB
  + libode                                           0.16.2  h7a76584_11       conda-forge            491kB
  + libabseil                                    20230125.3  cxx17_h000cb23_0  conda-forge              1MB
  + urdfdom_headers                                   1.1.0  h1b54a9f_0        conda-forge             19kB
  + cli11                                             2.3.2  hf0c8a7f_0        conda-forge             75kB
  + gdbm                                               1.18  h8a0c380_2        conda-forge            134kB
  + swig                                              4.1.1  h62f31c0_1        conda-forge              1MB
  + libccd-double                                       2.1  he965462_3        conda-forge             33kB
  + gflags                                            2.2.2  hb1e8313_1004     conda-forge             95kB
  + libsodium                                        1.0.18  hbcb3906_1        conda-forge            529kB
  + ogre-next                                         2.2.6  h20040b6_0        conda-forge              3MB
  + lzo                                                2.10  haf1e3a3_1000     conda-forge           Cached
  + urdfdom                                           3.1.1  hd1da41a_0        conda-forge             93kB
  + libignition-utils1                                1.4.0  hb486fe8_0        conda-forge             26kB
  + ruby                                              3.2.2  hb3742b3_0        conda-forge              8MB
  + ogre                                            1.10.12  hd9b3965_10       conda-forge            121MB
  + fcl                                               0.7.0  h699cf62_3        conda-forge              1MB
  + zeromq                                            4.3.5  h93d8f39_0        conda-forge            294kB
  + libarchive                                        3.6.2  h6d8d9f1_0        conda-forge            723kB
  + libignition-common4                               4.6.1  h9877c73_1        conda-forge            521kB
  + libignition-tools1                                1.5.0  h4425e3a_3        conda-forge             34kB
  + dartsim                                          6.12.2  hf8c8161_3        conda-forge             16MB
  + cppzmq                                           4.10.0  h478c2f3_0        conda-forge             29kB
  + libsdformat12                                    12.6.0  h1fb05c2_2        conda-forge            734kB
  + libignition-plugin1                               1.3.0  h75ea093_0        conda-forge             85kB
  + libignition-msgs8                                 8.7.0  hc16761c_2        conda-forge            870kB
  + libignition-rendering6                            6.5.1  h58813d6_1        conda-forge              4MB
[... 35 lines cut ...]
  + ros-humble-ros-ign-image                        0.244.9  py310h5aa156f_3   robostack-staging       20kB
  + ros-humble-ros-gz-sim-demos                     0.244.9  py310h5aa156f_3   robostack-staging       38kB
  + ros-humble-image-pipeline                         3.0.0  py310h5aa156f_3   robostack-staging       13kB
  + ros-humble-simulation                            0.10.0  py310h5aa156f_3   robostack-staging       14kB
  + ros-humble-ros-ign-gazebo-demos                 0.244.9  py310h5aa156f_3   robostack-staging       19kB
  + ros-humble-perception                            0.10.0  py310h5aa156f_3   robostack-staging       14kB
  + ros-humble-desktop-full                          0.10.0  py310h5aa156f_3   robostack-staging       15kB

  Downgrade:
[... cut ...]
  - pybullet                                           3.24  py310hcd8b382_0   conda-forge           Cached
  + pybullet                                           3.21  py310hcd8b382_4   conda-forge             67MB
  - bullet                                             3.24  ha188af9_0        conda-forge           Cached
  + bullet                                             3.21  ha188af9_4        conda-forge              9kB

  Summary:

  Install: 69 packages
  Downgrade: 3 packages

  Total download: 288MB
```

strongly-typed, 2023-10-22:
I can break it down to the installation of `ros-humble-sdformat-urdf`. Adding this package breaks!

```
mamba install --yes ros-humble-sdformat-urdf

Looking for: ['ros-humble-sdformat-urdf']

conda-forge/osx-64                                          Using cache
conda-forge/noarch                                          Using cache
robostack-staging/osx-64                                      No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.10.*

Transaction

  Prefix: /usr/local/Caskroom/miniconda/base/envs/ros_env

  Updating specs:

   - ros-humble-sdformat-urdf
   - ca-certificates
   - certifi
   - openssl

  Package                     Version  Build            Channel                Size
─────────────────────────────────────────────────────────────────────────────────────
  Install:
─────────────────────────────────────────────────────────────────────────────────────

  + urdfdom_headers             1.1.0  h1b54a9f_0       conda-forge            19kB
  + gts                         0.7.6  h53e17e3_4       conda-forge           281kB
  + cli11                       2.3.2  hf0c8a7f_0       conda-forge            75kB
  + gdbm                         1.18  h8a0c380_2       conda-forge           134kB
  + urdfdom                     3.1.1  hd1da41a_0       conda-forge            93kB
  + libignition-utils1          1.4.0  hb486fe8_0       conda-forge            26kB
  + ruby                        3.2.2  hb3742b3_0       conda-forge             8MB
  + libignition-common4         4.6.1  h9877c73_1       conda-forge           521kB
  + libignition-tools1          1.5.0  h4425e3a_3       conda-forge            34kB
  + libsdformat12              12.6.0  h1fb05c2_2       conda-forge           734kB
  + ros-humble-sdformat-urdf    1.0.1  py310h5aa156f_3  robostack-staging      88kB

  Summary:

  Install: 11 packages

  Total download: 10MB

─────────────────────────────────────────────────────────────────────────────────────

urdfdom_headers                                     18.5kB @  83.7kB/s  0.2s
urdfdom                                             93.4kB @ 391.9kB/s  0.2s
cli11                                               74.6kB @ 277.6kB/s  0.3s
libignition-utils1                                  26.4kB @  62.1kB/s  0.2s
gdbm                                               134.2kB @ 291.4kB/s  0.5s
gts                                                281.0kB @ 549.2kB/s  0.5s
libignition-common4                                521.1kB @ 866.8kB/s  0.4s
libignition-tools1                                  33.6kB @  53.7kB/s  0.1s
ros-humble-sdformat-urdf                            88.4kB @ 139.2kB/s  0.4s
libsdformat12                                      734.4kB @ 803.0kB/s  0.5s
ruby                                                 8.1MB @   2.4MB/s  2.9s

Downloading and Extracting Packages:

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
ament_append_unique_value:14: command not found: ament_zsh_to_array
/usr/local/Caskroom/miniconda/base/envs/ros_env/setup.sh:120: command not found: ament_zsh_to_array
```

strongly-typed, 2023-10-22:
I cannot reproduce it with Linux (using lima on macOS):

```
(ros_env) sascha@lima-default:/tmp/ros2_ws$ mamba deactivate && mamba activate ros_env
(ros_env) sascha@lima-default:/tmp/ros2_ws$ colcon build --symlink-install --packages-select urdf_tutorial_r2d2
Starting >>> urdf_tutorial_r2d2
--- stderr: urdf_tutorial_r2d2
/home/sascha.linux/miniconda3/envs/ros_env/lib/python3.10/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/sascha.linux/miniconda3/envs/ros_env/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< urdf_tutorial_r2d2 [3.11s]

Summary: 1 package finished [3.68s]
  1 package had stderr output: urdf_tutorial_r2d2
(ros_env) sascha@lima-default:/tmp/ros2_ws$ source install/setup.bash
(ros_env) sascha@lima-default:/tmp/ros2_ws$ ros2 launch urdf_tutorial_r2d2 demo_launch.py
[INFO] [launch]: All log files can be found below /home/sascha.linux/.ros/log/2023-10-22-20-53-55-219162-lima-default-24149
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [24150]
[INFO] [state_publisher-2]: process started with pid [24152]
[robot_state_publisher-1] [INFO] [1698008035.363621336] [robot_state_publisher]: got segment axis
[robot_state_publisher-1] [INFO] [1698008035.364168310] [robot_state_publisher]: got segment body
[robot_state_publisher-1] [INFO] [1698008035.364574001] [robot_state_publisher]: got segment box
[robot_state_publisher-1] [INFO] [1698008035.364911223] [robot_state_publisher]: got segment head
[robot_state_publisher-1] [INFO] [1698008035.365335660] [robot_state_publisher]: got segment leg1
[robot_state_publisher-1] [INFO] [1698008035.365869315] [robot_state_publisher]: got segment leg2
[robot_state_publisher-1] [INFO] [1698008035.366214504] [robot_state_publisher]: got segment rod
[state_publisher-2] OpenBLAS WARNING - could not determine the L2 cache size on this system, assuming 256k
[state_publisher-2] [INFO] [1698008035.602978474] [state_publisher]: state_publisher started
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1698008043.180156511] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 24150]
[INFO] [state_publisher-2]: process has finished cleanly [pid 24152]
(ros_env) sascha@lima-default:/tmp/ros2_ws$ mamba install --yes ros-humble-sdformat-urdf

Looking for: ['ros-humble-sdformat-urdf']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
robostack-staging/linux-64                                    No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.10.*

Transaction

  Prefix: /home/sascha.linux/miniconda3/envs/ros_env

  Updating specs:

   - ros-humble-sdformat-urdf
   - ca-certificates
   - certifi
   - openssl

  Package                     Version  Build            Channel                Size
─────────────────────────────────────────────────────────────────────────────────────
  Install:
─────────────────────────────────────────────────────────────────────────────────────

  + urdfdom_headers             1.1.0  h924138e_0       conda-forge            18kB
  + cli11                       2.3.2  hcb278e6_0       conda-forge            75kB
  + gdbm                         1.18  h0a1914f_2       conda-forge           195kB
  + urdfdom                     3.1.1  h8a9b4ce_0       conda-forge           114kB
  + libignition-utils1          1.4.0  h27087fc_0       conda-forge            28kB
  + ruby                        3.2.2  hff50039_0       conda-forge             8MB
  + libignition-common4         4.6.1  hedc608d_1       conda-forge           673kB
[... 38 lines cut ...]
Finished <<< urdf_tutorial_r2d2 [3.39s]

Summary: 1 package finished [3.98s]
  1 package had stderr output: urdf_tutorial_r2d2
(ros_env) sascha@lima-default:/tmp/ros2_ws$ source install/setup.bash
(ros_env) sascha@lima-default:/tmp/ros2_ws$ ros2 launch urdf_tutorial_r2d2 demo_launch.py
[INFO] [launch]: All log files can be found below /home/sascha.linux/.ros/log/2023-10-22-20-54-50-800224-lima-default-26219
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [26220]
[INFO] [state_publisher-2]: process started with pid [26222]
[robot_state_publisher-1] [INFO] [1698008090.921886105] [robot_state_publisher]: got segment axis
[robot_state_publisher-1] [INFO] [1698008090.921985781] [robot_state_publisher]: got segment body
[robot_state_publisher-1] [INFO] [1698008090.921993005] [robot_state_publisher]: got segment box
[robot_state_publisher-1] [INFO] [1698008090.921997224] [robot_state_publisher]: got segment head
[robot_state_publisher-1] [INFO] [1698008090.922020312] [robot_state_publisher]: got segment leg1
[robot_state_publisher-1] [INFO] [1698008090.922023841] [robot_state_publisher]: got segment leg2
[robot_state_publisher-1] [INFO] [1698008090.922027280] [robot_state_publisher]: got segment rod
[state_publisher-2] OpenBLAS WARNING - could not determine the L2 cache size on this system, assuming 256k
[state_publisher-2] [INFO] [1698008091.137399029] [state_publisher]: state_publisher started
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1698008093.510333282] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 26220]
[INFO] [state_publisher-2]: process has finished cleanly [pid 26222]
(ros_env) sascha@lima-default:/tmp/ros2_ws$
```

strongly-typed, 2023-10-22:
The solution seems to be

`ln -s libsdformat_urdf_plugin.so libsdformat_urdf_plugin.dylib` in `/usr/local/Caskroom/miniconda/base/envs/ros_env/lib`

Info from here: [ArduPilot/ardupilot_gz#21](https://github.com/ArduPilot/ardupilot_gz#21-library-extension)

Might it be possible to add this as a patch to robostack?

Tobias-Fischer, 2023-10-23:
Good debugging!

@traversaro - do you think it's better to do `set(CMAKE_SHARED_LIBRARY_SUFFIX ".dylib")` or to create a symlink such that both ".so" and ".dylib" are created?

Tobias-Fischer, 2023-11-06:
Quick reminder @traversaro

traversaro, 2023-11-07:
> @traversaro - do you think it's better to do `set(CMAKE_SHARED_LIBRARY_SUFFIX ".dylib")` or to create a symlink such that both ".so" and ".dylib" are created?

I guess that `libsdformat_urdf_plugin.so` ends with `.so` on macOS instead of `.dylib` as it is a `MODULE` , see https://github.com/ros/sdformat_urdf/blob/ddf83ed07e5fa5aeaea1aa2609ba3d2a03102f20/sdformat_urdf/CMakeLists.txt#L72 , so if we want it to end with `.dylib` I guess we need to simply change `MODULE` to `SHARED` in https://github.com/ros/sdformat_urdf/blob/ddf83ed07e5fa5aeaea1aa2609ba3d2a03102f20/sdformat_urdf/CMakeLists.txt#L72 , as done for the similar urdf parser plugin in https://github.com/ros2/urdf/blob/1d257eb6a3fa34593c0da67f5d16cd0155d7d91b/urdf/CMakeLists.txt#L58 . I opened a PR upstream for this in https://github.com/ros/sdformat_urdf/pull/22 .

Note that the root issue is https://github.com/ros/pluginlib/issues/200 .

Tobias-Fischer, 2023-11-07:
Thanks a lot @traversaro! I've added your patch here so this is being resolved in the next rebuild.

Tobias-Fischer, 2024-02-06:
This should be fixed with the latest packages - please let us know if not.

closed 2024-02-06 by Tobias-Fischer