RoboStack/ros-humble #106: Error Encountered When Running 'gz gui' Command for Gazebo GUI Access

fodaysahid, 2023-09-24:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Experienced the error below when running the `gz gui` command to access Gazebo GUI.
[image]

### Installed packages

```shell
# Name                    Version                   Build  Channel
aiohttp                   3.8.5           py310h2aa6e3c_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
argcomplete               3.1.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
boost                     1.78.0          py310h629746b_4    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.0.9                h1a8c8d9_9    conda-forge
brotli-bin                1.0.9                h1a8c8d9_9    conda-forge
bullet                    3.21                 h69fb684_4    conda-forge
bullet-cpp                3.21            py310h2b830bf_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.1               hb547adb_0    conda-forge
c-compiler                1.6.0                hd291e01_0    conda-forge
ca-certificates           2023.7.22            hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hd1ac623_14    conda-forge
cctools_osx-arm64         973.0.1             h2a25c60_14    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h2399d43_3    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
clang                     15.0.7               hce30654_3    conda-forge
clang-15                  15.0.7          default_h5dc8d65_3    conda-forge
clang_osx-arm64           15.0.7               h77e971b_3    conda-forge
clangxx                   15.0.7          default_h610c423_3    conda-forge
clangxx_osx-arm64         15.0.7               h768a7fd_3    conda-forge
cli11                     2.3.2                hb7217d7_0    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3                    pypi_0    pypi
colcon-bash               0.4.2                    pypi_0    pypi
colcon-cd                 0.1.1                    pypi_0    pypi
colcon-cmake              0.2.27                   pypi_0    pypi
colcon-common-extensions  0.3.0                    pypi_0    pypi
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8                    pypi_0    pypi
colcon-devtools           0.2.3                    pypi_0    pypi
colcon-library-path       0.2.1                    pypi_0    pypi
colcon-metadata           0.2.5                    pypi_0    pypi
colcon-notification       0.2.15          py310hbe9552e_1    conda-forge
colcon-output             0.2.13                   pypi_0    pypi
colcon-package-information 0.3.3                    pypi_0    pypi
colcon-package-selection  0.2.10                   pypi_0    pypi
colcon-parallel-executor  0.2.4                    pypi_0    pypi
colcon-pkg-config         0.1.0                    pypi_0    pypi
colcon-powershell         0.3.7                    pypi_0    pypi
colcon-python-setup-py    0.2.8                    pypi_0    pypi
colcon-recursive-crawl    0.2.1                    pypi_0    pypi
colcon-ros                0.3.23                   pypi_0    pypi
colcon-test-result        0.3.8                    pypi_0    pypi
colcon-zsh                0.4.0                    pypi_0    pypi
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
[... 632 lines cut ...]
         python version : 3.11.4.final.0
       virtual packages : __archspec=1=arm64
                          __osx=13.6=0
                          __unix=0=0
       base environment : /Users/fsnk/miniconda3  (writable)
      conda av data dir : /Users/fsnk/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/osx-arm64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-arm64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/fsnk/miniconda3/pkgs
                          /Users/fsnk/.conda/pkgs
       envs directories : /Users/fsnk/miniconda3/envs
                          /Users/fsnk/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.7.4 requests/2.31.0 CPython/3.11.4 Darwin/22.6.0 OSX/13.6
                UID:GID : 501:20
             netrc file : /Users/fsnk/.netrc
           offline mode : False
```

Tobias-Fischer, 2023-09-25:
Hi @fodaysahid - thanks for reporting this! It appears as if this is an issue that is not specific to RoboStack. It is probably best to open an issue in https://github.com/conda-forge/libignition-gazebo-feedstock/issues

/cc @traversaro

traversaro, 2023-09-25:
Hello @fodaysahid ! It seems that you are running `gz gui`, that is a command present since Gazebo Garden (i.e. gz-sim 7), while ROS Humble uses Ignition Fortress (i.e. ignition-gazebo 6). However, I guess that you have installed Gazebo Garden via homebrew, and hence the error message as you try to mix homebrew and conda libraries (that in general it is not supported, as you could have the same library installed both by homebrew and by conda, with different ABIs).

Can you try to run `ign gui`? On my macOS arm64 system it works fine in an environment similar to you.

However, note that `ign gui` is not the gui for the Gazebo simulator. That is launched via `ign gazebo -g` (or `gz sim -g`, since Gazebo Garden), and as of Ignition Fortress it is not supported on macOS, see https://github.com/gazebosim/gz-sim/issues/44 , as it is only supported with Gazebo Garden, see https://github.com/gazebosim/gz-sim/pull/1225 .

traversaro, 2023-09-25:
xref: https://github.com/conda-forge/gz-sim-feedstock/issues/24 .

fodaysahid, 2023-09-25:
> Hi @fodaysahid - thanks for reporting this! It appears as if this is an issue that is not specific to RoboStack. It is probably best to open an issue in https://github.com/conda-forge/libignition-gazebo-feedstock/issues
> 
> /cc @traversaro

Thank you @Tobias-Fischer

Tobias-Fischer, 2023-11-06:
Hi @fodaysahid @traversaro - can we close here, or is this still an issue?

closed 2023-12-19 by Tobias-Fischer