RoboStack/ros-humble #107: Upstream MoveIt2 patches

traversaro, 2023-09-27:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

@tylerjw mentioned that he is interested in upstreaming some of the patches that we use here for MoveIt2, I open this issue to add some comments to the existing patches.

### Installed packages

```shell
.
```

### Environment info

```shell
.
```

traversaro, 2023-09-27:
See discussion in https://github.com/ros-planning/moveit2/issues/2391 . On our side, probably we can just remove:
* https://github.com/RoboStack/ros-humble/blob/main/patch/ros-humble-moveit-common.win.patch
* https://github.com/RoboStack/ros-humble/blob/main/patch/ros-humble-moveit-planners-ompl.patch

Tobias-Fischer, 2024-01-28:
Closing here as tracked upstream

closed 2024-01-28 by Tobias-Fischer