RoboStack/ros-humble #108: Missing local_setup.ps1 on win-64 ros humble

ardiya, 2023-09-29:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Getting the following error. Missing local_setup.ps1
```
&: C:\Users\xxx\Software\Miniforge\etc\conda\activate.d\ros-humble-ros-workspace_activate.ps1:10
Line |
  10 |  & "${env:CONDA_PREFIX}\Library\local_setup.ps1"
     |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     | The term 'C:\Users\xxx\Software\Miniforge\Library\local_setup.ps1' is not recognized as a name of a cmdlet, function, script file, or executable program. Check the spelling of the name, or if a path
     | was included, verify that the path is correct and try again.
```

For reference, I can see the `local_setup.bat` but not `local_setup.ps1`. I think we need to find how the `.ps1` is linked/created. Not sure if issue from robostack or from ros/ament.
[image]

Reproduce steps:
```
mamba create -n ros_env
mamba activate ros_env

conda config --env --add channels conda-forge
conda config --env --add channels robostack-staging
conda config --env --remove channels defaults

mamba install ros-humble-ros-base ros-humble-ros-core
```

Open new tab and then the error message above starts appearing.

### Installed packages

```shell
argcomplete               3.1.2              pyhd8ed1ab_0    conda-forge
boltons                   23.0.0             pyhd8ed1ab_0    conda-forge
brotli-python             1.1.0           py310h00ffb61_0    conda-forge
bullet                    3.25                 h3bcd3f7_0    conda-forge
bullet-cpp                3.25                 hecd3228_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cairo                     1.16.0            h412253b_1017    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h628cb3f_3    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
cmake                     3.27.4               hf0feee3_4    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
conda                     23.3.1          py310h5588dad_0    conda-forge
conda-libmamba-solver     23.3.0             pyhd8ed1ab_0    conda-forge
conda-package-handling    2.2.0              pyh38be061_0    conda-forge
conda-package-streaming   0.9.0              pyhd8ed1ab_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppcheck                  2.12.1          py310h47ce786_1    conda-forge
cryptography              41.0.3          py310h6e82f81_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1          py310h5588dad_2    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
flake8                    6.1.0              pyhd8ed1ab_0    conda-forge
fmt                       9.1.0                h181d51b_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
[... 349 lines cut ...]
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=11.6=0
                          __win=0=0
       base environment : C:\Users\xxx\Software\miniforge  (writable)
      conda av data dir : C:\Users\xxx\Software\miniforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/robostack/win-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\xxx\Software\miniforge\pkgs
                          C:\Users\xxx\.conda\pkgs
                          C:\Users\xxx\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\xxx\Software\miniforge\envs
                          C:\Users\xxx\.conda\envs
                          C:\Users\xxx\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Windows/10 Windows/10.0.19045
          administrator : False
             netrc file : None
           offline mode : Fals
```

traversaro, 2023-09-29:
As mentioned in the docs (in https://robostack.github.io/GettingStarted.html), powershell is not supported/tested at the moment, see https://github.com/RoboStack/ros-galactic/issues/91 . Some work was done in https://github.com/RoboStack/vinca/pull/34, but not tested in depth. I guess the issue is on the vinca/robostack side.

ardiya, 2023-09-29:
Thanks for quick response @traversaro. I am able to get ros-galactic to work with win64.

Will check more regarding ros-humble when I am off work.

Tobias-Fischer, 2023-12-06:
Ah, great find! `colcon-powershell` already exists on conda-forge. Seems to be a matter of finding out how to incorporate it into the build process.

/cc @traversaro

Tobias-Fischer, 2023-12-07:
I guess one issue is that we use catkin/ament for building our packages, not colcon. So I'm not sure how it would work, we might need to change the build process potentially.

Tobias-Fischer, 2023-12-07:
The harder / better way would probably be to add templates for ament_package at https://github.com/ament/ament_package/tree/rolling/ament_package/template

traversaro, 2025-07-04:
In case someone is looking at the issue, note that this is mostly a problem for `conda` users, while `pixi` handles activation scripts differently, using `.bat` script on all shells, and so it is not affected by this problem.