RoboStack/ros-humble #110: Not able to install slam toolbox

Akshit0601, 2023-10-10:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Not able to install Slam toolbox in ROS humble on Linux aarch64 due to dependency issues.
Error:

Could not solve for environment specs
The following package could not be installed
ros-humble-slam-toolbox   is installable and it requires
   xorg-libx11 >=1.8.6,<2.0a0 , which requires
       libxcb >=1.15,<1.16.0a0 , which can be installed.
 tried with the given three version '2.6.3', '2.6.4', '2.6.5' but the error persists

### Installed packages

```shell
# Name                    Version                   Build  Channel
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-aarch64_curr_repodata_hack 4                   h57d6b7b_13    conda-forge
aiohttp                   3.8.6           py310hb299538_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h4e544f5_0    conda-forge
aom                       3.5.0                headf329_0    conda-forge
argcomplete               3.1.2              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h3fd5656_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hf4e84e4_1    conda-forge
attr                      2.5.1                h4e544f5_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
binutils                  2.40                 h64c2a2e_0    conda-forge
binutils_impl_linux-aarch64 2.40                 h870a726_0    conda-forge
binutils_linux-aarch64    2.40                 h94bbfa1_2    conda-forge
blosc                     1.21.5               h2f3a684_0    conda-forge
boost                     1.78.0          py310h5e48e15_5    conda-forge
boost-cpp                 1.78.0               h484ac13_2    conda-forge
brotli                    1.1.0                h31becfc_1    conda-forge
brotli-bin                1.1.0                h31becfc_1    conda-forge
bullet                    3.21                 h7d5ade6_4    conda-forge
bullet-cpp                3.21            py310h9e1bc84_4    conda-forge
bzip2                     1.0.8                hf897c2e_4    conda-forge
c-ares                    1.19.1               h31becfc_0    conda-forge
c-compiler                1.6.0                h31becfc_0    conda-forge
ca-certificates           2023.7.22            hcefe29a_0    conda-forge
cairo                     1.16.0            hd19fb6e_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
ceres-solver              2.1.0                h45b26fb_2    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py310hce94938_0    conda-forge
cfitsio                   4.2.0                h8b262d6_0    conda-forge
charset-normalizer        3.3.0              pyhd8ed1ab_0    conda-forge
chrpath                   0.16              hf897c2e_1002    conda-forge
cli11                     2.3.2                hd600fc2_0    conda-forge
cmake                     3.26.3               h5819af1_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py310h4c7bcd0_1    conda-forge
colcon-core               0.15.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.5              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
[... 319 lines cut ...]
       virtual packages : __archspec=1=aarch64
                          __glibc=2.35=0
                          __linux=6.2.0=0
                          __unix=0=0
       base environment : /home/akshit/miniforge3  (writable)
      conda av data dir : /home/akshit/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-aarch64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/robostack-humble/linux-aarch64
                          https://conda.anaconda.org/robostack-humble/noarch
                          https://conda.anaconda.org/conda-forge/linux-aarch64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack/linux-aarch64
                          https://conda.anaconda.org/robostack/noarch
          package cache : /home/akshit/miniforge3/pkgs
                          /home/akshit/.conda/pkgs
       envs directories : /home/akshit/miniforge3/envs
                          /home/akshit/.conda/envs
               platform : linux-aarch64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Linux/6.2.0-34-generic ubuntu/22.04.3 glibc/2.35
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Akshit0601, 2023-10-10:
Already tried that, it does install ros-humble-slam-toolbox but then I'm unable to install other humble packages like gazebo etc. It again shows incompatibility with the version of installed packges

traversaro, 2023-10-10:
I was able to reproduce your problem in a clean environment:
~~~
traversaro@IITICUBLAP257:~$ CONDA_SUBDIR=linux-aarch64 mamba create -n test110  -c conda-forge -c robostack-sttraversaro@IITICUBLAP257:~$ CONDA_SUBDIR=linux-aarch64 mamba create -n test110  -c conda-forge -c robostack-staging gazebo ros-humble-slam-toolbox

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.4.2) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['gazebo', 'ros-humble-slam-toolbox']

warning  libmamba Cache file "/home/traversaro/miniforge3/pkgs/cache/d9d93fba.json" was modified by another program
warning  libmamba Cache file "/home/traversaro/miniforge3/pkgs/cache/09cdf8bf.json" was modified by another program
robostack-staging/noarch                           135.0 B @  93.0 B/s  1.6s
robostack-staging/linux-aarch64                    202.4kB @ 109.2kB/s  2.0s
conda-forge/linux-aarch64                           10.4MB @   3.7MB/s  3.8s
conda-forge/noarch                                  14.1MB @   4.9MB/s  4.0s
Could not solve for environment specs
The following packages are incompatible
├─ gazebo   is installable with the potential options
│  ├─ gazebo 11.12.0 would require
│  │  └─ libcurl >=8.1.0,<9.0a0 , which can be installed;
│  ├─ gazebo 11.11.0 would require
│  │  └─ boost-cpp >=1.74.0,<1.74.1.0a0 , which can be installed;
│  ├─ gazebo [11.11.0|11.12.0] would require
│  │  └─ libxcb >=1.13,<1.14.0a0 , which can be installed;
│  ├─ gazebo 11.13.0 would require
│  │  └─ libcurl >=8.1.2,<9.0a0 , which can be installed;
│  ├─ gazebo [11.13.0|11.14.0] would require
│  │  └─ assimp >=5.3.1,<5.3.2.0a0 , which can be installed;
│  ├─ gazebo 11.13.0 would require
│  │  └─ libcurl >=8.3.0,<9.0a0 , which can be installed;
│  └─ gazebo 11.13.0 would require
│     └─ libcurl >=8.2.1,<9.0a0 , which can be installed;
└─ ros-humble-slam-toolbox   is uninstallable because it requires
   ├─ ros-humble-rviz-default-plugins   with the potential options
   │  ├─ ros-humble-rviz-default-plugins 11.2.5 would require
   │  │  └─ ros-humble-resource-retriever   with the potential options
   │  │     ├─ ros-humble-resource-retriever 3.1.1 would require
   │  │     │  └─ ros-humble-libcurl-vendor   with the potential options
   │  │     │     ├─ ros-humble-libcurl-vendor 3.1.1 would require
   │  │     │     │  └─ libcurl >=7.87.0,<8.0a0 , which conflicts with any installable versions previously reported;
   │  │     │     └─ ros-humble-libcurl-vendor 3.1.0 would require
   │  │     │        └─ ros2-distro-mutex 0.2.* , which can be installed;
   │  │     └─ ros-humble-resource-retriever 3.1.0 would require
   │  │        └─ ros2-distro-mutex 0.2.* , which can be installed;
   │  └─ ros-humble-rviz-default-plugins 11.2.4 would require
   │     └─ ros2-distro-mutex 0.2.* humble, which can be installed;
   ├─ ros-humble-rviz-ogre-vendor   with the potential options
   │  ├─ ros-humble-rviz-ogre-vendor 11.2.5 would require
   │  │  └─ assimp >=5.2.5,<5.2.6.0a0 , which conflicts with any installable versions previously reported;
   │  └─ ros-humble-rviz-ogre-vendor 11.2.4 would require
   │     └─ ros2-distro-mutex 0.2.* , which can be installed;
   ├─ ros2-distro-mutex 0.3.* , which requires
   │  └─ boost-cpp 1.78.* , which conflicts with any installable versions previously reported;
   └─ xorg-libx11 >=1.8.6,<2.0a0 , which requires
      └─ libxcb >=1.15,<1.16.0a0 , which conflicts with any installable versions previously reported.
~~~

traversaro, 2023-10-10:
Apparently it seems that there are no compatible packages (w.r.t. to pinnings) between `gazebò` and `ros-humble-slam-toolbox`, while it works fine in linux-64 . Just to understand, did this combination ever worked in the past for you on `linux-aarch64`?

Tobias-Fischer, 2023-11-06:
Looks like this needs a full rebuild with updated pinnings.

Tobias-Fischer, 2024-02-09:
This is done.

closed 2024-02-09 by Tobias-Fischer