RoboStack/ros-humble #112: Unable to include GL/gl.h

jfrancis71, 2023-11-03:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I am attempting to build a ROS node for the ORB3 SLAM monocular camera setup but am running into difficulties.

I can compile up Pangolin/ORB3 fine using the system wide installed compiler. However I understand that if I wish to integrate with RoboStack ROS2 I need to be using the compiler from the conda environment.

The core of the problem appears to be that after running the installation local tools command as per installation instructions I have a file:
/miniconda3/envs/ros2_4/include/GL/glu.h which has on line 38:
#include <GL/gl.h>

These are the files in my ~/miniconda3/envs/ros2_4/include/GL/:

-rw-rw-r-- 6 julian julian   11008 Apr  2  2018 freeglut_ext.h
-rw-rw-r-- 6 julian julian     681 Oct 21  2003 freeglut.h
-rw-rw-r-- 6 julian julian   27097 Feb 16  2022 freeglut_std.h
-rw-rw-r-- 6 julian julian    5871 Mar 26  2018 freeglut_ucall.h
-rw-rw-r-- 6 julian julian 1186601 Jul 31  2017 glew.h
-rw-rw-r-- 6 julian julian   17255 Mar 29  2020 glu.h
-rw-rw-r-- 6 julian julian    3315 Mar 29  2020 glu_mangle.h
-rw-rw-r-- 6 julian julian     639 Oct 21  2003 glut.h
-rw-rw-r-- 6 julian julian   73435 Jul 31  2017 glxew.h
-rw-rw-r-- 6 julian julian   63314 Jul 31  2017 wglew.h

As you can see there is no gl.h

However this is present in:
/usr/include/GL/gl.h
ie the system wide location. So the system wide compiler does find this file, but the conda compiler does not.

My reading of the documentation around Core Dependency Tree packages seems to suggest that the intention is for the conda compiler to fall back on the system wide files for this, but this does not seem to be happening. I am not sure I have fully understood this part, so I may be have misunderstood and any assistance would be appreciated.

Here is a simple program (test.cc) to illustrate:

    #include <GL/glu.h>
    #include <stdio.h>

    int
    main(int argc, char* argv)
    {
        printf("Hello I'm gl");
        return 0;
    }

and CMakeLists.txt:

    cmake_minimum_required(VERSION 3.10.0)

    project(fred)

    add_executable(${PROJECT_NAME} test.cc)

error message:
/home/julian/miniconda3/envs/ros2_4/include/GL/glu.h:38:10: fatal error: GL/gl.h: No such file or directory
   38 | #include <GL/gl.h>

Any help much appreciated.

### Installed packages

```shell
# packages in environment at /home/julian/miniconda3/envs/ros2_4:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohttp                   3.8.6           py310h2372a71_1    conda-forge
[... 657 lines cut ...]
          conda version : 23.5.0
    conda-build version : not installed
         python version : 3.10.10.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.1=0
                          __glibc=2.27=0
                          __linux=5.4.0=0
                          __unix=0=0
       base environment : /home/julian/miniconda3  (writable)
      conda av data dir : /home/julian/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/julian/miniconda3/pkgs
                          /home/julian/.conda/pkgs
       envs directories : /home/julian/miniconda3/envs
                          /home/julian/.conda/envs
               platform : linux-64
             user-agent : conda/23.5.0 requests/2.29.0 CPython/3.10.10 Linux/5.4.0-94-generic ubuntu/18.04.6 glibc/2.27 solver/libmamba conda-libmamba-solver/23.5.0 libmambapy/1.4.1
                UID:GID : 1000:1000
             netrc file : /home/julian/.netrc
           offline mode : False
```

traversaro, 2023-11-03:
> My reading of the documentation around Core Dependency Tree packages seems to suggest that the intention is for the conda compiler to fall back on the system wide files for this, but this does not [...]

I see the confusion here. Basically for OpenGL and similar conda-forge relies on the system version for what regards **loading libraries** when executing an executable or a shared library that uses it, but when **compiling** some C/C++ code that include those headers, it expects the cdt packages to be installed. Just for reference, the CDT libraries that we install while compiling packages that require opengl are:
* https://github.com/RoboStack/vinca/blob/f1d8bd35c705323faa620a6d5d59d18f8abb9d43/vinca/main.py#L494
* https://github.com/RoboStack/vinca/blob/f1d8bd35c705323faa620a6d5d59d18f8abb9d43/vinca/main.py#L835

So either:
~~~
mamba install -c conda-forge -c robotology-staging mesa-libgl-devel-cos7-x86_64 mesa-dri-drivers-cos7-x86_64 libselinux-cos7-x86_64 libxxf86vm-cos7-x86_64
~~~
or 
~~~
mamba install -c conda-forge -c robotology-staging mesa-libgl-devel-cos7-x86_64 mesa-dri-drivers-cos7-x86_64 libselinux-cos7-x86_64 libxdamage-cos7-x86_64 libxxf86vm-cos7-x86_64 libxfixes-cos7-x86_64 libxext-cos7-x86_64 libxau-cos7-x86_64 xorg-libx11 xorg-libxext
~~~

should solve your problem.

jfrancis71, 2023-11-07:
Excellent: Your first solution worked in my system conda setup.

Additionally: I am suspicious that the cmake package management plays well with conda package management, so I setup up 2 docker images to test both of your solutions, and both of them work.

I wonder if this mamba install information belongs in the install instructions or perhaps in the FAQ? I don't think I would have discovered it by myself without your help.

I now have the ORB3 examples running in ROS 2 RoboStack environment (now to get it running connected to a live camera!)

My lego robot and I thank you for your help, we are most appreciative.

closed 2023-11-07 by jfrancis71

traversaro, 2023-11-07:
> I wonder if this mamba install information belongs in the install instructions or perhaps in the FAQ? I don't think I would have discovered it by myself without your help.

Sure, would you be interested in proposing a PR against https://github.com/RoboStack/robostack.github.io/blob/master/docs/FAQ.md or somewhere in the docs? Thanks!

nmarticorena, 2024-02-07:
Hi I recently found myself in a similar problem and reading this issue help me solving it, I notice thought that the Pull request submited by @jfrancis71  was in your fork and not againts the RoboStack one.  

If you agree I would be more than happy to resubmit your PR againts the RoboStack one if you prefer.

traversaro, 2024-02-07:
> Hi I recently found myself in a similar problem and reading this issue help me solving it, I notice thought that the Pull request submited by @jfrancis71  was in your fork and not againts the [...]

Yes, thanks!

jfrancis71, 2024-02-07:
Hi all,
Apologies, I must have got the process slightly wrong. Please feel free to go ahead and get the pull request submitted correctly.
Thanks,
Julian.